chipweinberger/flutter_blue_plus #951: [Help]: Why can't my reading and listening receive any information

743306400, 2024-07-31:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.8

### Flutter Version

3.19.0

### What OS?

Android

### OS Version

Android 9

### Bluetooth Module

esp-12s

### What is your problem?

Why can't my reading and listening receive any information

connect:
[image]
write:
[image]
read:
[image]

### Logs

```shell
no
```

chipweinberger, 2024-07-31:
esp problem

743306400, 2024-07-31:
Sorry, I don't quite understand. Is it because the plugin doesn't support this ESP, or is it a bug on the ESP side that didn't put the attribute back

743306400, 2024-07-31:
> esp problem

Sorry, I don't quite understand. Is it because the plugin doesn't support this ESP, or is it a bug on the ESP side that didn't put the attribute back