chipweinberger/flutter_blue_plus #959: [Help]: Delay when disconnect

xOldeVx, 2024-08-08:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

5.3.1

### Flutter Version

3.22.1

### What OS?

Android

### OS Version

14

### Bluetooth Module

Nordic nRF52

### What is your problem?

When I'm connecting to the ble device the `device.connectionState.listen` fire immediately a state (connected) but when I'm power off my ble device, I get the disconnected event after about 5 seconds. Is there a way to change it? or to check if connecting manually?

### Logs

```shell
.
```

chipweinberger, 2024-08-08:
nothing we can do

closed 2024-08-08 by chipweinberger

xOldeVx, 2024-08-09:
Is there a way to check it manually (if the device is connect.. for check before performing a ble action)