chipweinberger/flutter_blue_plus #962: [Help]: Unused UUID on the BLE device causes error code 14 on IOS ONLY.

TheCodeDaniel, 2024-08-11:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.11

### Flutter Version

3.19.2

### What OS?

iOS

### OS Version

17.5.1

### Bluetooth Module

Nordic nRF5

### What is your problem?

I noticed an from my BLE device testing, this issue only occurs on IOS, it does not on Android.
If the device has an unused UUID and the mobile device of the user is IOS, they get an error: 
`Flutterblueplusexception readcharacteristic apple code 14 unlikely error fix ⁠`.

I don't get an error code 14 on android, the app works fine on that platform, but on the IOS platform, i get the error code 14,

I'm opening this issue so that just incase it's possible for the flutter_blue_plus team to handle the issue on IOS so that a issue of unused UUID does not cause error apple code 14 again ? just like it works fine on Android.

We later had to do some checks on the firmware to resolve it.

The package is really great though. everything else works flawlessly so far,

Thanks.

### Logs

```shell
 Flutterblueplusexception readcharacteristic apple code 14 unlikely error fix
```

chipweinberger, 2024-08-11:
List of Bluetooth GATT Errors
These GATT error codes are part of the BLE Specification.

**These are responses from your ble device because you are sending an invalid request.

FlutterBluePlus cannot fix these errors. You are doing something wrong & your device is responding with an error.**

GATT errors as they appear on iOS:

apple-code: 1  | The handle is invalid.
apple-code: 2  | Reading is not permitted.
apple-code: 3  | Writing is not permitted.
apple-code: 4  | The command is invalid.
apple-code: 6  | The request is not supported.
apple-code: 7  | The offset is invalid.
apple-code: 8  | Authorization is insufficient.
apple-code: 9  | The prepare queue is full.
apple-code: 10 | The attribute could not be found.
apple-code: 11 | The attribute is not long.
apple-code: 12 | The encryption key size is insufficient.
apple-code: 13 | The value's length is invalid.
apple-code: 14 | Unlikely error.
apple-code: 15 | Encryption is insufficient.
apple-code: 16 | The group type is unsupported.
apple-code: 17 | Resources are insufficient.
apple-code: 18 | Unknown ATT error.

chipweinberger, 2024-08-11:
im not sure why it works on android. but its an issue in your code.

closed 2024-08-11 by chipweinberger