chipweinberger/flutter_blue_plus #966: [Help]: ble pairing request popup appears twice

hiperioncn, 2024-08-14:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.11

### Flutter Version

3.22.2

### What OS?

Android

### OS Version

android14

### Bluetooth Module

esp32

### What is your problem?

When I run the example app, click "scan" button, show ble request popup twice. but when I run "BLE Scanner" only show ble request popup one time.
So I think it is not a Android OS bug.
Please watch the screen recording.

https://github.com/user-attachments/assets/cbf74962-43c4-45dc-9c54-926b28bd8f61

### Logs

```shell
nothing
```

chipweinberger, 2024-08-14:
this is in the readme already

search existing issue as well

closed 2024-08-14 by chipweinberger

hiperioncn, 2024-08-15:
I have already read "readme", but it just show how to solve when connect device.
You means below code?
`
final bsSubscription = **device**.bondState.listen((value) {
    print("$value prev:{$device.prevBondState}");
});

// cleanup: cancel subscription when disconnected
device.cancelWhenDisconnected(bsSubscription);

// Force the bonding popup to show now (Android Only) 
await device.createBond();

// remove bond
await device.removeBond();
`
But when search device, it doesn't know which devcie shoud be connected, so how to use below code.

Thanks a lot.

hiperioncn, 2024-08-15:
Does the example add code how to solve android pairing popup appears twice?
If not, could you add and test it?
Thank you.

hiperioncn, 2024-08-15:
I have try to add createbond when search, but it throws exception:
code:
`bool isCreateBond = false;
    searchStream = FlutterBluePlus.onScanResults.listen((results) async {
      if (results.isNotEmpty) {
        for (ScanResult r in results) {
          if (!isCreateBond) {
            isCreateBond = true;
            final bsSubscription = r.device.bondState.listen((value) {
              logs("$value bondState:${r.device.prevBondState}");
            });
            r.device.cancelWhenDisconnected(bsSubscription);
            await r.device.createBond();
            await r.device.removeBond();
          }
          String scanDeviceName = r.device.platformName;
       }
     }
   }
`
exception:

`Other error: FlutterBluePlusException | createBond | fbp-code: 6 | device is not connected
main.dart(182) StackTrace:#0      BluetoothDevice.createBond (package:flutter_blue_plus/src/bluetooth_device.dart:525:7)
BlueHelper.searchDevice.<anonymous closure> (package:watermonitor/core/utils/ble/blue_helper.dart:223:28)
_rootRunUnary (dart:async/zone.dart:1407:47)
_CustomZone.runUnary (dart:async/zone.dart:1308:19)
_CustomZone.runUnaryGuarded (dart:async/zone.dart:1217:7)
_BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:365:11)
_DelayedData.perform (dart:async/stream_impl.dart:541:14)
_PendingEvents.handleNext (dart:async/stream_impl.dart:646:11)
_PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:617:7)
_rootRun (dart:async/zone.dart:1391:47)
_CustomZone.run (dart:async/zone.dart:1301:19)
_CustomZone.runGuarded (dart:async/zone.dart:1209:7)
_CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1249:23) `

It maybe means createBond methond must call after decice connected

So how to solve pairing popup appears twice when searching?