chipweinberger/flutter_blue_plus #967: Unable to obtain Bluetooth devices connected to the system

xcjpdx, 2024-08-14:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.7

### Flutter Version

3.22.3

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

systemDevices

### What is your problem?

The following image is my permission list
[image]
The following is a code snippet of mine. What I want to implement is to obtain the required permissions and access the Bluetooth devices connected to the system：
import 'dart:convert';
import 'dart:async';
import 'package:flutter_blue_plus/flutter_blue_plus.dart' as flutter_blue;
import 'package:permission_handler/permission_handler.dart';
import './globals.dart';
import '../apis/app.dart';
import 'package:flutter/services.dart';
import 'dart:io';
import 'package:flutter_easyloading/flutter_easyloading.dart';
import 'dart:ui';

class BluetoothService {
  flutter_blue.BluetoothCharacteristic? targetCharacteristic;

  List<flutter_blue.BluetoothDevice> _systemDevices = [];
  flutter_blue.BluetoothDevice? targetDevice;
  StreamSubscription<List<int>>? _characteristicSubscription;

  bool isCheck = false; // 是否正在检查蓝牙权限
  bool isConnect = false; // 是否可以连接
  bool isLoad = false; // 是否正在加载

  String remoteId = ''; // 已连接蓝牙的唯一标识

  String systemLanguage = window.locale.languageCode; // 获取系统语言

  void init() async {
    Timer.periodic(Duration(milliseconds: 500), (timer) async {
      if (isConnect == false) {
        await _checkPermissions();
      }
      connectAndGetMacInfo();
    });
  }

  _checkPermissions() async {
    if (isCheck) {
      return;
    }
    isCheck = true;
    if (Platform.isAndroid) {
      Map<Permission, PermissionStatus> statuses = await [
        Permission.bluetoothConnect,
        Permission.bluetoothScan,
        Permission.locationWhenInUse,
      ].request();
      if (statuses.isEmpty ||
          statuses.values.any((status) => status.isDenied)) {
        // print('安卓蓝牙权限未打开');
        isConnect = false;
      } else {
        // print('安卓蓝牙权限已打开');
        isConnect = true;
      }
    } else {
      Map<Permission, PermissionStatus> statuses = await [
        Permission.bluetooth,
        Permission.bluetoothConnect,
        Permission.bluetoothScan,
        Permission.locationWhenInUse,
      ].request();
      if (statuses.values.any((status) => status.isDenied)) {
        // print('iOS蓝牙权限未打开');
        isConnect = false;
      } else {
        // print('iOS蓝牙权限已打开');
        isConnect = true;
      }
    }
    isCheck = false;
  }

  connectAndGetMacInfo() async {
    if (isConnect == false) return;
    if (isLoad) return;
    isLoad = true;
    try {
      _systemDevices = await flutter_blue.FlutterBluePlus.systemDevices;
      // print('蓝牙服务 搜索到的系统蓝牙设备: ${_systemDevices.length}个 $_systemDevices');
------------------------------------------------------------------------------------------------------------
Problem description: I can obtain system connected Bluetooth devices normally on the Chinese version of Samsung phones or other brands (such as Xiaomi, Vivo, Huawei) of Android phones, but I have been unable to obtain connected Bluetooth devices on the American version of Samsung S20

### Logs

```shell
null
```

chipweinberger, 2024-08-14:
I dont know the reason.

If you fix it, please tell me.

chipweinberger, 2024-08-15:
i think this is a OS or permissions issue

closed 2024-08-15 by chipweinberger