chipweinberger/flutter_blue_plus #970: [Help]: Can't parse the key bytes from advertisementData.manufacturerData

xOldeVx, 2024-08-18:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

5.3.1

### Flutter Version

3.22.1

### What OS?

Android

### OS Version

14

### Bluetooth Module

Nordic nRF52

### What is your problem?

I moving my code from `flutter_reactive_ble` to this package, in `flutter_reactive_ble` I could get the _device-id_ in one simple variable, in this package i noticed that the _device-id_ is live in a `Map manufacturerData`, the first 2 bytes is in the `key` and the rest (6 bytes in my case) is in the `value`, however I couldn't to parse the 2 first bytes correctly, instead of getting _L2_ I'm getting _㉌_ symbol.

Here's my result:
[image]

I would like to know how to parse it to correct String.

By the way it can be very useful to combine it in the package side like in flutter_reactive_ble.
[image]

### Logs

```shell
Above
```

chipweinberger, 2024-08-19:
the two bytes are probably reversed from what you expect.

chipweinberger, 2024-08-19:
or use AdvertisementData.msd

closed 2024-08-19 by chipweinberger