chipweinberger/flutter_blue_plus #972: [Help]: ble pairing request popup appears twice

hiperioncn, 2024-08-19:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.11

### Flutter Version

3.22.2

### What OS?

Android

### OS Version

android14

### Bluetooth Module

esp32

### What is your problem?

When I run the example app, click "scan" button, show ble request popup twice. but when I run "BLE Scanner" only show ble request popup one time.

 I have already read "readme", but it just show how to solve when connect device.
 I have try to add createbond when search, but it throws exception:
```
Other error: FlutterBluePlusException | createBond | fbp-code: 6 | device is not connected
main.dart(182) StackTrace:#0      BluetoothDevice.createBond (package:flutter_blue_plus/src/bluetooth_device.dart:525:7)
BlueHelper.searchDevice.<anonymous closure> (package:watermonitor/core/utils/ble/blue_helper.dart:223:28)
_rootRunUnary (dart:async/zone.dart:1407:47)
_CustomZone.runUnary (dart:async/zone.dart:1308:19)
_CustomZone.runUnaryGuarded (dart:async/zone.dart:1217:7)
_BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:365:11)
_DelayedData.perform (dart:async/stream_impl.dart:541:14)
_PendingEvents.handleNext (dart:async/stream_impl.dart:646:11)
_PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:617:7)
_rootRun (dart:async/zone.dart:1391:47)
_CustomZone.run (dart:async/zone.dart:1301:19)
_CustomZone.runGuarded (dart:async/zone.dart:1209:7)
_CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1249:23) 
```
 It maybe means createBond method must call after decice connected

So how to solve pairing popup appears twice when searching?

More info from https://github.com/boskokg/flutter_blue_plus/issues/966

### Logs

```shell
none
```

chipweinberger, 2024-08-19:
If I could ban you from creating issues, I would. 

You did not read FAQ in the readme: https://github.com/boskokg/flutter_blue_plus?tab=readme-ov-file#android-pairing-popup-appears-twice

chipweinberger, 2024-08-19:
thats not a pairing request. It's a permissions popup.

closed 2024-08-19 by chipweinberger

hiperioncn, 2024-08-21:
So why the example show permission popup twice？What can i do for it？

chipweinberger, 2024-08-21:
we ask for 2 permissions when scanning.

```
                    if (Build.VERSION.SDK_INT >= 31) { // Android 12 (October 2021)
                        permissions.add(Manifest.permission.BLUETOOTH_SCAN);
                        if (androidUsesFineLocation) {
                            permissions.add(Manifest.permission.ACCESS_FINE_LOCATION);
                        }
                        // it is unclear why this is needed, but some phones throw a
                        // SecurityException AdapterService getRemoteName, without it
                        permissions.add(Manifest.permission.BLUETOOTH_CONNECT);
                    }
```

but we only ask 1 at a time. you'll need to submit a PR to fix it. 

```
    private void askPermission(List<String> permissionsNeeded, OperationOnPermission operation)
    {
        // finished asking for permission? call callback
        if (permissionsNeeded.isEmpty()) {
            operation.op(true, null);
            return;
        }

        String nextPermission = permissionsNeeded.remove(0);

        operationsOnPermission.put(lastEventId, (granted, perm) -> {
            operationsOnPermission.remove(lastEventId);
            if (!granted) {
                operation.op(false, perm);
                return;
            }
            // recursively ask for next permission
            askPermission(permissionsNeeded, operation);
        });

        ActivityCompat.requestPermissions(
                activityBinding.getActivity(),
                new String[]{nextPermission},
                lastEventId);

        lastEventId++;
    }

```

chipweinberger, 2024-08-21:
you can try this code. You'll need to change the FBP code.

```
@Override
public boolean onRequestPermissionsResult(int requestCode,
                                          String[] permissions,
                                          int[] grantResults) {
    OperationOnPermission operation = operationsOnPermission.get(requestCode);

    if (operation != null && grantResults.length > 0) {
        for (int i = 0; i < permissions.length; i++) {
            if (grantResults[i] != PackageManager.PERMISSION_GRANTED) {
                // If any permission is denied, call the operation with false
                operation.op(false, permissions[i]);
                return true;
            }
        }
        // All permissions are granted
        operation.op(true, null);
        return true;
    } else {
        return false;
    }
}

private void ensurePermissions(List<String> permissions, OperationOnPermission operation) {
    // only request permission we don't already have
    List<String> permissionsNeeded = new ArrayList<>();
    for (String permission : permissions) {
        if (permission != null && ContextCompat.checkSelfPermission(context, permission)
                != PackageManager.PERMISSION_GRANTED) {
            permissionsNeeded.add(permission);
        }
    }

    // no work to do?
    if (permissionsNeeded.isEmpty()) {
        operation.op(true, null);
        return;
    }

    // Request all permissions at once
    askPermission(permissionsNeeded, operation);
}

private void askPermission(List<String> permissionsNeeded, OperationOnPermission operation) {
    // finished asking for permission? call callback
    if (permissionsNeeded.isEmpty()) {
        operation.op(true, null);
        return;
    }

    // Store the operation for later use in onRequestPermissionsResult
    operationsOnPermission.put(lastEventId, operation);

    // Request all needed permissions at once
    ActivityCompat.requestPermissions(
            activityBinding.getActivity(),
            permissionsNeeded.toArray(new String[0]),
            lastEventId);

    lastEventId++;
}
```