chipweinberger/flutter_blue_plus #973: [Help]: What if i just want to know if the bluetooth enabled or not what permissions should i put in android manifest

shehabmohamed0, 2024-08-21:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.12

### Flutter Version

3.24.0

### What OS?

Android

### OS Version

for all version

### Bluetooth Module

Don't know

### What is your problem?

What if i just want to know if the bluetooth enabled or not what permissions should i put in android manifest, and should i put uses-feature ? 

### Logs

```shell
_
```

chipweinberger, 2024-08-21:
not sure

closed 2024-08-21 by chipweinberger

tnc1997, 2024-08-21:
Hi @shehabmohamed0, if I am understanding your question correctly, you would like know the Android permissions that are required in order to get the state of adapter, which allows you to determine whether Bluetooth is on/off.

```dart
FlutterBluePlus.adapterState.listen(
  (adapterState) {
    if (adapterState == BluetoothAdapterState.on) {

    }
  },
);
```

For apps targeting [Build.VERSION_CODES#R](https://developer.android.com/reference/android/os/Build.VERSION_CODES#R) ([API level 30](https://developer.android.com/guide/topics/manifest/uses-sdk-element#ApiLevels)) or lower, this requires the [Manifest.permission#BLUETOOTH](https://developer.android.com/reference/android/Manifest.permission#BLUETOOTH) permission which can be gained with a simple <uses-permission> manifest tag:

```xml
<uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
```