chipweinberger/flutter_blue_plus #974: [Help]: Fultter Blue Plus 1.4 version does not support Android 14

2eagle88, 2024-08-22:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.4.0

### Flutter Version

3.0.5

### What OS?

Android

### OS Version

14

### Bluetooth Module

No errors involving the use of reported uuids

### What is your problem?

Accessing Bluetooth will close the app

### Logs

```shell
no log uuid question
```

2eagle88, 2024-08-22:
java.lang.ExceptionInInitializerError
	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$2.onScanResult(FlutterBluePlusPlugin.java:887)
	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:579)
	at android.os.Handler.handleCallback(Handler.java:958)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:224)
	at android.os.Looper.loop(Looper.java:318)
	at android.app.ActivityThread.main(ActivityThread.java:8777)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:561)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1013)
Caused by: java.lang.IllegalArgumentException: UUID string too large
	at java.util.UUID.fromStringJava11(UUID.java:238)
	at java.util.UUID.fromString(UUID.java:226)
	at com.boskokg.flutter_blue_plus.ProtoMaker.<clinit>(ProtoMaker.java:35)
	... 10 more

chipweinberger, 2024-08-22:
your FBP version is no longer supported.

closed 2024-08-22 by chipweinberger

MohsinP07, 2024-08-28:
Hey @2eagle88 did you find any solution as I am facing same issue.

MohsinP07, 2024-08-28:
@chipweinberger from which versions FBP supports Android 14, as I core feature is dependent on FBP and I am using 1.4.0 version and its works perfectly b only issue is in case of Android 14.

Saphney, 2024-09-15:
Does anyone have any infirmation in this?

MohsinP07, 2024-09-16:
@Saphney I solved this by updating FBS to latest version(1.32.12).

Saphney, 2024-09-16:
> @Saphney I solved this by updating FBS to latest version(1.32.12).

That sounds promising. Unfortunately I don't know what you mean with FBS.

MohsinP07, 2024-09-17:
> > @Saphney I solved this by updating FBS to latest version(1.32.12).
> 
> That sounds promising. Unfortunately I don't know what you mean with FBS.

Hey actually I mean flutter_blue_plus latest version,

chipweinberger, 2024-09-17:
upgrading is pretty easy. just some variable renames mostly.

https://github.com/chipweinberger/flutter_blue_plus/blob/master/MIGRATION.md

Saphney, 2024-09-17:
Thanks for the feedback. It's what CoPilot told me yesterday too. With it's help I got the app running with FB+. I am quit pleased, especially since I am not familiar with Flutter.

MohsinP07, 2024-09-17:
Yes @chipweinberger ! and new version is quite fast and responsive as well.