chipweinberger/flutter_blue_plus #977: [Help]: MacOS Terminating app due to uncaught exception 'NSInvalidArgumentException

czhah, 2024-08-24:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.8

### Flutter Version

3.3.4

### What OS?

macOS

### OS Version

11.5.2

### Bluetooth Module

MT6580 has a built-in integrated 6627 Bluetooth chip

### What is your problem?

Bluetooth automatically disconnects after connecting, and crashes when reconnecting

### Logs

```shell
Process:               appconnection [790]
Path:                  /Applications/appconnection.app/Contents/MacOS/appconnection
Identifier:            com.cz.appconnection
Version:               1.0.0 (1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           appconnection [790]
User ID:               502

Date/Time:             2024-07-26 10:16:12.669 +0200
OS Version:            macOS 11.5.2 (20G95)
Report Version:        12
Anonymous UUID:        12207EAF-E682-0289-20AD-06DF4A8F4D0D

Sleep/Wake UUID:       73E59CFE-77BC-4B2F-A4CE-E1757491C5D6

Time Awake Since Boot: 1800 seconds
Time Since Wake:       110000000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]: attempt to insert nil object from objects[1]'
abort() called
terminating with uncaught exception of type NSException

Application Specific Backtrace 1:
0   CoreFoundation                      0x00007fff2069a83b __exceptionPreprocess + 242
1   libobjc.A.dylib                     0x00007fff203d2d92 objc_exception_throw + 48
2   CoreFoundation                      0x00007fff207502aa _CFThrowFormattedException + 202
3   CoreFoundation                      0x00007fff2075bd91 -[__NSPlaceholderDictionary initWithCapacity:].cold.1 + 0
4   CoreFoundation                      0x00007fff205a9e5d -[__NSPlaceholderDictionary initWithObjects:forKeys:count:] + 251
5   CoreFoundation                      0x00007fff205a9d48 +[NSDictionary dictionaryWithObjects:forKeys:count:] + 49
6   flutter_blue_plus                   0x000000010ca4b3f3 -[FlutterBluePlusPlugin bmBluetoothCharacteristic:characteristic:] + 2490
7   flutter_blue_plus                   0x000000010ca4a602 -[FlutterBluePlusPlugin bmBluetoothService:service:] + 311
8   flutter_blue_plus                   0x000000010ca461ba -[FlutterBluePlusPlugin peripheral:didDiscoverDescriptorsForCharacteristic:error:] + 1303
9   CoreBluetooth                       0x00007fff30abbcaa -[CBPeripheral handleAttributeEvent:args:attributeSelector:delegateSelector:delegateFlag:] + 219
10  CoreBluetooth                       0x00007fff30abbde1 -[CBPeripheral handleCharacteristicEvent:characteristicSelector:delegateSelector:delegateFlag:] + 113
11  CoreBluetooth                       0x00007fff30ab84f6 -[CBPeripheral handleMsg:args:] + 291
12  CoreBluetooth                       0x00007fff30aaddda -[CBCentralManager handleMsg:args:] + 189
13  CoreBluetooth                       0x00007fff30aa0037 -[CBManager xpcConnectionDidReceiveMsg:args:] + 177
14  CoreBluetooth                       0x00007fff30a9ff73 __30-[CBXpcConnection _handleMsg:]_block_invoke + 66
15  libdispatch.dylib                   0x00007fff2037c623 _dispatch_call_block_and_release + 12
16  libdispatch.dylib                   0x00007fff2037d806 _dispatch_client_callout + 8
17  libdispatch.dylib                   0x00007fff203835ea _dispatch_lane_serial_drain + 606
18  libdispatch.dylib                   0x00007fff203840e0 _dispatch_lane_invoke + 417
19  libdispatch.dylib                   0x00007fff20389a08 _dispatch_main_queue_callback_4CF + 613
20  CoreFoundation                      0x00007fff2065d5a8 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
21  CoreFoundation                      0x00007fff2061f7a2 __CFRunLoopRun + 2755
22  CoreFoundation                      0x00007fff2061e61c CFRunLoopRunSpecific + 563
23  HIToolbox                           0x00007fff28864a83 RunCurrentEventLoopInMode + 292
24  HIToolbox                           0x00007fff288647e5 ReceiveNextEventCommon + 587
25  HIToolbox                           0x00007fff28864583 _BlockUntilNextEventMatchingListInModeWithFilter + 70
26  AppKit                              0x00007fff22e26172 _DPSNextEvent + 864
27  AppKit                              0x00007fff22e24945 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1364
28  AppKit                              0x00007fff22e16c69 -[NSApplication run] + 586
29  AppKit                              0x00007fff22deae6c NSApplicationMain + 816
30  appconnection                       0x000000010ca2b919 main + 9
31  libdyld.dylib                       0x00007fff20542f3d start + 1
```

chipweinberger, 2024-08-26:
more information needed

"attempt to insert nil object from objects[1]'"

which key is nil? please add logs to the obj-c code

chipweinberger, 2024-08-26:
add logs here

```
- (NSDictionary*)bmBluetoothCharacteristic:(CBPeripheral *)peripheral
                            characteristic:(CBCharacteristic *)characteristic
{
    // descriptors
    NSMutableArray *descriptors = [NSMutableArray new];
    for (CBDescriptor *d in [characteristic descriptors])
    {
        // See: BmBluetoothDescriptor
        NSDictionary* desc = @{
            @"remote_id":              [peripheral.identifier UUIDString],
            @"service_uuid":           [d.characteristic.service.UUID uuidStr],
            @"secondary_service_uuid": [NSNull null],
            @"characteristic_uuid":    [d.characteristic.UUID uuidStr],
            @"descriptor_uuid":        [d.UUID uuidStr],
        };

        [descriptors addObject:desc];
    }

    ServicePair *pair = [self getServicePair:peripheral characteristic:characteristic];

    CBCharacteristicProperties props = characteristic.properties;

    // See: BmCharacteristicProperties
    NSDictionary* propsMap = @{
        @"broadcast":                    @((props & CBCharacteristicPropertyBroadcast) != 0),
        @"read":                         @((props & CBCharacteristicPropertyRead) != 0),
        @"write_without_response":       @((props & CBCharacteristicPropertyWriteWithoutResponse) != 0),
        @"write":                        @((props & CBCharacteristicPropertyWrite) != 0),
        @"notify":                       @((props & CBCharacteristicPropertyNotify) != 0),
        @"indicate":                     @((props & CBCharacteristicPropertyIndicate) != 0),
        @"authenticated_signed_writes":  @((props & CBCharacteristicPropertyAuthenticatedSignedWrites) != 0),
        @"extended_properties":          @((props & CBCharacteristicPropertyExtendedProperties) != 0),
        @"notify_encryption_required":   @((props & CBCharacteristicPropertyNotifyEncryptionRequired) != 0),
        @"indicate_encryption_required": @((props & CBCharacteristicPropertyIndicateEncryptionRequired) != 0),
    };

    // See BmBluetoothCharacteristic
    return @{
        @"remote_id":              [peripheral.identifier UUIDString],
        @"service_uuid":           [pair.primary.UUID uuidStr],
        @"secondary_service_uuid": pair.secondary ? [pair.secondary.UUID uuidStr] : [NSNull null],
        @"characteristic_uuid":    [characteristic.UUID uuidStr],
        @"descriptors":            descriptors,
        @"properties":             propsMap,
    };
}
```

chipweinberger, 2024-08-27:
closing until we have enough data to fix it.

closed 2024-08-27 by chipweinberger