chipweinberger/flutter_blue_plus #984: [Help]: Android 14+ devices default to requesting an MTU of 517

ztaylor54, 2024-09-01:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.12

### Flutter Version

3.24.0

### What OS?

Android

### OS Version

android 14

### Bluetooth Module

Espressif ESP32-S3-DevkitC-M1

### What is your problem?

I am trying to implement the `splitWrite` function described in the documentation, but the Android platform implementation is blocking my attempt at writing data longer than 512 bytes:

```
I/flutter (25835): PlatformException(writeCharacteristic, data longer than allowed. dataLen: 514 > max: 512 (withResponse, noLongWrite), null, null)
```

I've got callbacks registered on my device firmware that confirm the MTU is changed to 517 bytes immediately after a connection is established:

```
[ 42609][I][device.cpp:329] onConnect(): [device] BLE client connected
[ 43455][I][device.cpp:364] onMtuChanged(): [device] BLE MTU changed to 517
```

Attempts to renegotiate MTU fail, as Android has changed MTU negotiation behavior starting in Android 14: https://developer.android.com/about/versions/14/behavior-changes-all#mtu-set-to-517

TL;DR, we can no longer negotiate a preferred MTU, and since 517 is greater than the max of 512 enforced by this library, I cannot actually perform batched writes unless I change the MTU negotiation on my ESP32 device.

Is there an easy workaround here, or does the library need to be updated? Thanks!

### Logs

```shell
Referenced in the issue description.
```

chipweinberger, 2024-09-01:
just enforce a max data len of 512 in your splitWrite function

thanks for reporting

chipweinberger, 2024-09-01:
```
import 'dart:math';
// split write should be used with caution.
//    1. due to splitting, `characteristic.read()` will return partial data.
//    2. it can only be used *with* response to avoid data loss
//    3. The characteristic must be designed to support split data
extension splitWrite on BluetoothCharacteristic {
  Future<void> splitWrite(List<int> value, {int timeout = 15}) async {
    int chunk = min(device.mtuNow - 3, 512); // 3 bytes BLE overhead, 512 bytes max
    for (int i = 0; i < value.length; i += chunk) {
      List<int> subvalue = value.sublist(i, min(i + chunk, value.length));
      await write(subvalue, withoutResponse:false, timeout: timeout);
    }
  }
}
```

closed 2024-09-01 by chipweinberger

ztaylor54, 2024-09-01:
Thanks for the quick reply, your suggestion worked great!