chipweinberger/flutter_blue_plus #985: [Help]: setLogLevel always request for permission

beastoin, 2024-09-03:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.12

### Flutter Version

3.24.1

### What OS?

iOS

### OS Version

iOS 17

### Bluetooth Module

nrf52840

### What is your problem?

Whenever calling for the setLogLevel, I see the request permission popup.

### Logs

```shell
None.
```

chipweinberger, 2024-09-03:
is this a new issue? only in 1.32.12?

chipweinberger, 2024-09-03:
try 1.32.13

beastoin, 2024-09-03:
yes man, I just saw that fixes. https://github.com/chipweinberger/flutter_blue_plus/blob/3abe40b3b86c3cc5084ff362599130bc6e091d66/lib/src/flutter_blue_plus.dart#L599

thanks!

closed 2024-09-03 by beastoin

beastoin, 2024-09-03:
Ah man, @chipweinberger , what do you think about the default value of LogLevel ? I mean should we use INFO instead of DEBUG ? 

As you know that if the application was published to user with default level as DEBUG, logs are expensived in term of resource, so the performance will be dramatically impacted.

thịnh