chipweinberger/flutter_blue_plus #986: [Help]: setNotifyValue | fbp-code: 1 | Timed out after 15s for a particular characteristic

devtyagi-c, 2024-09-05:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.12

### Flutter Version

3.22.3

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

Nordic nRF

### What is your problem?

My app flow is :
DeviceDiscoveryPage --> List Devices --> Tap on one device --> Move to DeviceMonitoringPage --> Connect to Device --> Click on Start Monitoring Button --> Perform the below set of BLE Operations:
    1. Enable notifications on characteristic 86f66001-f706-58a0-95b2-1fb9261e4dc7
    2. Enable notifications on characteristic 86f65002-f706-58a0-95b2-1fb9261e4dc7
    3. Write 0x0213C87462 on characteristic 86f65001-f706-58a0-95b2-1fb9261e4dc7
    4. Write 0x66 on characteristic 86f65001-f706-58a0-95b2-1fb9261e4dc7

I am able to enable notifications on the 6001 characteristic but keep getting `FlutterBluePlusException | setNotifyValue | fbp-code: 1 | Timed out after 15s` for the 5002 characteristic.

### Logs

```shell
D/BluetoothGatt(32345): onClientConnectionState() - status=0 clientIf=12 device=CE:BA:5C:8E:E6:C2
D/[FBP-Android](32345): [FBP] onConnectionStateChange:connected
D/[FBP-Android](32345): [FBP]   status: SUCCESS
D/BluetoothGatt(32345): onConfigureMTU() - Device=CE:BA:5C:8E:E6:C2 mtu=247 status=0
D/[FBP-Android](32345): [FBP] onMtuChanged:
D/[FBP-Android](32345): [FBP]   mtu: 247
D/[FBP-Android](32345): [FBP]   status: GATT_SUCCESS (0)
[log] ConnectDevice: Connection attempt completed
D/BluetoothGatt(32345): onConnectionUpdated() - Device=CE:BA:5C:8E:E6:C2 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(32345): onConnectionUpdated() - Device=CE:BA:5C:8E:E6:C2 interval=36 latency=0 timeout=500 status=0
D/BluetoothGatt(32345): onConnectionUpdated() - Device=CE:BA:5C:8E:E6:C2 interval=36 latency=0 timeout=600 status=0
2
V/AutofillManager(32345): requestHideFillUi(null): anchor = null
D/[FBP-Android](32345): [FBP] onMethodCall: discoverServices
D/BluetoothGatt(32345): discoverServices() - device: CE:BA:5C:8E:E6:C2
D/BluetoothGatt(32345): onSearchComplete() = Device=CE:BA:5C:8E:E6:C2 Status=0
D/[FBP-Android](32345): [FBP] onServicesDiscovered:
D/[FBP-Android](32345): [FBP]   count: 8
D/[FBP-Android](32345): [FBP]   status: 0GATT_SUCCESS
D/[FBP-Android](32345): [FBP] onMethodCall: setNotifyValue
D/BluetoothGatt(32345): setCharacteristicNotification() - uuid: 00002a05-0000-1000-8000-00805f9b34fb enable: true
D/[FBP-Android](32345): [FBP] onDescriptorWrite:
D/[FBP-Android](32345): [FBP]   chr: 2a05
D/[FBP-Android](32345): [FBP]   desc: 2902
D/[FBP-Android](32345): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](32345): [FBP] onMethodCall: setNotifyValue
D/BluetoothGatt(32345): setCharacteristicNotification() - uuid: 86f66001-f706-58a0-95b2-1fb9261e4dc7 enable: true
D/[FBP-Android](32345): [FBP] onDescriptorWrite:
D/[FBP-Android](32345): [FBP]   chr: 86f66001-f706-58a0-95b2-1fb9261e4dc7
D/[FBP-Android](32345): [FBP]   desc: 2902
D/[FBP-Android](32345): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](32345): [FBP] onMethodCall: setNotifyValue
D/BluetoothGatt(32345): setCharacteristicNotification() - uuid: 86f65002-f706-58a0-95b2-1fb9261e4dc7 enable: true
[log] Error: FlutterBluePlusException | setNotifyValue | fbp-code: 1 | Timed out after 15s
```

devtyagi-c, 2024-09-05:
This is happening only on Android :/ @chipweinberger

chipweinberger, 2024-09-05:
this is likely a problem with your peripheral

you can try increasing the timeout

devtyagi-c, 2024-09-08:
@chipweinberger I did increase the timeout, it still doesn't work

I tried with the nRF Connect app and other methods, my peripheral works fine

Tried with this custom python script on my laptop:
```python
import asyncio
from bleak import BleakClient, BleakScanner

# Replace with your BLE device's address
device_address = "E909871B-A209-3AE7-3195-F7C3EAD5EA7D"

# Replace with your BLE characteristic UUIDs
characteristic1_uuid = "86f66001-f706-58a0-95b2-1fb9261e4dc7"
characteristic2_uuid = "86f65002-f706-58a0-95b2-1fb9261e4dc7"
characteristic3_uuid = "86f65001-f706-58a0-95b2-1fb9261e4dc7"

# Define the data to be written
data1 = bytearray([0x02, 0x13, 0xC8, 0x74, 0x62])
data2 = bytearray([0x66])
# data2 = bytearray([0x98])

# Notification handler
async def notification_handler(sender, data):
    print(f"Notification from {sender}: {data}")

async def main():
    async with BleakClient(device_address) as client:
        print(f"Connected: {client.is_connected}")

        # Enable notifications on characteristic1
        await client.start_notify(characteristic1_uuid, notification_handler)
        print(f"Notifications enabled on {characteristic1_uuid}")

        # Enable notifications on characteristic2
        await client.start_notify(characteristic2_uuid, notification_handler)
        print(f"Notifications enabled on {characteristic2_uuid}")

        # Write 0x0213C87462 on characteristic3
        await client.write_gatt_char(characteristic3_uuid, data1)
        print(f"Written {data1} to {characteristic3_uuid}")

        # Write 0x66 on characteristic3
        await client.write_gatt_char(characteristic3_uuid, data2)
        print(f"Written {data2} to {characteristic3_uuid}")

        # Keep the connection open to receive notifications
        while True:
            await asyncio.sleep(1)

# Run the main function
asyncio.run(main())
```

it works just fine

chipweinberger, 2024-09-08:
okay. you'll need to debug the FBP code

chipweinberger, 2024-09-08:
from your logs it looks like android is not responding with `onDescriptorWrite`

so seems like android problem

devtyagi-c, 2024-09-13:
`from your logs it looks like android is not responding with onDescriptorWrite`

Most likely yes, I tried everything... Can't seem to find any solution :/

chipweinberger, 2024-09-17:
closing. not enough information to solve the problem. 

please reopen a new issue if you can show there is a problem in FBP.

closed 2024-09-17 by chipweinberger