chipweinberger/flutter_blue_plus #987: [Help]: How can I track the fact that bluetooth devices are connected/disconnected to a smartphone, if this devices connect not from my app?

GAG888, 2024-09-06:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.12

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

12

### Bluetooth Module

-

### What is your problem?

How can I track the fact that bluetooth devices are connected/disconnected to a smartphone, if this devices connect not from my app?_

### Logs

```shell
-
```

chipweinberger, 2024-09-06:
check if it is in `systemDevices`

closed 2024-09-06 by chipweinberger

efraespada, 2024-09-09:
@chipweinberger  @GAG888  The `systemDevices` is returning an empty list:

[image]

There are BT devices connected to my Android smartphone.