chipweinberger/flutter_blue_plus #988: [Help]: Bluetooth adapter was disabled, but the connection was not disconnected on the Android 14 device

cow-yoon, 2024-09-08:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.12

### Flutter Version

3.24.2

### What OS?

Android

### OS Version

Android 14

### Bluetooth Module

Nordic nRF52832

### What is your problem?

**Scenario:**

On android 14,
1. connect() is called (with autoConnect: true)
2. Bluetooth is working normally, then Bluetooth is disabled on the smartphone.
3. disconnect() is called.
4. disconnect() does not work properly (gatt.close() is not called). (work properly on Android 13 and under)

---

When the autoConnect parameter is set to true in the connect() method, if Bluetooth is disabled (BluetoothAdapterState.off) and the disconnect() method is called, gatt.close() is not invoked.

In FlutterBluePlusPlugin.java, when newState is BluetoothProfile.STATE_DISCONNECTED, if the condition (mAutoConnected.containsKey(remoteId)) is true, only the log message "autoconnect is true. skipping gatt.close()" is currently logged. 
However, I believe that if mBluetoothAdapter.isEnabled() == false, gatt.close() should be called.

Ex)
```
if (mAutoConnected.containsKey(remoteId)) {
    /// add ~
    if (mBluetoothAdapter.isEnabled() == false) {
        gatt.close();
    } else {
        log(LogLevel.DEBUG, "autoconnect is true. skipping gatt.close()");
    }
    /// ~ add
} else {
    // it is important to close after disconnection, otherwise we will 
    // quickly run out of bluetooth resources, preventing new connections
    gatt.close();
}
```

---

I would appreciate it if you could let me know if there is another way to do gatt.close() in the above situation.
Thank you for reading my problem. 

### Logs

```shell
// bluetooth adapter turnOff on android cell phone
D/[FBP-Android](17240): [FBP] OnAdapterStateChanged: turningOff
D/[FBP-Android](17240): [FBP] disconnectAllDevices(adapterTurnOff)
D/[FBP-Android](17240): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](17240): [FBP]   status: SUCCESS
D/[FBP-Android](17240): [FBP] autoconnect is true. skipping gatt.close()
I/flutter (17240): [FBP] [[ OnAdapterStateChanged ]] result: {adapter_state: 5}
I/flutter (17240): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: E9:1B:A2:A0:11:34, connection_state: 0}
D/[FBP-Android](17240): [FBP] onCharacteristicChanged:
D/[FBP-Android](17240): [FBP]   chr: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
D/[FBP-Android](17240): [FBP] OnAdapterStateChanged: off
D/[FBP-Android](17240): [FBP] disconnectAllDevices(adapterTurnOff)

-----------------------------------------------------------------------
// call disconnect() method
D/[FBP-Android](17240): [FBP] onMethodCall: disconnect
D/[FBP-Android](17240): [FBP] already disconnected
```

chipweinberger, 2024-09-08:
if bluetooth is off, how is your phone still connected to the device? seems impossible

>  "However, I believe that if mBluetoothAdapter.isEnabled() == false, gatt.close() should be called."

please try it. modify the FBP code and try it.

chipweinberger, 2024-09-13:
closing until there is more evidence the connection is still active after bluetooth is turned off.

closed 2024-09-13 by chipweinberger