chipweinberger/flutter_blue_plus #989: [Help]: Can't compile Example for Android

olivierkastel, 2024-09-08:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.13

### Flutter Version

3.24.2

### What OS?

Android

### OS Version

14

### Bluetooth Module

N/A

### What is your problem?

I get this error when compiling the app for Android:

```
Execution failed for task ':app:checkDebugAarMetadata'
```

### Logs

```shell
$ cd example
$ flutter run
Resolving dependencies... 
Downloading packages... 
+ characters 1.3.0
+ collection 1.18.0 (1.19.0 available)
+ flutter 0.0.0 from sdk flutter
+ flutter_blue_plus 1.32.13 from path ..
+ material_color_utilities 0.11.1 (0.12.0 available)
+ meta 1.15.0
+ sky_engine 0.0.99 from sdk flutter
+ vector_math 2.1.4
Changed 8 dependencies!
2 packages have newer versions incompatible with dependency constraints.
Try `flutter pub outdated` for more information.
Launching lib/main.dart on Pixel 6a in debug mode...
You are applying Flutter's app_plugin_loader Gradle plugin imperatively using the apply script method, which is deprecated and will be removed in a future release. Migrate to applying Gradle plugins with the declarative plugins block: https://flutter.dev/to/flutter-gradle-plugin-apply

You are applying Flutter's main Gradle plugin imperatively using the apply script method, which is deprecated and will be removed in a future release. Migrate to applying Gradle plugins with the declarative plugins block: https://flutter.dev/to/flutter-gradle-plugin-apply

Warning: unable to detect project KGP version. Skipping version checking. 
This may be because you have applied KGP after the Flutter Gradle Plugin.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:checkDebugAarMetadata'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.CheckAarMetadataWorkAction
   > 14 issues were found when checking AAR metadata:

       1.  Dependency 'androidx.fragment:fragment:1.7.1' requires libraries and applications that
[... 264 lines cut ...]

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 4s
Running Gradle task 'assembleDebug'...                              5.5s
Error: Gradle task assembleDebug failed with exit code 1

$ flutter doctor
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 3.24.2, on macOS 15.0 24A5331b darwin-arm64 (Rosetta), locale en-US)
[✓] Android toolchain - develop for Android devices (Android SDK version 30.0.3)
[✓] Xcode - develop for iOS and macOS (Xcode 15.4)
[✓] Chrome - develop for the web
[✓] Android Studio (version 2024.1)
[✓] VS Code (version 1.92.2)
[✓] Connected device (6 available)
[✓] Network resources

• No issues found!
```

olivierkastel, 2024-09-08:
Changing compileSdkVersion 34 on the build.gradle fixed the issue.

chipweinberger, 2024-09-08:
please open a PR thanks

chipweinberger, 2024-09-09:
fixed

https://github.com/chipweinberger/flutter_blue_plus/commit/8ea7bed87789e689bf430733275977e18af26cb5

closed 2024-09-09 by chipweinberger