chipweinberger/flutter_blue_plus #993: [Help]: Android race condition between gatt.disconnect and Bluetooth stack establishing connection is still happening

samnbd, 2024-09-10:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.12

### Flutter Version

3.22.3 

### What OS?

Android

### OS Version

13

### Bluetooth Module

BLE watch

### What is your problem?

The issue reported previously [here](https://github.com/chipweinberger/flutter_blue_plus/issues/934) was marked as resolved but I still have the same issue on the latest plugin version **_1.32.12_**
Note that this issue happens when trying to connect a device after disconnecting it.

### Logs

```shell
D/BluetoothGatt(27345): connect() - device: CA:26:03:F6:2E:90, auto: true
D/BluetoothGatt(27345): registerApp()
D/BluetoothGatt(27345): registerApp() - UUID=1e39eef6-9cce-466d-bbb6-be3308119f24
D/BluetoothGatt(27345): onClientRegistered() - status=0 clientIf=8
I/flutter (27345): [FBP] <connect> result: true
12
E/gralloc4(27345): Empty SMPTE 2094-40 data
D/TrafficStats(27345): tagSocket(313) with statsTag=0xffffffff, statsUid=-1
47
E/gralloc4(27345): Empty SMPTE 2094-40 data
I/flutter (27345): [FBP] <disconnect> args: 
D/[FBP-Android](27345): [FBP] onMethodCall: disconnect
D/[FBP-Android](27345): [FBP] already disconnected
I/flutter (27345): [FBP] <disconnect> result: false
E/gralloc4(27345): Empty SMPTE 2094-40 data
I/flutter (27345): [FBP] <connect> args: {remote_id: CA:26:03:F6:2E:90, auto_connect: 1}
D/[FBP-Android](27345): [FBP] onMethodCall: connect
D/[FBP-Android](27345): [FBP] already connecting
I/flutter (27345): [FBP] <connect> result: true
21
E/gralloc4(27345): Empty SMPTE 2094-40 data
D/BluetoothGatt(27345): onClientConnectionState() - status=0 clientIf=8 device=CA:26:03:F6:2E:90
D/[FBP-Android](27345): [FBP] onConnectionStateChange:connected
D/[FBP-Android](27345): [FBP]   status: SUCCESS
I/flutter (27345): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: CA:26:03:F6:2E:90, connection_state: 1}
31
E/gralloc4(27345): Empty SMPTE 2094-40 data
I/flutter (27345): [FBP] <discoverServices> args: CA:26:03:F6:2E:90
D/[FBP-Android](27345): [FBP] onMethodCall: discoverServices
D/BluetoothGatt(27345): discoverServices() - device: CA:26:03:F6:2E:90
D/BluetoothGatt(27345): onSearchComplete() = Device=CA:26:03:F6:2E:90 Status=0
D/[FBP-Android](27345): [FBP] onServicesDiscovered:
D/[FBP-Android](27345): [FBP]   count: 11
D/[FBP-Android](27345): [FBP]   status: 0GATT_SUCCESS
I/flutter (27345): [FBP] <discoverServices> result: true
E/gralloc4(27345): Empty SMPTE 2094-40 data
I/flutter (27345): [FBP] [[ OnDiscoveredServices ]] result: {error_string: GATT_SUCCESS, success: 1, remote_id: CA:26:03:F6:2E:90, error_code: 0, services: [{included_services: [], characteristics: [{descriptors: [], service_uuid: 1800, remote_id: CA:26:03:F6:2E:90, characteristic_uuid: 2a00, properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 1, authenticated_signed_writes: 0, extended_properties: 0, indicate: 0, indicate_encryption_required: 0, write: 0, notify: 0}}, {descriptors: [], service_uuid: 1800, remote_id: CA:26:03:F6:2E:90, characteristic_uuid: 2a01, properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 1, authenticated_signed_writes: 0, extended_properties: 0, indicate: 0, indicate_encryption_required: 0, write: 0, notify: 0}}, {descriptors: [], service_uuid: 1800, remote_id: CA:26:03:F6:2E:90, characteristic_uuid: 2a04, properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 1, authenticated_signed_write
I/flutter (27345): [FBP] <setNotifyValue> args: {remote_id: CA:26:03:F6:2E:90, service_uuid: 1801, secondary_service_uuid: null, characteristic_uuid: 2a05, force_indications: false, enable: true}
D/[FBP-Android](27345): [FBP] onMethodCall: setNotifyValue
D/BluetoothGatt(27345): setCharacteristicNotification() - uuid: 00002a05-0000-1000-8000-00805f9b34fb enable: true
E/gralloc4(27345): Empty SMPTE 2094-40 data
I/flutter (27345): [FBP] <setNotifyValue> result: true
2
E/gralloc4(27345): Empty SMPTE 2094-40 data
D/[FBP-Android](27345): [FBP] onDescriptorWrite:
D/[FBP-Android](27345): [FBP]   chr: 2a05
D/[FBP-Android](27345): [FBP]   desc: 2902
D/[FBP-Android](27345): [FBP]   status: GATT_SUCCESS (0)
I/flutter (27345): [FBP] [[ OnDescriptorWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: 1801, success: 1, remote_id: CA:26:03:F6:2E:90, descriptor_uuid: 2902, error_code: 0, characteristic_uuid: 2a05, value: 0200}
I/flutter (27345): [FBP] <readCharacteristic> args: {remote_id: CA:26:03:F6:2E:90, service_uuid: 180a, secondary_service_uuid: null, characteristic_uuid: 2a26}
D/[FBP-Android](27345): [FBP] onMethodCall: readCharacteristic
E/gralloc4(27345): Empty SMPTE 2094-40 data
I/flutter (27345): [FBP] <readCharacteristic> result: true
2
E/gralloc4(27345): Empty SMPTE 2094-40 data
D/[FBP-Android](27345): [FBP] onCharacteristicRead:
D/[FBP-Android](27345): [FBP]   chr: 2a26
D/[FBP-Android](27345): [FBP]   status: GATT_SUCCESS (0)
I/flutter (27345): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: 180a, success: 1, remote_id: CA:26:03:F6:2E:90, error_code: 0, characteristic_uuid: 2a26, value: 362e31302e30}
E/gralloc4(27345): Empty SMPTE 2094-40 data
I/flutter (27345): [FBP] <writeCharacteristic> args: {remote_id: CA:26:03:F6:2E:90, service_uuid: 156e7000-a300-4fea-897b-86f698d74461, secondary_service_uuid: null, characteristic_uuid: 7001, write_type: 0, allow_long_write: 0, value: 120d0000000000}
D/[FBP-Android](27345): [FBP] onMethodCall: writeCharacteristic
I/flutter (27345): [FBP] <writeCharacteristic> result: true
7
E/gralloc4(27345): Empty SMPTE 2094-40 data
D/[FBP-Android](27345): [FBP] onCharacteristicWrite:
D/[FBP-Android](27345): [FBP]   chr: 7001
D/[FBP-Android](27345): [FBP]   status: GATT_SUCCESS (0)
E/gralloc4(27345): Empty SMPTE 2094-40 data
I/flutter (27345): [FBP] [[ OnCharacteristicWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: 156e7000-a300-4fea-897b-86f698d74461, success: 1, remote_id: CA:26:03:F6:2E:90, error_code: 0, characteristic_uuid: 7001, value: 120d0000000000}
I/flutter (27345): [FBP] <writeCharacteristic> args: {remote_id: CA:26:03:F6:2E:90, service_uuid: 156e1000-a300-4fea-897b-86f698d74461, secondary_service_uuid: null, characteristic_uuid: 1005, write_type: 0, allow_long_write: 0, value: 3052221002092404}
D/[FBP-Android](27345): [FBP] onMethodCall: writeCharacteristic
I/flutter (27345): [FBP] <writeCharacteristic> result: true
2
E/gralloc4(27345): Empty SMPTE 2094-40 data
```

samnbd, 2024-09-10:
Note in this case, the pairing stream emits 2 disconnected states before it emits connected state when trying to connect to the device

chipweinberger, 2024-09-10:
this is not the same issue. 

the previous was a race condition in Android. 

your issue is caused by this:

```
D/[FBP-Android](27345): [FBP] onMethodCall: disconnect
D/[FBP-Android](27345): [FBP] already disconnected
I/flutter (27345): [FBP] <disconnect> result: false
```

chipweinberger, 2024-09-10:
to debug this, I would add logs every time `mCurrentlyConnectingDevices` and `mConnectedDevices`, is updated or checked

in this FBP code below, we check those variables, and in your case it is ignoring the call to disconnect when it should not be.

```
                case "disconnect":
                {
                    String remoteId = (String) call.arguments;

                    // already disconnected?
                    BluetoothGatt gatt = null;
                    if (gatt == null) {
                        gatt = mCurrentlyConnectingDevices.get(remoteId);
                        if (gatt != null) {
                            log(LogLevel.DEBUG, "disconnect: cancelling connection in progress");
                        }
                    }
                    if (gatt == null) {
                        gatt = mConnectedDevices.get(remoteId);
                    }
                    if (gatt == null) {
                        gatt = mAutoConnected.get(remoteId);
                        if (gatt != null) {
                            log(LogLevel.DEBUG, "already disconnected. disabling autoconnect");
                            mAutoConnected.remove(remoteId);
                            gatt.disconnect();
                            gatt.close();
                            result.success(false);  // no work to do
                            return;
                        }
                    }
                    if (gatt == null) {
                        log(LogLevel.DEBUG, "already disconnected");
                        result.success(false);  // no work to do
                        return;
                    }

```

chipweinberger, 2024-09-10:
in your logs, it looks like you are calling 

1. connect
2. disconnect
3. connect

is that right?

chipweinberger, 2024-09-10:
also, please include more logs.

it looks like your first call to `connect` is cut off in the logs

chipweinberger, 2024-09-10:
> Note in this case, the pairing stream emits 2 disconnected states before it emits connected state when trying to connect to the device

I dont see this in your logs

chipweinberger, 2024-09-10:
please include more logs.

samnbd, 2024-09-12:
I'm not calling any disconnect from my end

samnbd, 2024-09-12:
Also, I don't know why did you change the title ?! It's exactly the same issue that is happening in the issue that was marked as resolved [here](https://github.com/chipweinberger/flutter_blue_plus/issues/934)

samnbd, 2024-09-12:
> > Note in this case, the pairing stream emits 2 disconnected states before it emits connected state when trying to connect to the device
> 
> I dont see this in your logs

it's visible in the logs: 
```
I/flutter (27345): [FBP] <disconnect> args: 
D/[FBP-Android](27345): [FBP] onMethodCall: disconnect //first disconnect, this wasn't called by me
D/[FBP-Android](27345): [FBP] already disconnected //second call of disconnect, this also isn't called by me
I/flutter (27345): [FBP] <disconnect> result: false
E/gralloc4(27345): Empty SMPTE 2094-40 data

I/flutter (27345): [FBP] <connect> args: {remote_id: CA:26:03:F6:2E:90, auto_connect: 1}
```

chipweinberger, 2024-09-12:
> I'm not calling any disconnect from my end 

`D/[FBP-Android](27345): [FBP] onMethodCall: disconnect`

you call it right here

chipweinberger, 2024-09-12:
> it's visible in the logs:
>
> ```
>
> I/flutter (27345): [FBP] <disconnect> args:
>
> D/[FBP-Android](27345): [FBP] onMethodCall: disconnect //first disconnect, this wasn't called by me [...]

you are misinterpreting the logs. this is a single call

chipweinberger, 2024-09-13:
> Also, I don't know why did you change the title ?! It's exactly the same issue that is happening in the issue that was marked as resolved [here](https://github.com/chipweinberger/flutter_blue_plus/issues/934)

its a different issue. his had this log:

`D/[FBP-Android](30007): [FBP] disconnect: cancelling connection in progress`

yours does not.

samnbd, 2024-09-13:
As I mentioned before, I'm not calling disconnect at all, it's being emitted by the plugin without me calling it

chipweinberger, 2024-09-13:
"log(LogLevel.DEBUG, "onMethodCall: " + call.method);"

this log only happens when disconnect is called from dart. I don't see anywhere in FBP that we call disconnect from dart. do you?

chipweinberger, 2024-09-13:
maybe you're hitting this code, the connection timeout

```
        BmConnectionStateResponse response = await futureState
            .fbpEnsureAdapterIsOn("connect")
            .fbpTimeout(timeout.inSeconds, "connect")
            .catchError((e) async {
          if (e is FlutterBluePlusException && e.code == FbpErrorCode.timeout.index) {
            await FlutterBluePlus._invokeMethod('disconnect', remoteId.str); // cancel connection attempt
          }
          throw e;
        });
```

chipweinberger, 2024-09-24:
closing. nothing I can do without more data from OP.

closed 2024-09-24 by chipweinberger