chipweinberger/flutter_blue_plus #994: [Help]: IOS, The device is not connecting on autoconnect in the background

Maqcel, 2024-09-11:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.12

### Flutter Version

3.22.2 

### What OS?

iOS

### OS Version

17.6 (21G5066d)

### Bluetooth Module

Own product can't discuss, All I can share: Bluetooth Low Energy device which uses GATT protocols along Bluetooth Advertising packets to transmit temperature readings measured by the device’s sensor.

### What is your problem?

1 off all thank you for this package it's really cool how easy it was to pick up.

```
[FBP-iOS] didConnectPeripheral
[FBP-iOS] didDisconnectPeripheral:
[FBP-iOS]  error: The specified device has disconnected from us.
```

This error cancels autoconnect altogether (in the background), and somehow when disconnecting I'm not getting the connection status change from my stream:
```
  Stream<NotConnected> _listenDisconnected(BluetoothDevice device) =>
      device.connectionState
          .where((state) => state != BluetoothConnectionState.connected)
          .skip(1)
          .map((event) => NotConnected());
```
I tried to find the place in the library responsible for the error but failed to do so. I'm currently unable to reproduce this error in debug mode, but error sometimes persist (The notification about the `NotConnected` is not send only in case this error happening)

### Logs

```shell
By my surprise, I can't replicate same error today when I opened the app to give full logs. But the issue persist in the version on my PM's phone.
```

chipweinberger, 2024-09-11:
"The specified device has disconnected from us."

this is an error string from iOS, not FBP

Maqcel, 2024-09-11:
Thank you for the response, haven't seen 846 will proceed with things from there!

closed 2024-09-13 by chipweinberger

Maqcel, 2024-09-13:
Ahh Yes I forgot to do so, the upgrade from v13 helped! Thanks