chipweinberger/flutter_blue_plus #995: [Help]: Possibility to connect to a hidden device

elvysferrari, 2024-09-13:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.12

### Flutter Version

3.10.5

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

ESP32, libraries BLEServer, BLEClient, BLEUtils, BLEScan, BLEAddress

### What is your problem?

Hi, how are you?

Is it possible for me to connect to a hidden device without having to pair the device?

I have an ESP32 configured to hide Bluetooth, is it possible for me to connect my cell phone to it without having to pair it? It would be similar to connecting via MAC Address on the old Bluetooth.

I'm using the last version of Flutter Blue Plus.

### Logs

```shell
No logs.
```

chipweinberger, 2024-09-13:
> ESP32 configured to hide Bluetooth

what does that mean

chipweinberger, 2024-09-18:
closing. not enough info.

closed 2024-09-18 by chipweinberger