chipweinberger/flutter_blue_plus #996: [Help]: FlutterBluePlus.systemDevices is not working on iOS 18.

anhvu2503, 2024-09-18:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.12

### Flutter Version

3.19.5

### What OS?

iOS

### OS Version

IOS 18

### Bluetooth Module

Nordic nRF52832

### What is your problem?

When my phone is updated to the latest iOS 18 version, and I use `FlutterBluePlus.systemDevices` in my app to retrieve connected system devices, the result returns an empty list. However, when using a phone with an earlier version, everything works fine.

### Logs

```shell
No logs.
```

chipweinberger, 2024-09-18:
seems like it might be a new privacy restriction in iOS:

https://www.reddit.com/r/iOSProgramming/comments/1dctwmo/whats_with_the_new_bluetooth_privacy_features/

chipweinberger, 2024-09-18:
In FBP `systemDevices` is a wrapper around [retrieveConnectedPeripheralsWithServices](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/retrieveconnectedperipherals(withservices:))

We use the Generic Access Service `0x1800`, but perhaps Apple has blocked this in iOS because it basically returns all devices.

Try forking FBP and looking for a specific service instead. We might need to update the API to be compatible with iOS 18.

```
        else if ([@"getSystemDevices" isEqualToString:call.method])
        {
            // Cannot pass blank UUID list for security reasons.
            // Assume all devices have the Generic Access service 0x1800
            CBUUID* gasUuid = [CBUUID UUIDWithString:@"1800"];

            // this returns devices connected by *any* app
            NSArray *periphs = [self.centralManager retrieveConnectedPeripheralsWithServices:@[gasUuid]];

            // Devices
            NSMutableArray *deviceProtos = [NSMutableArray new];
            for (CBPeripheral *p in periphs) {
                [deviceProtos addObject:[self bmBluetoothDevice:p]];
            }

            // See BmDevicesList
            NSDictionary* response = @{
                @"devices": deviceProtos,
            };

            result(response);
        }
```

chipweinberger, 2024-09-18:
if you fix this issue, please open a PR. 

I don't plan to fix it myself.

if new api is needed, let's do a breaking change and switch to `FlutterBluePlus.systemDevices({Guid? uuid}) `

xiongsi1101, 2024-09-20:
@anhvu2503 您好，你这个问题解决了吗？请求怎么解决的，能贴一下你的解决方案吗，谢谢

chipweinberger, 2024-09-24:
actually, api should take an array

FlutterBluePlus.systemDevices({Guid serviceUuid = []})

anhvu2503, 2024-09-24:
> @anhvu2503 您好，你这个问题解决了吗？请求怎么解决的，能贴一下你的解决方案吗，谢谢

Currently, I am busy with other tasks, so I haven’t had time to address the fixes yet.

chipweinberger, 2024-09-24:
> The same issue. Is there any way to solve it currently?

yes, fork FBP and make the edits I suggested above.

juggCan, 2024-09-26:
> > The same issue. Is there any way to solve it currently?
> 
> yes, fork FBP and make the edits I suggested above.

I've tried it without any problem. Author, will this point be updated in the next version?

chipweinberger, 2024-09-26:
yes please open high quality PR

chipweinberger, 2024-09-27:
pushed this fix, please try it: 2e2305c03110b83a5e3243a98fc87c0048b70508

point your `pubspec.yaml` to master branch

AlexSmirnov9107, 2024-09-27:
not working. tried with 1812 GUID. (HID)

xcjpdx1218, 2024-09-29:
Hello author, I manually updated the version of the plugin to 1.32.13 in the pubspec.yaml file of the project: flutter-blue_plus: ^ 1.32.13. However, when I run the project, I encounter the following error:

**Waiting for another flutter command to release the startup lock...
Flutter assets will be downloaded from https://storage.flutter-io.cn. Make sure you trust this source!
Resolving dependencies...
Because roboshine depends on flutter_blue_plus ^1.32.13 which doesn't match any versions, version solving failed.
You can try the following suggestion to make the pubspec resolve:
* Consider downgrading your constraint on flutter_blue_plus: flutter pub add flutter_blue_plus:^1.32.11
exit code 1**

Can you tell me how to solve this problem?

xcjpdx1218, 2024-09-29:
@chipweinberger Hello author, I manually updated the version of the plugin to 1.32.13 in the pubspec.yaml file of the project: flutter-blue_plus: ^ 1.32.13. However, when I run the project, I encounter the following error:

Waiting for another flutter command to release the startup lock...
Flutter assets will be downloaded from https://storage.flutter-io.cn/. Make sure you trust this source!
Resolving dependencies...
Because roboshine depends on flutter_blue_plus ^1.32.13 which doesn't match any versions, version solving failed.
You can try the following suggestion to make the pubspec resolve:

Consider downgrading your constraint on flutter_blue_plus: flutter pub add flutter_blue_plus:^1.32.11
exit code 1

Can you tell me how to solve this problem?

chipweinberger, 2024-10-04:
fixed in 1.33.0

closed 2024-10-04 by chipweinberger