chipweinberger/flutter_blue_plus #997: [Help]: Admissible throughput: delayed "Keep Alive" Messages Over Bluetooth Characteristic

lorenzosfarra, 2024-09-20:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.12

### Flutter Version

3.24.1

### What OS?

Android, iOS

### OS Version

*

### Bluetooth Module

Nordic

### What is your problem?

Hi,

I'm trying to send a "keep alive" message every 120ms (that's something I need to do according to persons in charge of the device/firmware) through a Bluetooth characteristic, but the messages don't seem to be dequeued at the same pace they're written in the code. There is a noticeable delay in transmission.

So my questions are:

- Is there a hard limit on how fast Bluetooth messages can be sent (of course is also hardware-dependent)?
- Is sending messages at 120ms intervals a good practice? If it's not, I could try to ask for a fix in the firmware

I've tried both with a classic:

```
Timer.periodic(const Duration(milliseconds: 120), (timer) async {
   await  bluetoothCharacteristicThrottle.write([keepAliveMessage]);
}
```
I've also tried a while loop with a `sleep` call.

The characteristic does not support the "withoutResponse" parameter, I don't know if I can try something else.

Thanks for any hint you can provide.

Have a nice day,
  Lorenzo

### Logs

```shell
open question, the write operation logs are OK.
```

chipweinberger, 2024-09-20:
this is a bad idea. 

FBP can encounter delays at any time for many reasons.

120ms is too short.

I would recommend 1 second minimum, with 2 seconds being better.

chipweinberger, 2024-09-20:
btw, BLE already has KeepAlive messages built in.

You can just listen for a Disconnect event. 

> Bluetooth Low Energy (BLE) does maintain a connection between devices using connection intervals and supervision timeouts, very similar to Keep-Alive.
> 
> - **Connection Interval**: BLE devices communicate at regular intervals (connection events). This interval is negotiated when a connection is established and can range from 7.5 ms to 4 seconds. During each connection event, the central and peripheral devices can exchange data packets. If no data is sent, the devices still acknowledge the event to maintain the connection.
> 
> - **Supervision Timeout**: This is the maximum time that can pass without receiving a valid packet from the connected peer. If the supervision timeout is reached without communication, the connection is considered lost. The supervision timeout is typically set to a value between 100 ms and 32 seconds.
> 
> These connection intervals and supervision timeouts serve as a form of implicit "keep-alive" mechanism, ensuring that devices remain connected or detect a connection loss.

lorenzosfarra, 2024-09-21:
Thank you very much, that's a great source of information.
lorenzosfarra.com - @lrnzsfr

lorenzosfarra, 2024-09-21:
Thanks for your info, I can close the ticket