chipweinberger/flutter_blue_plus #999: [Help]: example app crashes on macOS

WBP20, 2024-09-24:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.13

### Flutter Version

3.24.3

### What OS?

macOS

### OS Version

macOS 13.6.5

### Bluetooth Module

macos native Bluetooth

### What is your problem?

When I try to run the example app (flutter run). It crashes.
It told me the app's Info.plist must contain an NSBluetoothAlwaysUsageDescription key with a string value explaining to the user how the app uses this data.

I checked in flutter_blue_plus/example/macos/Runner/info.plist

`	<key>NSBluetoothAlwaysUsageDescription</key>
	<string>We need BT access because it's a BT App.</string>`

the key is there.

I ran the example app several time successfuly in the past. 

### Logs

```shell
Building macOS application...
✓ Built build/macos/Build/Products/Debug/macos.app
flutter: [FBP] <setLogLevel> args: 5
2024-09-24 19:40:46.245 macos[2137:15448] [FBP-iOS] handleMethodCall: setLogLevel
Syncing files to device macOS...                                    45ms

Flutter run key commands.
r Hot reload. 🔥🔥🔥
R Hot restart.
h List all available interactive commands.
d Detach (terminate "flutter run" but leave application running).
c Clear the screen
q Quit (terminate the application on the device).

A Dart VM Service on macOS is available at: http://127.0.0.1:49363/QcRKj0vesmc=/
flutter: [FBP] <setLogLevel> result: true
flutter: [FBP] <getAdapterState> args: null
2024-09-24 19:40:46.533 macos[2137:15448] [FBP-iOS] handleMethodCall: flutterRestart
2024-09-24 19:40:46.533 macos[2137:15448] [FBP-iOS] initializing CBCentralManager
2024-09-24 19:40:46.533 macos[2137:15448] [FBP-iOS] showPowerAlert: yes
2024-09-24 19:40:46.533 macos[2137:15448] [FBP-iOS] restoreState: no
2024-09-24 19:40:46.533 macos[2137:15448] [FBP-iOS] initializing checkForMtuChangesTimer
2024-09-24 19:40:46.533 macos[2137:15448] [FBP-iOS] disconnectAllDevices(flutterRestart)
2024-09-24 19:40:46.533 macos[2137:15448] [FBP-iOS] connectedPeripherals: 0
2024-09-24 19:40:46.533 macos[2137:15448] [FBP-iOS] handleMethodCall: getAdapterState
The Flutter DevTools debugger and profiler on macOS is available at:
http://127.0.0.1:9100?uri=http://127.0.0.1:49363/QcRKj0vesmc=/
flutter: [FBP] <getAdapterState> result: {adapter_state: 0}
Lost connection to device.

------------------------------------------------------------------

Crashed Thread:        3  Dispatch queue: com.apple.root.default-qos

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace TCC, Code 0 
This app has crashed because it attempted to access privacy-sensitive data without a usage description. The app's Info.plist must contain an NSBluetoothAlwaysUsageDescription key with a string value explaining to the user how the app uses this data.
```

chipweinberger, 2024-09-24:
I've never had this problem. maybe a bug in macOS.

chipweinberger, 2024-09-27:
dont think we can fix this

closed 2024-09-27 by chipweinberger