chipweinberger/flutter_blue_plus #1002: Android: SCAN_FAILED_ALREADY_STARTED after turning Bluetooth Off then On #849

tucici, 2024-09-26:
### FlutterBluePlus Version

1.32.13

### Flutter Version

3.24.1 

### What OS?

Android

### OS Version

Android. 13

### Bluetooth Module

not clear

### What is your feature request?

To reproduce the issue, do the following:
turn on bluetooth---> scan -- > quickly turn off bluetooth -- > quickly turn on bluetooth -- > scan again,
and it will  appear.
"SCAN_FAILED_ALREADY_STARTED"   

[Android: SCAN_FAILED_ALREADY_STARTED after turning Bluetooth Off then On #849](https://github.com/chipweinberger/flutter_blue_plus/issues/849)

### Logs

```shell
Launching lib/main.dart on M2103K19C in debug mode...
You are applying Flutter's app_plugin_loader Gradle plugin imperatively using the apply script method, which is deprecated and will be removed in a future release. Migrate to applying Gradle plugins with the declarative plugins block: https://flutter.dev/to/flutter-gradle-plugin-apply

You are applying Flutter's main Gradle plugin imperatively using the apply script method, which is deprecated and will be removed in a future release. Migrate to applying Gradle plugins with the declarative plugins block: https://flutter.dev/to/flutter-gradle-plugin-apply

Warning: unable to detect project KGP version. Skipping version checking.
This may be because you have applied KGP after the Flutter Gradle Plugin.
✓ Built build/app/outputs/flutter-apk/app-debug.apk
Connecting to VM Service at ws://127.0.0.1:53453/PRtMcjS9JL0=/ws
Connected to the VM Service.
I/flutter (27473): [FBP] <setLogLevel> args: 5
D/[FBP-Android](27473): [FBP] onMethodCall: setLogLevel
D/[FBP-Android](27473): [FBP] initializing BluetoothAdapter
D/hw-ProcessState(27473): Binder ioctl to enable oneway spam detection failed: Invalid argument
I/flutter (27473): [FBP] <setLogLevel> result: true
I/flutter (27473): [FBP] <getAdapterState> args: null
D/[FBP-Android](27473): [FBP] onMethodCall: flutterRestart
D/[FBP-Android](27473): [FBP] disconnectAllDevices(flutterRestart)
D/[FBP-Android](27473): [FBP] connectedPeripherals: 0
D/[FBP-Android](27473): [FBP] onMethodCall: getAdapterState
E/gralloc4(27473): Empty SMPTE 2094-40 data
D/BLASTBufferQueue(27473): [SurfaceView[com.lib.flutter_blue_plus_example/com.lib.flutter_blue_plus_example.MainActivity]#1](f:0,a:1) acquireNextBufferLocked size=1080x2356 mFrameNumber=1 applyTransaction=true mTimestamp=625261675763267(auto) mPendingTransactions.size=0 graphicBufferId=117995636523008 transform=0
D/Drawable(27473): getShouldDeclineWhiteColor false
D/Drawable(27473): getShouldDeclineWhiteColor false
D/Drawable(27473): getShouldDeclineWhiteColor false
I/ue_plus_example(27473): ProcessProfilingInfo new_methods=1481 is saved saved_to_disk=1 resolve_classes_delay=8000
D/ue_plus_example(27473): MiuiProcessManagerServiceStub setSchedFifo
I/MiuiProcessManagerImpl(27473): setSchedFifo pid:27473, mode:3
E/gralloc4(27473): Empty SMPTE 2094-40 data
D/BLASTBufferQueue(27473): [VRI[MainActivity]#0](f:0,a:1) acquireNextBufferLocked size=1080x2400 mFrameNumber=1 applyTransaction=true mTimestamp=625261714127882(auto) mPendingTransactions.size=0 graphicBufferId=117995636523009 transform=0
W/Parcel  (27473): Expecting binder but got null!
D/DecorView[](27473): onWindowFocusChanged hasWindowFocus true
I/HandWritingStubImpl(27473): refreshLastKeyboardType: 1
I/HandWritingStubImpl(27473): getCurrentKeyboardType: 1
I/flutter (27473): [FBP] <getAdapterState> result: {adapter_state: 4}
I/HandWritingStubImpl(27473): getCurrentKeyboardType: 1
E/gralloc4(27473): Empty SMPTE 2094-40 data
D/DecorView[](27473): getWindowModeFromSystem  windowmode is 1
[... 406 lines cut ...]
E/gralloc4(27473): Empty SMPTE 2094-40 data
I/ue_plus_example(27473): oneway function results for code 1599295570 on binder at 0xb4000073949a4780 will be dropped but finished with status UNKNOWN_TRANSACTION
I/ue_plus_example(27473): oneway function results for code 1599295570 on binder at 0xb4000073949a4780 will be dropped but finished with status UNKNOWN_TRANSACTION
D/[FBP-Android](27473): [FBP] OnAdapterStateChanged: on
I/flutter (27473): [FBP] [[ OnAdapterStateChanged ]] result: {adapter_state: 4}
E/gralloc4(27473): Empty SMPTE 2094-40 data
E/gralloc4(27473): Empty SMPTE 2094-40 data
D/DecorView[](27473): onWindowFocusChanged hasWindowFocus true
I/HandWritingStubImpl(27473): refreshLastKeyboardType: 1
I/HandWritingStubImpl(27473): getCurrentKeyboardType: 1

I/flutter (27473): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 1, android_legacy: false, android_scan_mode: 2, android_uses_fine_location: false}
D/[FBP-Android](27473): [FBP] onMethodCall: startScan
D/BluetoothAdapter(27473): isLeEnabled(): ON
E/[FBP-Android](27473): [FBP] onScanFailed: SCAN_FAILED_ALREADY_STARTED
I/flutter (27473): [FBP] <startScan> result: true
I/flutter (27473): [FBP] [[ OnScanResponse ]] result: {error_string: SCAN_FAILED_ALREADY_STARTED, advertisements: [], success: 0, error_code: 1}

I/flutter (27473): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 1, android_legacy: false, android_scan_mode: 2, android_uses_fine_location: false}
D/[FBP-Android](27473): [FBP] onMethodCall: startScan
D/BluetoothAdapter(27473): isLeEnabled(): ON
E/[FBP-Android](27473): [FBP] onScanFailed: SCAN_FAILED_ALREADY_STARTED
I/flutter (27473): [FBP] <startScan> result: true
I/flutter (27473): [FBP] [[ OnScanResponse ]] result: {error_string: SCAN_FAILED_ALREADY_STARTED, advertisements: [], success: 0, error_code: 1}
```

tucici, 2024-09-26:
@chipweinberger  Inspiration comes from here [](https://stackoverflow.com/questions/29781393/android-ble-scan-scan-failed-already-started-and-getting-hold-of-a-bluetoothdev?rq=4)

tucici, 2024-09-26:
@chipweinberger Inspiration comes from here :  [Android BLE scan, SCAN_FAILED_ALREADY_STARTED and getting hold of a BluetoothDevice instance](https://stackoverflow.com/questions/29781393/android-ble-scan-scan-failed-already-started-and-getting-hold-of-a-bluetoothdev)

chipweinberger, 2024-09-26:
this was our previous fix

https://github.com/chipweinberger/flutter_blue_plus/commit/99740a73c221c9d79837fe2be804962583e3dcfd

chipweinberger, 2024-09-26:
can you please debug why https://github.com/chipweinberger/flutter_blue_plus/commit/99740a73c221c9d79837fe2be804962583e3dcfd does not stop the scan?

chipweinberger, 2024-09-26:
> turn on bluetooth---> scan -- > quickly turn off bluetooth -- > quickly turn on bluetooth -- > scan again,

does the problem only happen when you do it "quickly"?

tucici, 2024-09-26:
> > turn on bluetooth---> scan -- > quickly turn off bluetooth -- > quickly turn on bluetooth -- > scan again,
> 
> does the problem only happen when you do it "quickly"?

No, in fact, before the scan duration is complete, turn off Bluetooth, turn it on again, and do the next scan, this error will definitely occur.
 
For example, with your demo, the scan duration is 15 seconds, turn off Bluetooth before 15 seconds, turn it on again,  and then perform a scan,and it will appear

chipweinberger, 2024-11-08:
I just tried your steps, in the Example app

1. press scan
2. turn off bluetooth before scan finished
3. turn on bluetooth
4. press scan

I did not have problems

closed 2024-11-08 by chipweinberger