chipweinberger/flutter_blue_plus #1007: [Help]: Characteristic errors are not propagated from Android to dart

jefflongo, 2024-10-04:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.13

### Flutter Version

3.24.3

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

ESP32C3

### What is your problem?

To test if pairing is successful, I read an encrypted characteristic. If the bond was not successful (i.e. user dismissed PIN or entered wrong PIN), reading the characteristic will result in an authentication error. I expect to receive an exception on the Flutter side when this happens. Instead, the list `[0]` is returned.

The cause of this is this line: https://github.com/chipweinberger/flutter_blue_plus/blob/master/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java#L2294

Any error in the characteristic read (and notifications/indications, for that matter) on the Android side is logged and thrown away and treated as success in the dart side. If an encrypted characteristic is read without a successful bond, the status code is 137 (`GATT_AUTH_FAIL`). This status code is currently treated as `UNKNOWN_GATT_ERROR`. The real status code should be sent to `onCharacteristicReceived` instead of `BluetoothGatt.GATT_SUCCESS`, so the dart code can properly throw an exception.

### Logs

```shell
E/[FBP-Android](31089): [FBP] onCharacteristicRead:
E/[FBP-Android](31089): [FBP]   chr: 827ebcc0-0d9a-521f-a336-39072e84edce
E/[FBP-Android](31089): [FBP]   status: UNKNOWN_GATT_ERROR (137) (137)
I/flutter (31089): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: e78333a3-57fb-5891-8ac2-fd83c79dea82, success: 1, remote_id: 10:91:A8:59:0F:9A, error_code: 0, characteristic_uuid: 827ebcc0-0d9a-521f-a336-39072e84edce, value: 00}
```

chipweinberger, 2024-10-04:
thanks. you're right.

chipweinberger, 2024-10-04:
fixed 1.33.0

closed 2024-10-04 by chipweinberger

jefflongo, 2024-10-04:
Ah, you beat me to the PR. I posted mine anyway which adds the fix for notifications/indications and adds some more GATT error strings.