chipweinberger/flutter_blue_plus #1009: [BUG]: Characteristic read errors in IOS

mu-fazil-vk, 2024-10-05:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.33.0

### Flutter Version

3.22.2

### What OS?

iOS

### OS Version

iOS 17.5.1

### Bluetooth Module

ESP32C3

### What is your problem?

When I try to read the characteristic using `.read()`, getting:
`FlutterBluePlusException | readCharacteristic | fbp-code: 6 | Device is disconnected` or `time out` as exception. In Android, its working fine. Please fix this issue.

### Logs

```shell
didUpdateValueForCharacteristic:
chr: 6e400005-b5a3-f393-e0a9-e50e24dcca9e
 handleMethodCall: readCharacteristic
 didDisconnectPeripheral:
```

chipweinberger, 2024-10-07:
you need to connect first

closed 2024-10-07 by chipweinberger

mu-fazil-vk, 2024-10-08:
> you need to check if connected
> 
> your device may disconnect at any time, for many reasons

There is already a connection check implemented above the read() function, and this read issue only happening on iOS; on Android, it's working fine.

chipweinberger, 2024-10-08:
need logs

chipweinberger, 2024-10-08:
eother way its probably not an issue in FBP

more likely a problem with iOS or your ble device.

mu-fazil-vk, 2024-10-08:
But the Issue is occuring frequently not always. Yesterday I was able to connect multiple times and the read() method worked perfectly. But now it’s again throwing error when using the read() method. Above the read() method, I am again checking whether the device is disconnected or not and connecting again if disconnected.

```
Future<String> sendWifiData(String wifiName, String wifiPassword,
      String userId, String deviceId) async {
    print(‘=======READ CALLING=======‘);
    if (device?.isConnected == false) {
      await _connectDevice();
    }
    List<int> rawDeviceIdCandidate = [];
    try {
      if (deviceIdCharacteristic.properties.read) {
        rawDeviceIdCandidate = await deviceIdCharacteristic.read(timeout: 300);
      }
    } catch (e) {
      log(device.toString());
      log(e.toString());
    }

    (‘========READ CALLING FINISHED========‘);
    //String deviceId = Int(rawDeviceId);
    String machineId = “”;
    if (rawDeviceIdCandidate.isEmpty) {
      machineId = deviceId;
    } else {
      machineId = String.fromCharCodes(rawDeviceIdCandidate);
    }

    Logger.debug(
        “Writing $userId, SSID $wifiName, wifiPassowrd: $wifiPassword. deviceId $machineId”);
    await userIdCharacteristic.write(userId.codeUnits, allowLongWrite: true);
    await deviceIdCharacteristic.write(machineId.codeUnits,
        allowLongWrite: true);
    await ssidCharacteristic.write(wifiName.codeUnits, allowLongWrite: true);
    await passwordCharacteristic.write(wifiPassword.codeUnits,
        allowLongWrite: true);
    Logger.info(“Machine uuid: $machineId”);

    return machineId;
  }
```

And after this condition the connected flag is true. But when I call  deviceIdCharacteristic.read(timeout: 300) method, I am getting this exception after a few seconds and the connected flag is false.

FlutterBluePlusException | readCharacteristic | fbp-code: 6 | Device is disconnected

Logs
```
[FBP-iOS] handleMethodCall: discoverServices
[FBP-iOS] didDiscoverServices:
[FBP-iOS]  svc: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
[FBP-iOS]  svc: 88dcd081-e4bd-422b-8be4-44fd4d46d030
[FBP-iOS] didDiscoverCharacteristicsForService:
[FBP-iOS]  svc: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
[FBP-iOS]   chr: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
[FBP-iOS]   chr: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
[FBP-iOS]   chr: 6e400004-b5a3-f393-e0a9-e50e24dcca9e
[FBP-iOS]   chr: 6e400005-b5a3-f393-e0a9-e50e24dcca9e
[FBP-iOS]   chr: 6e400006-b5a3-f393-e0a9-e50e24dcca9e
[FBP-iOS] didDiscoverCharacteristicsForService:
[FBP-iOS]  svc: 88dcd081-e4bd-422b-8be4-44fd4d46d030
[FBP-iOS]   chr: 88dcd082-e4bd-422b-8be4-44fd4d46d030
[FBP-iOS]   chr: 88dcd083-e4bd-422b-8be4-44fd4d46d030
[FBP-iOS]   chr: 88dcd084-e4bd-422b-8be4-44fd4d46d030
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]  chr: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]  chr: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]  chr: 6e400004-b5a3-f393-e0a9-e50e24dcca9e
[FBP-iOS]   desc: 2902
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]  chr: 6e400005-b5a3-f393-e0a9-e50e24dcca9e
[FBP-iOS]   desc: 2902
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]  chr: 6e400006-b5a3-f393-e0a9-e50e24dcca9e
[FBP-iOS]   desc: 2902
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]  chr: 88dcd082-e4bd-422b-8be4-44fd4d46d030
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]  chr: 88dcd083-e4bd-422b-8be4-44fd4d46d030
[FBP-iOS] didDiscoverDescriptorsForCharacteristic:
[FBP-iOS]  chr: 88dcd084-e4bd-422b-8be4-44fd4d46d030
flutter: ========ANALYZED=======
[FBP-iOS] handleMethodCall: setNotifyValue
[FBP-iOS] didUpdateNotificationStateForCharacteristic:
[FBP-iOS]  chr: 6e400004-b5a3-f393-e0a9-e50e24dcca9e
[FBP-iOS] handleMethodCall: setNotifyValue
[FBP-iOS] didUpdateNotificationStateForCharacteristic:
[FBP-iOS]  chr: 6e400005-b5a3-f393-e0a9-e50e24dcca9e
[FBP-iOS] handleMethodCall: setNotifyValue
[FBP-iOS] didUpdateNotificationStateForCharacteristic:
[FBP-iOS]  chr: 6e400006-b5a3-f393-e0a9-e50e24dcca9e
[FBP-iOS] didUpdateValueForCharacteristic:
[FBP-iOS]  chr: 6e400005-b5a3-f393-e0a9-e50e24dcca9e
flutter: \^[[32mDevice mqtt state updated DeviceMQTTState.OWNS_CERTIFICATES<…>
flutter: =======READ CALLING=======
[FBP-iOS] handleMethodCall: readCharacteristic
[FBP-iOS] didDisconnectPeripheral:
[FBP-iOS]  error: The connection has timed out unexpectedly.
```

chipweinberger, 2024-10-08:
> error: The connection has timed out unexpectedly.

this is your problem. Look at the README. We talk about this problem.