chipweinberger/flutter_blue_plus #1014: [Bug]: java.lang.NullPointerException: Attempt to invoke interface method 'void com.lib.flutter_blue_plus.FlutterBluePlusPlugin$OperationOnPermission.op(boolean, java.lang.String)' on a null object reference

BreadKey, 2024-10-08:
### FlutterBluePlus Version

1.33.3

### Flutter Version

3.24.3

### What OS?

Android

### OS Version

14

### Bluetooth Module

?

### What is your feature request?

When I accept to allow some permissions (in my case, Notifications), including Bluetooth by permission_handler, the app crashes.

### Logs

```shell
E/AndroidRuntime( 7608): java.lang.RuntimeException: Failure delivering result ResultInfo{who=@android:requestPermissions:, request=24, result=-1, data=Intent { act=android.content.pm.action.REQUEST_PERMISSIONS (has extras) }} to activity {io.openb.linkview/io.openb.linkview.MainActivity}: java.lang.NullPointerException: Attempt to invoke interface method 'void com.lib.flutter_blue_plus.FlutterBluePlusPlugin$OperationOnPermission.op(boolean, java.lang.String)' on a null object reference
E/AndroidRuntime( 7608): 	at android.app.ActivityThread.deliverResults(ActivityThread.java:5527)
E/AndroidRuntime( 7608): 	at android.app.ActivityThread.handleSendResult(ActivityThread.java:5566)
E/AndroidRuntime( 7608): 	at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:67)
E/AndroidRuntime( 7608): 	at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
E/AndroidRuntime( 7608): 	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:139)
E/AndroidRuntime( 7608): 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:96)
E/AndroidRuntime( 7608): 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2443)
E/AndroidRuntime( 7608): 	at android.os.Handler.dispatchMessage(Handler.java:106)
E/AndroidRuntime( 7608): 	at android.os.Looper.loopOnce(Looper.java:205)
E/AndroidRuntime( 7608): 	at android.os.Looper.loop(Looper.java:294)
E/AndroidRuntime( 7608): 	at android.app.ActivityThread.main(ActivityThread.java:8177)
E/AndroidRuntime( 7608): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime( 7608): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
E/AndroidRuntime( 7608): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971)
E/AndroidRuntime( 7608): Caused by: java.lang.NullPointerException: Attempt to invoke interface method 'void com.lib.flutter_blue_plus.FlutterBluePlusPlugin$OperationOnPermission.op(boolean, java.lang.String)' on a null object reference
E/AndroidRuntime( 7608): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.onRequestPermissionsResult(FlutterBluePlusPlugin.java:1537)
E/AndroidRuntime( 7608): 	at io.flutter.embedding.engine.FlutterEngineConnectionRegistry$FlutterEngineActivityPluginBinding.onRequestPermissionsResult(FlutterEngineConnectionRegistry.java:746)
E/AndroidRuntime( 7608): 	at io.flutter.embedding.engine.FlutterEngineConnectionRegistry.onRequestPermissionsResult(FlutterEngineConnectionRegistry.java:405)
E/AndroidRuntime( 7608): 	at io.flutter.embedding.android.FlutterActivityAndFragmentDelegate.onRequestPermissionsResult(FlutterActivityAndFragmentDelegate.java:912)
E/AndroidRuntime( 7608): 	at io.flutter.embedding.android.FlutterActivity.onRequestPermissionsResult(FlutterActivity.java:967)
E/AndroidRuntime( 7608): 	at android.app.Activity.dispatchRequestPermissionsResult(Activity.java:9089)
E/AndroidRuntime( 7608): 	at android.app.Activity.dispatchActivityResult(Activity.java:8947)
E/AndroidRuntime( 7608): 	at android.app.ActivityThread.deliverResults(ActivityThread.java:5520)
E/AndroidRuntime( 7608): 	... 13 more
I/Process ( 7608): Sending signal. PID: 7608 SIG: 9
Lost connection to device.

Exited.
```

chipweinberger, 2024-10-08:
please open PR

chipweinberger, 2024-10-08:
try 1.33.4

closed 2024-10-08 by chipweinberger

shimulxx, 2024-10-08:
Yes, same issue appears. Downgrade to 1.32.12 seems to works well.

chipweinberger, 2024-10-08:
@shimulxx try 1.33.4

shimulxx, 2024-10-08:
okk