chipweinberger/flutter_blue_plus #1015: [Help]: IOS: FlutterBluePlus.startScan() automatically opens AppSettings

minhdang2803, 2024-10-08:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.33.3

### Flutter Version

3.22.3

### What OS?

iOS

### OS Version

14.6.1

### Bluetooth Module

No 

### What is your problem?

The AppSettings is automatically open when startScan(), How can i prevent it from opening?

### Logs

```shell
No Logs
```

chipweinberger, 2024-10-08:
please include screen recording

minhdang2803, 2024-10-08:
https://drive.google.com/file/d/14oRYyo_k_-HZFcHy9x0wEZkIIAtNHJwo/view?usp=share_link

here is screen record, please check it for me

minhdang2803, 2024-10-08:
> https://drive.google.com/file/d/14oRYyo_k_-HZFcHy9x0wEZkIIAtNHJwo/view?usp=share_link
> 
> here is screen record, please check it for me

When i tap on the "Bluetooth configuration" to open ble scanning page -> it invokes FlutterBlePlus.startScan(), then it automatically move to the AppSettings

chipweinberger, 2024-10-08:
read the README

maybe your permissions are wrong

closed 2024-10-08 by minhdang2803