chipweinberger/flutter_blue_plus #1020: [Help]: device is not disconnecting in iOS, disconnect method not works. But in Android it is OK.

shimulxx, 2024-10-16:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.33.4

### Flutter Version

3.24.3

### What OS?

iOS

### OS Version

16.7

### Bluetooth Module

Nano Pi, Neo Plus 2 Ubuntu

### What is your problem?

in iOS, once connect it never disconnects. If i called disconnect it changes it status to disconnect but actually it is connected. No other device is found at this moment. One way is to turn off bluetooth in iOS devices manually to disconnect. In Android all is ok, disconnects works.

### Logs

```shell
Already described! Please solve the issue !!!! Thank you very much.....
```

chipweinberger, 2024-10-16:
disconnect works fine for me on iOS

please give logs

shimulxx, 2024-10-17:
Test with example app.

1, After disconnect from iOS, try to connect with Android:

FAILED LOG:

I/ViewRootImpl( 7276): ViewRoot's Touch Event : ACTION_DOWN
I/ViewRootImpl( 7276): ViewRoot's Touch Event : ACTION_UP
I/flutter ( 7276): [FBP] <connect> args: {remote_id: E2:A1:97:24:0A:70, auto_connect: 0}
D/[FBP-Android]( 7276): [FBP] onMethodCall: connect
D/BluetoothGatt( 7276): connect() - device: E2:A1:97:24:0A:70, auto: false, eattSupport: false
D/BluetoothGatt( 7276): registerApp()
D/BluetoothGatt( 7276): registerApp() - UUID=e710a3e7-3fc4-4a62-ae5b-b3dfab3dce09
D/BluetoothGatt( 7276): onClientRegistered() - status=0 clientIf=7
I/flutter ( 7276): [FBP] <connect> result: true
I/ViewRootImpl( 7276): ViewRoot's Touch Event : ACTION_DOWN
I/ViewRootImpl( 7276): ViewRoot's Touch Event : ACTION_UP
D/BluetoothGatt( 7276): onClientConnectionState() - status=255 clientIf=7 device=E2:A1:97:24:0A:70
D/[FBP-Android]( 7276): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android]( 7276): [FBP]   status: UNKNOWN_HCI_ERROR (255)
D/BluetoothGatt( 7276): close()
D/BluetoothGatt( 7276): unregisterApp() - mClientIf=7
I/flutter ( 7276): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 255, disconnect_reason_string: UNKNOWN_HCI_ERROR (255), remote_id: E2:A1:97:24:0A:70, connection_state: 0}

2. After turn off iPhone Bluetooth of iOS then it connect successfully.

SUCCESS LOG:

I/ViewRootImpl( 7276): ViewRoot's Touch Event : ACTION_DOWN
I/ViewRootImpl( 7276): ViewRoot's Touch Event : ACTION_UP
I/flutter ( 7276): [FBP] <connect> args: {remote_id: E2:A1:97:24:0A:70, auto_connect: 0}
D/[FBP-Android]( 7276): [FBP] onMethodCall: connect
D/BluetoothGatt( 7276): connect() - device: E2:A1:97:24:0A:70, auto: false, eattSupport: false
D/BluetoothGatt( 7276): registerApp()
D/BluetoothGatt( 7276): registerApp() - UUID=c42497c8-3d03-4f03-b018-2b7fe4884898
I/flutter ( 7276): [FBP] <connect> result: true
D/BluetoothGatt( 7276): onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt( 7276): onClientConnectionState() - status=0 clientIf=7 device=E2:A1:97:24:0A:70
D/[FBP-Android]( 7276): [FBP] onConnectionStateChange:connected
D/[FBP-Android]( 7276): [FBP]   status: SUCCESS
I/flutter ( 7276): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: E2:A1:97:24:0A:70, connection_state: 1}
I/flutter ( 7276): [FBP] <readRssi> args: E2:A1:97:24:0A:70
D/[FBP-Android]( 7276): [FBP] onMethodCall: readRssi
D/BluetoothGatt( 7276): readRssi() - device: E2:A1:97:24:0A:70
I/flutter ( 7276): [FBP] <readRssi> result: true
D/[FBP-Android]( 7276): [FBP] onReadRemoteRssi:
D/[FBP-Android]( 7276): [FBP]   rssi: -62
D/[FBP-Android]( 7276): [FBP]   status: GATT_SUCCESS (0)
I/flutter ( 7276): [FBP] [[ OnReadRssi ]] result: {error_string: GATT_SUCCESS, rssi: -62, success: 1, remote_id: E2:A1:97:24:0A:70, error_code: 0}
D/BluetoothGatt( 7276): onConnectionUpdated() - Device=E2:A1:97:24:0A:70 interval=36 latency=0 timeout=500 status=42
D/BluetoothGatt( 7276): onConnectionUpdated() - Device=E2:A1:97:24:0A:70 interval=36 latency=0 timeout=500 status=0

PLEASE HELP...

chipweinberger, 2024-10-17:
> UNKNOWN_HCI_ERROR (255)

There is no easy solution. This is Android reporting an unknown error. Not very helpful. 

Try looking at your peripheral's firmware logs.

littleantfly, 2024-10-23:
我也遇到了同样的问题。我和一台设备通过低功耗蓝牙连接。当我主动调用disconnect(queue: false) 时，手机系统蓝牙并不会断开与设备的连接，设备端也无法收到蓝牙断开信号。
flutter_blue_plus: ^1.33.4
IOS 18.0.1

littleantfly, 2024-10-23:
I also encountered the same problem. I was connected to a device via low-power Bluetooth. When I actively called disconnect(queue: false), the mobile phone system Bluetooth did not disconnect from the device, and the device could not receive the Bluetooth disconnect signal.
flutter_blue_plus: ^1.33.4
IOS 18.0.1

shimulxx, 2024-10-23:
chipweinberger this is the exact issue. Can u solve this? This is not android problem. This problem belongs to iOS. Some BLE devices iOS disconnect works correctly but some device still connected after calling disconnect function. Can u solve it ?

chipweinberger, 2024-10-23:
@shimulxx , please solve it and open a PR. thanks.