chipweinberger/flutter_blue_plus #1023: [Help]: CONNECTION_TERMINATED_BY_LOCAL_HOST getting error in android connection

JaydipGateway, 2024-10-17:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.33.4

### Flutter Version

3.24.3

### What OS?

Android

### OS Version

14

### Bluetooth Module

Nordic Nrf5

### What is your problem?

I am getting the below error while connection with the device.

[[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 22, disconnect_reason_string: CONNECTION_TERMINATED_BY_LOCAL_HOST, remote_id: D5:47:B8:AF:98:DB, connection_state: 0}

### Logs

```shell
D/InputMethodManagerUtils( 4665): startInputInner - Id : 0
I/InputMethodManager( 4665): startInputInner - IInputMethodManagerGlobalInvoker.startInputOrWindowGainedFocus
D/[FBP-Android]( 4665): [FBP] OnBondStateChanged: bond-none prev: bonding
I/flutter ( 4665): [FBP] [[ OnBondStateChanged ]] result: {bond_state: 0, remote_id: D5:47:B8:AF:98:DB, prev_state: 1}
D/BluetoothGatt( 4665): onConnectionUpdated() - Device=XX:XX:XX:XX:98:DB interval=24 latency=24 timeout=3000 status=0
D/BluetoothGatt( 4665): onClientConnectionState() - status=22 clientIf=6 device=XX:XX:XX:XX:98:DB
D/[FBP-Android]( 4665): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android]( 4665): [FBP]   status: CONNECTION_TERMINATED_BY_LOCAL_HOST
D/BluetoothGatt( 4665): close()
D/BluetoothGatt( 4665): unregisterApp() - mClientIf=6
I/flutter ( 4665): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 22, disconnect_reason_string: CONNECTION_TERMINATED_BY_LOCAL_HOST, remote_id: D5:47:B8:AF:98:DB, connection_state: 0}
```

chipweinberger, 2024-10-17:
> In BLE (Bluetooth Low Energy), the error code CONNECTION_TERMINATED_BY_LOCAL_HOST refers to a situation where the connection between two devices is intentionally terminated by the device that initiated the disconnection. This can happen for a variety of reasons, such as:
> 
> * The local device (the host) decided to disconnect due to an internal event (e.g., resource management, user request, or timeout).
> * The application running on the local device explicitly requested the disconnection.
> * The local device detected an issue with the connection and opted to terminate it.
> * 
> Essentially, it means that the device you are controlling or using as the host was the one that terminated the connection, not the remote device.

closed 2024-10-17 by chipweinberger

Rupsero, 2025-03-05:
Have you found any solution to your issue?
I´m having the same issues with 2 phones running Android 14 and the example flutter_blue_plus app