chipweinberger/flutter_blue_plus #1024: [Help]: Why does clearGattCache need device to be connected?

shubhamsinghmutualmobile, 2024-10-18:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.33.5

### Flutter Version

3.24.3

### What OS?

Android

### OS Version

Android 10 (Q)

### Bluetooth Module

Nordic nRF52

### What is your problem?

I am facing problems in `connect`ing to my BLE device. That's why I through trying `clearGattCache` as a method to recover from this issue, but seems like the function does work until the device is connected, which kinda defeats the purpose of having this method in the first place.

Are there any complications of removing the `connected` check from the logic that I am missing at the moment?

## Concerned Library Code

[image]

### Logs

```shell
D/[FBP-Android](24196): [FBP] onMethodCall: connect
D/BluetoothGatt(24196): connect() - device: DF:57:39:36:BB:90, auto: true
D/BluetoothGatt(24196): registerApp()
D/BluetoothGatt(24196): registerApp() - UUID=9cad8822-9012-4af5-9ddd-6ee82f208ee2
D/BluetoothGatt(24196): onClientRegistered() - status=0 clientIf=11
I/flutter (24196): FlutterBluePlusException | discoverServices | fbp-code: 6 | device is not connected
I/flutter (24196): null ->> 
I/flutter (24196): #0      BluetoothDevice.discoverServices (package:flutter_blue_plus/src/bluetooth_device.dart:250:7)
I/flutter (24196): #1      BluetoothDeviceExtension.connectAndUnlock (package:core/ble_utils.dart:55:34)
I/flutter (24196): <asynchronous suspension>
I/flutter (24196): #2      _DeviceFoundScreenState._handleMugConnection (package:drinkware/features/2_device_pairing/view/device_found_screen.dart:180:14)
```

chipweinberger, 2024-10-18:
I don't remember.

but i think the function will fail if not connected

but remove the check and try it

shubhamsinghmutualmobile, 2024-10-21:
> I don't remember.
> 
> but i think the function will fail if not connected
> 
> remove the check and try it

You're right, seems like the function `gatt.getClass().getMethod("refresh")` can only be called on a connected device.

I'm closing this ticket.
Thanks for your time and help.

closed 2024-10-21 by shubhamsinghmutualmobile