chipweinberger/flutter_blue_plus #1025: [Help]: PlatformException(iosException, Client Characteristic Configuration descriptors must be configured using setNotifyValue:forCharacteristic:, in flutter_blue_plus

EINS-SD-MOHAMMED, 2024-10-19:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.33.5

### Flutter Version

3.24.3

### What OS?

iOS

### OS Version

18.0.1

### Bluetooth Module

Nedap Nvite Reader

### What is your problem?

I am trying to connect to the Nedap Nvite reader using the Flutter Blue Plus package in my mobile app. **The connection process works correctly on Android, allowing me to communicate with the device and enable notifications without issues.**

However, on iOS, after successfully connecting to the Nedap Nvite reader and retrieving the characteristic properties, I encounter an error when attempting to enable notifications. Specifically, after calling setNotifyValue(true), when I try to write to the descriptor, I receive the following error:

**PlatformException(iosException, Client Characteristic Configuration descriptors must be configured using setNotifyValue:forCharacteristic:, in flutter_blue_plus)**

This error indicates that the notification configuration is not being handled correctly on iOS, while the same steps work fine on Android.

I have also gone through the closed issues related to this problem but have not been able to resolve the issue.

For a better understanding of the issue, I will be sharing the logs from both Android and iOS. Any guidance on resolving this iOS-specific issue would be greatly appreciated!

### Logs

```shell
iOS Logs
flutter: 080000000000000123
flutter: 4b76a404f0cdd23c93ac540cc89148e7
[FBP-iOS] handleMethodCall: startScan
flutter: BluetoothDevice{remoteId: 3FDD4B45-0565-AF43-DE53-6BA68C1203E8, platformName: EINS, services: null}
flutter: EINS found! rssi: -78
[FBP-iOS] handleMethodCall: stopScan
[FBP-iOS] handleMethodCall: connect
[FBP-iOS] didConnectPeripheral
[FBP-iOS] handleMethodCall: discoverServices
[FBP-iOS] didDiscoverServices:
[FBP-iOS]   svc: 180a
[FBP-iOS]   svc: 87b1de8d-e7cb-4ea8-a8e4-290209522c83
[FBP-iOS] didDiscoverCharacteristicsForService:
[FBP-iOS]   svc: 180a
[FBP-iOS]     chr: 2a25
[FBP-iOS]     chr: 2a27
[FBP-iOS]     chr: 2a26
[FBP-iOS]     chr: 2a28
[FBP-iOS]     chr: 2a29
[FBP-iOS]     chr: 2a24
[FBP-iOS] didDiscoverCharacteristicsForService:
[FBP-iOS]   svc: 87b1de8d-e7cb-4ea8-a8e4-290209522c83
[FBP-iOS]     chr: e68a5c09-aef8-4447-8f10-f3339898dee9
[FBP-iOS]     chr: 54080bd6-d573-11e5-ab30-625662870761
[... 94 lines cut ...]
D/[FBP-Android](15805): [FBP] onMethodCall: writeCharacteristic
D/[FBP-Android](15805): [FBP] onCharacteristicChanged:
D/[FBP-Android](15805): [FBP]   chr: 540810c2-d573-11e5-ab30-625662870761
D/[FBP-Android](15805): [FBP] onCharacteristicWrite:
D/[FBP-Android](15805): [FBP]   chr: e68a5c09-aef8-4447-8f10-f3339898dee9
D/[FBP-Android](15805): [FBP]   status: GATT_SUCCESS (0)
I/flutter (15805): inside
I/flutter (15805): 4b76a404f0cdd23c93ac540cc89148e7
I/flutter (15805): [87, 115, 207, 107, 71, 99, 191, 123]
I/flutter (15805): RNDA: e5250dde15c6d92e
I/flutter (15805): RNDB*: 73cf6b4763bf7b57
I/flutter (15805): RNDA + RNDB*: e5250dde15c6d92e73cf6b4763bf7b57
I/flutter (15805): 4b76a404f0cdd23c93ac540cc89148e7
I/flutter (15805): 95c294e5e565c949fcd55abf9f03976a
I/flutter (15805): Encrypted message: 95c294e5e565c949fcd55abf9f03976a
I/flutter (15805): 080000000000000123 is of chart
D/[FBP-Android](15805): [FBP] onMethodCall: writeCharacteristic
D/[FBP-Android](15805): [FBP] onCharacteristicChanged:
D/[FBP-Android](15805): [FBP]   chr: 540810c2-d573-11e5-ab30-625662870761
D/[FBP-Android](15805): [FBP] onCharacteristicWrite:
D/[FBP-Android](15805): [FBP]   chr: 54080bd6-d573-11e5-ab30-625662870761
D/[FBP-Android](15805): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](15805): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](15805): [FBP]   status: REMOTE_USER_TERMINATED_CONNECTION
```

chipweinberger, 2024-10-21:
android and iOS have slightly different rules.

ideally we would prevent this on Android as well, so they both behave the same.

but ya, you can't do what you are doing on iOS. Only Android.

closed 2024-10-21 by chipweinberger

EINS-SD-MOHAMMED, 2024-10-21:
It seems that on iOS, we're not receiving any data, even though the BLE device has sent it. We’re getting a List<int> with a length of zero, which suggests that the notifications are not being properly received or processed. We have also used chr.setNotifyValue on the characteristics, but the issue persists.The same is working correctly on Android even after removing the desc.write(..) and just calling the chr.setNotifyValue

hezhipengzipp, 2025-01-23:
how to fix it ?

EINS-SD-MOHAMMED, 2025-01-23:
The issue was resolved by first calling the following line of code to read the characteristic for iOS:

`await notificationCharacteristics.read();`

Once the characteristic was successfully read, I then used the following code to subscribe to the characteristic and listen for incoming notifications:

```
final subscription = notificationCharacteristics.onValueReceived.listen((value) {
    // onValueReceived is updated:
    //   - anytime read() is called
    //   - anytime a notification arrives (if subscribed)
});
```

This sequence of calls allowed me to read the characteristic and properly listen for updates without further issues in iOS.