chipweinberger/flutter_blue_plus #1028: [Help]:  Ble auto connection issue when bluetooth turn on/off

ImanGhasemiArani, 2024-10-21:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.33.5

### Flutter Version

3.5.3

### What OS?

Android

### OS Version

14

### Bluetooth Module

?

### What is your problem?

hello
When the device is connected, turn off the bluetooth, after the device is 10 meters away, then turn on the bluetooth, then the device will not connect anymore, just restart the app to connect.

### Logs

```shell
D/[FBP-Android](14651): [FBP] onCharacteristicChanged:
D/[FBP-Android](14651): [FBP]   chr: fff7
D/BluetoothGatt(14651): onConnectionUpdated() - Device=XX:XX:XX:XX:12:61 interval=21 latency=0 timeout=400 status=0
D/BluetoothAdapter(14651): onStateChanged 13
D/[FBP-Android](14651): [FBP] OnAdapterStateChanged: turningOff
D/[FBP-Android](14651): [FBP] disconnectAllDevices(adapterTurnOff)
D/[FBP-Android](14651): [FBP] OnAdapterStateChanged: turningOff
D/[FBP-Android](14651): [FBP] disconnectAllDevices(adapterTurnOff)
D/[FBP-Android](14651): [FBP] OnAdapterStateChanged: turningOff
D/[FBP-Android](14651): [FBP] disconnectAllDevices(adapterTurnOff)
D/[FBP-Android](14651): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](14651): [FBP]   status: SUCCESS
D/[FBP-Android](14651): [FBP] autoconnect is true. skipping gatt.close()
I/flutter (14651): [W] TIME: 2024-10-21T22:49:17.390608 BLE status: false
I/flutter (14651): ┌───────────────────────────────────────────────────────────────────────────────────────────────────
I/flutter (14651): │ 22:49:17.428
I/flutter (14651): ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
I/flutter (14651): │ 👾 Disconnected from E2:B0:90:00:12:61 isAutoConnect <<true>>
I/flutter (14651): └───────────────────────────────────────────────────────────────────────────────────────────────────
I/BluetoothAdapter(14651): onBluetoothStateChange: up=false
2
I/BluetoothAdapter(14651): Bluetooth is turned off, stop adv
D/[FBP-Android](14651): [FBP] OnAdapterStateChanged: off
D/[FBP-Android](14651): [FBP] disconnectAllDevices(adapterTurnOff)
D/[FBP-Android](14651): [FBP] OnAdapterStateChanged: off
D/[FBP-Android](14651): [FBP] disconnectAllDevices(adapterTurnOff)
D/[FBP-Android](14651): [FBP] OnAdapterStateChanged: off
D/[FBP-Android](14651): [FBP] disconnectAllDevices(adapterTurnOff)
D/BluetoothAdapter(14651): onStateChanged 15
D/BluetoothAdapter(14651): onStateChanged 16
D/BluetoothAdapter(14651): onBluetoothServiceDown
I/BluetoothAdapter(14651): onDisableBLE
D/BluetoothGatt(14651): onDisableBLE()
D/BluetoothGatt(14651): Google BLE is turned off, close(5)
D/BluetoothAdapter(14651): onStateChanged 10
D/BluetoothGatt(14651): close()
2
I/BluetoothAdapter(14651): There are no active google scan apps, stop scan
D/BluetoothLeScanner(14651): stopAllScan standalone boolean value is = false
D/BluetoothLeScanner(14651): Exiting stopping all callback scans
D/BluetoothGatt(14651): unregisterApp() - mClientIf=5
D/BluetoothAdapter(14651): onBluetoothServiceUp: android.bluetooth.IBluetooth$Stub$Proxy@d069375
D/BluetoothAdapter(14651): onStateChanged 14
D/BluetoothAdapter(14651): onStateChanged 15
D/BluetoothAdapter(14651): onStateChanged 11
3
D/[FBP-Android](14651): [FBP] OnAdapterStateChanged: turningOn
I/BluetoothAdapter(14651): onBluetoothStateChange: up=true
D/BluetoothAdapter(14651): onStateChanged 12
3
D/[FBP-Android](14651): [FBP] OnAdapterStateChanged: on
I/flutter (14651): [W] TIME: 2024-10-21T22:49:48.617088 BLE status: true
D/[FBP-Android](14651): [FBP] onMethodCall: connect
D/BluetoothAdapter(14651): getBleEnabledArray(): ON
D/BluetoothGatt(14651): connect() - device: XX:XX:XX:XX:12:61, auto: true
D/BluetoothGatt(14651): registerApp()
D/BluetoothGatt(14651): registerApp() - UUID=90076b8c-e9d2-4839-985b-cdcdafe51877
D/BluetoothGatt(14651): onClientRegistered() - status=0 clientIf=5
D/BluetoothAdapter(14651): getBleEnabledArray(): ON
```

chipweinberger, 2024-10-22:
please open a PR to fix it, thanks.

closed 2024-10-22 by chipweinberger

ImanGhasemiArani, 2024-10-23:
no,
the bug is for FBP. 
when bluetooth turned to on, you are call connect, but if the device is not available for 30-60 seconds, after that device douse not connect anymore,
after 60 seconds, I call disconnect(with queue=false) then connect(with autoConnect = true) manually, and works properly.

ImanGhasemiArani, 2024-10-23:
> the logs look fine. We call connect. There is nothing more FBP can do.
> 
> ```
> D/[FBP-Android](14651): [FBP] onMethodCall: connect
> D/BluetoothAdapter(14651): getBleEnabledArray(): ON
> D/BluetoothGatt(14651): connect() - device: XX:XX:XX:XX:12:61, auto: true
> ```
> 
> Open an issue with Google / AOSP.

no,
the bug is for FBP. 
when bluetooth turned to on, you are call connect, but if the device is not available for 30-60 seconds, after that device douse not connect anymore,
after 60 seconds, I call disconnect(with queue=false) then connect(with autoConnect = true) manually, and works properly.

chipweinberger, 2024-10-23:
> but if the device is not available for 30-60 seconds, after that device douse not connect anymore,

not true, we call with `autoconnect:true` parameter. It should not time out

chipweinberger, 2024-10-24:
thanks for reporting. autoconnect can be slow to detect the new device. try waiting longer maybe.

ImanGhasemiArani, 2024-10-25:
> thanks for reporting. autoconnect can be slow to detect the new device. try waiting longer maybe.

douse not connect at all.

Creiger, 2025-02-03:
Hi. i am having the same problem. When i connect ble button using autoconnect, then even when on bluetooth off listener i disconnect button and after bluetooth is on i call again autoconnect, it does not connect. Also tryed disconnecting button, then restart bluetooth and new scan. it can find the button but never autoconnects.

chipweinberger, 2025-02-03:
@Creiger , thanks for reporting. There's nothing I know of that could be improved. It works for me. But if you diagnose your problem further please report back.

Creiger, 2025-02-04:
Ok for me it was two things.
1. i did not know that when i have autoconnect: true and turn off the button, so connectionState is set to disconnected, i dont need to call autoconnect again when button is turned on again.
2. i can see in logs now that when the bluetooth is turned off, it is a different story and it calls disconnect on all connected buttons.

Creiger, 2025-02-04:
But i am still confused about one thing. If i disconnect the button manually i can see in logs cancelOpen. If it is done by bleutooth off trigger, this log is not there. If thne on bluetooth on i see in logs that the button was connected again, there is obviously no connection listener on it. What is cleared automaticly by the bluetooth off by FBP?

chipweinberger, 2025-02-04:
turning Bluetooth off does not disable auto connect

Flutter Blue Plus will automatically try to connect when Bluetooth is turned back on

vinayvishnu725, 2025-03-12:
so we need perment disconnect eventthough autoconnect is enabled ,while scanning the device is autconnects  the listner should not listen while scanning ,