chipweinberger/flutter_blue_plus #1032: Connected device delete from OS bluetooth device again connect the same device

balakrmu, 2024-10-23:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.2

### Flutter Version

 3.24.2

### What OS?

All

### OS Version

Android 12

### Bluetooth Module

AIROC

### What is your problem?

Below mentioned state disconnected device unable to connect again.
If I check the connection State it shows the disconnected. 
D/[FBP-Android]( 2814): [FBP] onMethodCall: connect
D/[FBP-Android]( 2814): [FBP] already connecting

### Logs

```shell
D/[FBP-Android]( 2423): [FBP] onMethodCall: removeBond
W/[FBP-Android]( 2423): [FBP] already not bonded
D/[FBP-Android]( 2423): [FBP] onMethodCall: connect
D/[FBP-Android]( 2423): [FBP] already connecting
I/flutter ( 2423): _connectAndBond() error: FlutterBluePlusException | createBond | fbp-code: 6 | device is not connected
```

balakrmu, 2024-10-23:
Bluetooth Module 
-

chipweinberger, 2024-10-23:
probably a problem with your peripheral BLE device.

closed 2024-10-23 by chipweinberger