chipweinberger/flutter_blue_plus #1033: [Help]: connectionState Returns duplicate status after stream subscription

CallMeBossC, 2024-10-23:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.26.0

### Flutter Version

3.9.12

### What OS?

Android, All

### OS Version

HarmonyOs4.2.0

### Bluetooth Module

Not sure

### What is your problem?

I use a Huawei phone to connect to a Bluetooth low energy device. I will subscribe to the device's status stream every time I connect. Whenever the connection is unsuccessful and I retry multiple connections, connectionState will return repeated device status, although I cancel the subscription to the connectionState stream after each connection.

### Logs

```shell
I/BluetoothLeAdvertiser(21284): stopAdvertising is called
I/BluetoothLeAdvertiser(21284): startAdvertising is called
I/BluetoothAdapter(21284): isLeEnabled(): ON
I/BluetoothLeAdvertiser(21284): startAdvertisingSet, calling package name = com.mesh.easysmart
I/BluetoothAdapter(21284): isLeEnabled(): ON
I/BluetoothLeAdvertiser(21284): startAdvertisingSet and gatt is: android.bluetooth.IBluetoothGatt$Stub$Proxy@1b64e69 and callback is: android.bluetooth.le.BluetoothLeAdvertiser$1@b3ff8f0
I/flutter (21284): printf_message:  _scanSubscription.cancel() LateInitializationError: Field '_scanSubscription@1917439335' has not been initialized.;
I/BluetoothLeAdvertiser(21284): stopAdvertising is called
I/BluetoothLeAdvertiser(21284): stopAdvertisingSet, calling package name = com.mesh.easysmart callback = android.bluetooth.le.BluetoothLeAdvertiser$1@b3ff8f0
I/BluetoothAdapter(21284): isLeEnabled(): ON
I/BluetoothLeAdvertiser(21284): startAdvertising is called
I/BluetoothAdapter(21284): isLeEnabled(): ON
I/BluetoothLeAdvertiser(21284): startAdvertisingSet, calling package name = com.mesh.easysmart
I/BluetoothAdapter(21284): isLeEnabled(): ON
I/BluetoothLeAdvertiser(21284): startAdvertisingSet and gatt is: android.bluetooth.IBluetoothGatt$Stub$Proxy@f31cdfa and callback is: android.bluetooth.le.BluetoothLeAdvertiser$1@8dc6825
D/BluetoothLeScanner(21284): onScannerRegistered() - status=0 scannerId=16 mScannerId=0
I/BluetoothLeAdvertiser(21284): advertise stopped and remove callback: android.bluetooth.le.BluetoothLeAdvertiser$1@b3ff8f0
I/BluetoothDevice(21284): get remote device name: null
I/BluetoothDevice(21284): get remote device name: null
I/BluetoothDevice(21284): get remote device name: null
I/BluetoothDevice(21284): get remote device name: null
I/BluetoothDevice(21284): get remote device name: null
I/BluetoothDevice(21284): get remote device name: null
I/BluetoothDevice(21284): get remote device name: null
I/BluetoothDevice(21284): get remote device name: null
I/BluetoothDevice(21284): get remote device name: smart lamp
I/BluetoothDevice(21284): get remote device name: null
I/BluetoothAdapter(21284): isLeEnabled(): ON
I/BluetoothLeAdvertiser(21284): stopAdvertising is called
I/BluetoothLeAdvertiser(21284): stopAdvertisingSet, calling package name = com.mesh.easysmart callback = android.bluetooth.le.BluetoothLeAdvertiser$1@8dc6825
I/BluetoothLeAdvertiser(21284): startAdvertising is called
I/BluetoothAdapter(21284): isLeEnabled(): ON
I/BluetoothLeAdvertiser(21284): startAdvertisingSet, calling package name = com.mesh.easysmart
I/BluetoothAdapter(21284): isLeEnabled(): ON
I/BluetoothLeAdvertiser(21284): advertise stopped and remove callback: android.bluetooth.le.BluetoothLeAdvertiser$1@8dc6825
[... 70 lines cut ...]
D/BluetoothGatt(21284): onClientConnectionState() - status=133 clientIf=17 device=FF:FF:**:**:**:FF
D/BluetoothGatt(21284): onClientConnectionState() - status=133 clientIf=16 device=FF:FF:**:**:**:FF
D/BluetoothGatt(21284): close()
D/BluetoothGatt(21284): close()
D/BluetoothGatt(21284): unregisterApp() - mClientIf=16
D/BluetoothGatt(21284): unregisterApp() - mClientIf=17
I/flutter (21284): printf_message:监听到设备状态BluetoothDevice{remoteId: FF:FF:FF:FF:FF:FF, platformName: smart lamp, services: null} BluetoothConnectionState.disconnected
I/flutter (21284): printf_message:监听到设备状态BluetoothDevice{remoteId: FF:FF:FF:FF:FF:FF, platformName: smart lamp, services: null} BluetoothConnectionState.disconnected
I/flutter (21284): printf_message:监听到设备状态BluetoothDevice{remoteId: FF:FF:FF:FF:FF:FF, platformName: smart lamp, services: null} BluetoothConnectionState.disconnected
I/flutter (21284): printf_message:监听到设备状态BluetoothDevice{remoteId: FF:FF:FF:FF:FF:FF, platformName: smart lamp, services: null} BluetoothConnectionState.disconnected
I/flutter (21284): printf_message:监听到设备状态BluetoothDevice{remoteId: FF:FF:FF:FF:FF:FF, platformName: smart lamp, services: null} BluetoothConnectionState.disconnected
I/flutter (21284): printf_message:监听到设备状态BluetoothDevice{remoteId: FF:FF:FF:FF:FF:FF, platformName: smart lamp, services: null} BluetoothConnectionState.disconnected
I/flutter (21284): printf_message:监听到设备状态BluetoothDevice{remoteId: FF:FF:FF:FF:FF:FF, platformName: smart lamp, services: null} BluetoothConnectionState.disconnected
I/flutter (21284): printf_message:监听到设备状态BluetoothDevice{remoteId: FF:FF:FF:FF:FF:FF, platformName: smart lamp, services: null} BluetoothConnectionState.disconnected
I/flutter (21284): printf_message:监听到设备状态BluetoothDevice{remoteId: FF:FF:FF:FF:FF:FF, platformName: smart lamp, services: null} BluetoothConnectionState.disconnected
I/flutter (21284): printf_message:监听到设备状态BluetoothDevice{remoteId: FF:FF:FF:FF:FF:FF, platformName: smart lamp, services: null} BluetoothConnectionState.disconnected
I/BluetoothDevice(21284): connectGatt: callingPid=21284 callingUid=10394
I/BluetoothDevice(21284): callingApp=com.mesh.easysmart
D/BluetoothGatt(21284): connect() - device: FF:FF:**:**:**:FF, auto: false
D/BluetoothGatt(21284): registerApp()
D/BluetoothGatt(21284): registerApp() - UUID=03643880-84c2-4230-b60a-caeef628709e
D/BluetoothGatt(21284): onClientRegistered() - status=0 clientIf=16
I/flutter (21284): printf_message:监听到设备状态BluetoothDevice{remoteId: FF:FF:FF:FF:FF:FF, platformName: smart lamp, services: null} BluetoothConnectionState.disconnected
[GETX] CLOSE DIALOG 559089158
```

chipweinberger, 2024-10-23:
this is expected behavior.

you can use .`skip(1)` to avoid it

closed 2024-10-23 by chipweinberger

CallMeBossC, 2024-10-24:
> this is expected behavior.
> 
> you can use .`skip(1)` to avoid it

Is there an API that can do it every time you use the Bluetooth plug-in library, it caches ah streams or subscriptions are cleared