chipweinberger/flutter_blue_plus #1035: [Help]: Flutter Blue Plus Error in writing on a characteristic

Miragaia, 2024-10-25:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.33.5

### Flutter Version

3.24.2

### What OS?

Android

### OS Version

Android 14

### Bluetooth Module

Bluetooth 5.0

### What is your problem?

Using flutter_blue_plus: ^1.33.5

I have a smartwatch code generating services and characteristics. The smartphone connects to the smartwatch and uses "discoverServices", iterates every service and every characteristic to find one that is writable.

Whenever the writable characteristic is found it tries to write to it, but an Unknown Error is given (19).

Uint8List data = Uint8List data = Uint8List.fromList([0x01, 0x02, 0x03, 0x04]);

Future<void> _sendDataToDevice(Uint8List data) async {

    if (connectedDevice != null) {
      try {
        await Future.delayed(const Duration(seconds: 2));
        List<BluetoothService> services = await connectedDevice!.discoverServices();
        for (BluetoothService service in services) {
          print('Service: ${service.uuid}');
          for (BluetoothCharacteristic characteristic in service.characteristics) {
            print('Characteristic: ${characteristic.uuid}');
            print('Properties: ${characteristic.properties}');
            if (characteristic.properties.write || characteristic.properties.writeWithoutResponse) { //characteristic.properties.write && 
              print('Found writable characteristic');
              print('UUID: ${characteristic.uuid}');
              // Enable notifications if needed
              if (characteristic.properties.notify) {
                print('Enabling notifications');
                await characteristic.setNotifyValue(true);
              }
              
              // Write data with appropriate write type
              print('Writing data...');
              if (characteristic.properties.write) {
                await characteristic.write(data);
                print('Data sent to smartwatch with response: $data');
              } else {
                print('Characteristic does not support write with response');
              }

              print('Data sent to smartwatch: $data');
              return;
            }
          }
        }
        print('No writable characteristic found');
      } catch (e) {
        print('Error sending data: $e');
      }
    } else {
      print('No device connected');
    }
  }

I get that error, I would like to know how to fix it, it seems that it is only on this line "await characteristic.write(data); "

### Logs

```shell
Logs and error:
/flutter (15410): Connected to Galaxy Watch7 (JB0B)
I/ViewRootImpl@95c32e5[MainActivity](15410): ViewPostIme pointer 0
I/ViewRootImpl@95c32e5[MainActivity](15410): ViewPostIme pointer 1
D/[FBP-Android](15410): [FBP] onMethodCall: discoverServices
D/BluetoothGatt(15410): discoverServices() - device: XX:XX:XX:XX:02:8E
D/BluetoothGatt(15410): onSearchComplete() = Device=XX:XX:XX:XX:02:8E Status=0
D/[FBP-Android](15410): [FBP] onServicesDiscovered:
D/[FBP-Android](15410): [FBP]   count: 4
D/[FBP-Android](15410): [FBP]   status: 0GATT_SUCCESS
D/[FBP-Android](15410): [FBP] onMethodCall: setNotifyValue
D/BluetoothGatt(15410): setCharacteristicNotification() - uuid: 00002a05-0000-1000-8000-00805f9b34fb enable: true
W/[FBP-Android](15410): [FBP] CCCD descriptor for characteristic not found: 2a05
I/flutter (15410): Service: 1801
I/flutter (15410): Characteristic: 2a05
I/flutter (15410): Properties: CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: false, write: false, notify: false, indicate: true, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}
I/flutter (15410): Characteristic: 2b3a
I/flutter (15410): Properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: false, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}
I/flutter (15410): Characteristic: 2b29
I/flutter (15410): Properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: false, write: true, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}
I/flutter (15410): Found writable characteristic
I/flutter (15410): UUID: 2b29
I/flutter (15410): Writing data...
D/[FBP-Android](15410): [FBP] onMethodCall: writeCharacteristic
E/[FBP-Android](15410): [FBP] onCharacteristicWrite:
E/[FBP-Android](15410): [FBP]   chr: 2b29
E/[FBP-Android](15410): [FBP]   status: UNKNOWN_GATT_ERROR (19) (19)
I/flutter (15410): Error sending data: FlutterBluePlusException | writeCharacteristic | android-code: 19 | UNKNOWN_GATT_ERROR (19)
```

Miragaia, 2024-10-26:
The smartwatch I am testing is the Galaxy Watch 7.
I have tested sending data to the Mi Band 4 and it sends the data as supposed, but the same code does not work for the Galaxy Watch 7, I receive the error.

chipweinberger, 2024-10-26:
that error is coming from your device. 

you'll need to check the firmware logs.

closed 2024-10-26 by chipweinberger