chipweinberger/flutter_blue_plus #1036: [Help]: Advertising Data

Yolo-cell-hash, 2024-10-28:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.12

### Flutter Version

 3.24.3

### What OS?

Android

### OS Version

Android 14

### Bluetooth Module

-

### What is your problem?

Can I somehow advertise my device data / any custom data to the BLE device ? I tried reading some references to get here 
```

class AdvertiseData {
  final FlutterBlue flutterBlue = FlutterBlue.instance;

  Future<void> advertiseDeviceData() async {
    final advertisementData = AdvertisementData(
      localName: 'My BLE Device',
      serviceUuids: [Guid('12345678-1234-5678-1234-56789abcdef0')],
    );

    FlutterBluePlus flb = FlutterBluePlus();
    await flb.startAdvertising(
      advertisementData: advertisementData,
      timeout: Duration(seconds: 30), // Advertise for 30 seconds
    );

    print('Advertising started');
  }
}
```

Is there any way to achieve this ?

### Logs

```shell
'AdvertisementData' isn't a function. 
'Guid' isn't a function
The method 'startAdvertising' isn't defined for the type 'FlutterBluePlus'.
```

MrCsabaToth, 2024-10-28:
I think what you want is so called "peripheral mode" of Bluetooth: when your device (probably a phone or tablet) augments a Bluetooth device. Most of the Bluetooth plugins (including FBP) do not support / target peripheral mode. The few plugins I saw and tried didn't work, but let me know if you find any because I'm also interested.

chipweinberger, 2024-10-28:
this is address in the readme:

> ## Introduction
> 
> FlutterBluePlus is a Bluetooth Low Energy plugin for [Flutter](https://flutter.dev). 
> 
> It supports BLE Central Role only (most common). 
> 
> If you need BLE Peripheral Role, you should check out [FlutterBlePeripheral](https://pub.dev/packages/flutter_ble_peripheral), or [bluetooth_low_energy](https://pub.dev/packages/bluetooth_low_energy).

the answer is no. Try https://pub.dev/packages/bluetooth_low_energy instead.

closed 2024-10-28 by chipweinberger