chipweinberger/flutter_blue_plus #1037: [Help]: Unable to Retrieve Real-Time Data from Masimo MightySat Pulse Oximeter

Fudal, 2024-10-29:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.33.5

### Flutter Version

3.24.4

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

Masimo MightySat

### What is your problem?

I am attempting to connect my Flutter app to a Masimo MightySat pulse oximeter using the flutter_blue_plus library. I can successfully connect to the device, discover services, and view characteristics, but I am not able to receive any real-time data (e.g., pulse rate, SpO₂, and other metrics) when subscribing to the available characteristics.

The MediM Health app demonstrates that it’s possible to receive real-time data from the device immediately upon connection, so I believe the issue is related to how data is accessed via flutter_blue_plus in my implementation.

### Steps to Reproduce
Connect to the Masimo MightySat device using flutter_blue_plus.
Discover services and characteristics successfully.
Attempt to subscribe to each available characteristic to retrieve data.
No data is received on any characteristic.

### Expected Behavior
Upon connection and subscription to the appropriate characteristics, real-time data (such as pulse rate and SpO₂) should be accessible and updated continuously.

### Actual Behavior
Although I successfully connect to the Masimo MightySat device and see the list of services and characteristics, no data is available when subscribing to them.
I am not receiving any updates for the pulse rate, SpO₂, or other metrics.

### Additional Context
The MediM Health app can display real-time data from the MightySat device immediately after connection, so I know the device supports this.
I suspect this might be related to the correct characteristic UUIDs or configuration settings specific to the MightySat.

I'm using Flutter Blue Plus example app. I can see every service and characteristic but after subscribe I don't have data, the MediM Health app always have real-time data after connect.

What can I do?
I really need to implement getting data from that pulse oximeter in my Flutter app. I think using the example of Flutter Blue Plus is the best way to have information about working that device with my phone, with Flutter.

### Logs

```shell
connect() - device: DE:9E:61:5A:91:74, auto: false, eattSupport: false
registerApp() - UUID=602b6e74-7811-41c6-a6f4-235736ba8a11

[FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -72, connectable: 1, manufacturer_data: {76: 1219d00383231e0069d2422e1f020c40e65291381c06c8711b01eb}, remote_id: FD:1A:02:1C:42:3B, service_data: {}}]}
[FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -71, connectable: 1, manufacturer_data: {}, remote_id: 40:88:79:D5:BA:13, service_data: {fd5a: 15a75002b1516d1ddd67bd94b7000000913d17c7}, service_uuids: [fd5a]}]}
[FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -78, connectable: 1, manufacturer_data: {}, remote_id: 52:7B:23:D2:5B:CF, service_data: {fd5a: 15a750025ed3a213380be0d3b700000010b7f695}, service_uuids: [fd5a]}]}

I/flutter (30567): [FBP] <setNotifyValue> args: {remote_id: DE:9E:61:5A:91:74, service_uuid: 54c21000-a720-4b4f-11e4-9fe20002a5d5, secondary_service_uuid: null, characteristic_uuid: 54c21002-a720-4b4f-11e4-9fe20002a5d5, force_indications: false, enable: true}
D/[FBP-Android](30567): [FBP] onMethodCall: setNotifyValue
D/BluetoothGatt(30567): setCharacteristicNotification() - uuid: 54c21002-a720-4b4f-11e4-9fe20002a5d5 enable: true
I/flutter (30567): [FBP] <setNotifyValue> result: true
D/[FBP-Android](30567): [FBP] onDescriptorWrite:
D/[FBP-Android](30567): [FBP]   chr: 54c21002-a720-4b4f-11e4-9fe20002a5d5
D/[FBP-Android](30567): [FBP]   desc: 2902
D/[FBP-Android](30567): [FBP]   status: GATT_SUCCESS (0)
I/flutter (30567): [FBP] [[ OnDescriptorWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: 54c21000-a720-4b4f-11e4-9fe20002a5d5, success: 1, remote_id: DE:9E:61:5A:91:74, descriptor_uuid: 2902, error_code: 0, characteristic_uuid: 54c21002-a720-4b4f-11e4-9fe20002a5d5, value: 0100}
```

chipweinberger, 2024-10-29:
Sorry, I'm not familiar with that device.

I wont be able to help you.

Try asking people that know your device,

closed 2024-10-29 by chipweinberger

Fudal, 2024-10-29:
@chipweinberger I just want to know what can I change in FBP example code, to have data. Even not how change but just what can I do. It is odd that FBP can see services and characters but doesn't have data from subscription. I don't know where can I find help.

chipweinberger, 2024-10-29:
this is already in the readme

https://github.com/chipweinberger/flutter_blue_plus?tab=readme-ov-file#onvaluereceived-is-never-called-or-lastvaluestream