chipweinberger/flutter_blue_plus #1038: [Help]: Ble auto connection issue when bluetooth turn on/off in Android 11

kklds, 2024-10-31:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.33.5

### Flutter Version

3.13.7

### What OS?

Android

### OS Version

Android 11

### Bluetooth Module

Not sure about it

### What is your problem?

Turn off Bluetooth, open the app and turn Bluetooth back on, Android 14 can connect automatically, but Android 11 auto-connect will report an errorr：‘’Error during connection or service discovery: FlutterBluePlusException | connect | android-code: 133 | ANDROID_SPECIFIC_ERROR‘’

### Logs

```shell
D/[FBP-Android](18822): [FBP] OnAdapterStateChanged: turningOn
I/flutter (18822): [FBP] [[ OnAdapterStateChanged ]] result: {adapter_state: 3}
D/[FBP-Android](18822): [FBP] OnAdapterStateChanged: on
I/flutter (18822): [FBP] [[ OnAdapterStateChanged ]] result: {adapter_state: 4}
I/flutter (18822): [FBP] <connect> args: {remote_id: C1:44:99:85:E3:FC, auto_connect: 0}
D/[FBP-Android](18822): [FBP] onMethodCall: connect
D/BluetoothGatt(18822): connect() - device: C1:44:99:85:E3:FC, auto: false
D/BluetoothGatt(18822): registerApp()
D/BluetoothGatt(18822): registerApp() - UUID=1deb5dd0-c6c6-4990-a397-c72b356e89c5
D/BluetoothGatt(18822): onClientRegistered() - status=0 clientIf=6
I/flutter (18822): [FBP] <connect> result: true
D/BluetoothGatt(18822): onClientConnectionState() - status=133 clientIf=6 device=C1:44:99:85:E3:FC
D/[FBP-Android](18822): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](18822): [FBP]   status: ANDROID_SPECIFIC_ERROR
D/BluetoothGatt(18822): close()
D/BluetoothGatt(18822): unregisterApp() - mClientIf=6
I/flutter (18822): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 133, disconnect_reason_string: ANDROID_SPECIFIC_ERROR, remote_id: C1:44:99:85:E3:FC, connection_state: 0}
I/flutter (18822): Error during connection or service discovery: FlutterBluePlusException | connect | android-code: 133 | ANDROID_SPECIFIC_ERROR
```

kklds, 2024-10-31:
This is the configuration inside my Android
[image]

chipweinberger, 2024-10-31:
see readme. impossible for us to fix it.

closed 2024-10-31 by chipweinberger