chipweinberger/flutter_blue_plus #1039: the Bluetooth automatically disconnects after 30 seconds

special521, 2024-11-01:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.33.5

### Flutter Version

3.18

### What OS?

Android

### OS Version

Android 14

### Bluetooth Module

IOT-08

### What is your problem?

setNotifyValue After subscription, if the device sends the data stream continuously, the data can be received normally, but after 30 seconds, the Bluetooth connection will be automatically disconnected, and the debugging will be done using the official flutter_blue_plus example, if the device does not automatically transmit the data stream after subscription, Then the Bluetooth will not automatically disconnect, the official example has such a bug, how to solve it

### Logs

```shell
Flutter assets will be downloaded from https://storage.flutter-io.cn. Make sure you trust this source!
Launching lib\main.dart on V2199A in debug mode...
Running Gradle task 'assembleDebug'...
Flutter assets will be downloaded from https://storage.flutter-io.cn. Make sure you trust this source!
√ Built build\app\outputs\flutter-apk\app-debug.apk
Installing build\app\outputs\flutter-apk\app-debug.apk...
I/flutter (17103): The Dart VM service is listening on ************************************
I/flutter (17687): The Dart VM service is listening on ************************************
I/flutter (17687): [FBP] <setLogLevel> args: 5
I/flutter (17687): [FBP] <setLogLevel> result: true
I/flutter (17687): [FBP] <getAdapterState> args: null
I/flutter (17687): [FBP] <getAdapterState> result: {adapter_state: 4}
I/flutter (17687): [FBP] <getSystemDevices> args: {with_services: [180f]}
I/flutter (17687): [FBP] <getSystemDevices> result: {devices: []}
I/flutter (17687): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 1, android_legacy: false, android_scan_mode: 2, android_uses_fine_location: false}
I/flutter (17687): [FBP] <startScan> result: true
I/flutter (17687): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -63, manufacturer_data: {6: 010920025b2c6ae6ad86375ff00e36bedb9a30f2ea7d7a4312db5e}, remote_id: *****************, service_data: {}}]}
I/flutter (17687): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -66, manufacturer_data: {6: 01092002d94e277239a4d5635eedcfc783c21330124ef8a97cd647}, remote_id: *****************, service_data: {}}]}
I/flutter (17687): 连接状态BluetoothConnectionState.disconnected
I/flutter (17687): 连接状态BluetoothConnectionState.disconnected
I/flutter (17687): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -79, connectable: 1, manufacturer_data: {17224: 2d30380000000000000000000000000000000000}, remote_id: *****************, platform_name: IOT-08, service_data: {}, adv_name: IOT-08
I/flutter (17687): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -86, manufacturer_data: {6: 010920025616db45326ef5b1613693a746cdbfc30db94c709c9a55}, remote_id: *****************, service_data: {}}]}
I/flutter (17687): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -63, manufacturer_data: {6: 01092002264b54e1b2ca7a8b53104be947e041f2ed5edbe15fec41}, remote_id: *****************, service_data: {}}]}
I/flutter (17687): 连接状态BluetoothConnectionState.disconnected
I/flutter (17687): 连接状态BluetoothConnectionState.disconnected
I/flutter (17687): 连接状态BluetoothConnectionState.disconnected
I/flutter (17687): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -66, manufacturer_data: {2103: 20b10180304944901c4445533635303d71c42d3133323030340000}, remote_id: *****************, service_data: {}}]}
I/flutter (17687): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -70, manufacturer_data: {6: 0109200228af1c2c9721cb1748bc06694e95f5006cc853229c346e}, remote_id: *****************, service_data: {}}]}
I/flutter (17687): 连接状态BluetoothConnectionState.disconnected
I/flutter (17687): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -75, connectable: 1, manufacturer_data: {2103: 834034363064313540000769514f203131033d71c42d0000c0a864644f000000000000000000}, remote_id: *****************, service_data: {}}]}
I/flutter (17687): 连接状态BluetoothConnectionState.disconnected
I/flutter (17687): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -103, manufacturer_data: {117: 4204012066200d000205200104000100000000000000000042040180667c0a3f751fa57e0a3f751fa401560000000000}, remote_id: *****************, platform_name: [Monitor] Samsung M7 (32), service_data: {}}]}
I/flutter (17687): 连接状态BluetoothConnectionState.disconnected
I/flutter (17687): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -86, connectable: 1, manufacturer_data: {0: 606e415285c1}, remote_id: *****************, platform_name: SFH_303Q_85C1, service_data: {}, adv_name: SFH_303Q_85C1, service_uuids: [fee7]}]}
I/flutter (17687): 连接状态BluetoothConnectionState.disconnected
[... 43 lines cut ...]
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff2, success: 1, remote_id: *****************, error_code: 0, characteristic_uuid: fff3, value: 01bd}
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff2, success: 1, remote_id: *****************, error_code: 0, characteristic_uuid: fff3, value: ac051d000000f0000000ef3c**************00}
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff2, success: 1, remote_id: *****************, error_code: 0, characteristic_uuid: fff3, value: 01bd}
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff2, success: 1, remote_id: *****************, error_code: 0, characteristic_uuid: fff3, value: ac051d000000f0000000ef3c**************00}
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff2, success: 1, remote_id: *****************, error_code: 0, characteristic_uuid: fff3, value: 01bd}
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff2, success: 1, remote_id: *****************, error_code: 0, characteristic_uuid: fff3, value: ac051d000000f0000000ef3c**************00}
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff2, success: 1, remote_id: *****************, error_code: 0, characteristic_uuid: fff3, value: 01bd}
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff2, success: 1, remote_id: *****************, error_code: 0, characteristic_uuid: fff3, value: ac051d000000f0000000ef3c**************00}
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff2, success: 1, remote_id: *****************, error_code: 0, characteristic_uuid: fff3, value: 01bd}
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff2, success: 1, remote_id: *****************, error_code: 0, characteristic_uuid: fff3, value: ac051d000000f0000000ef3c**************00}
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff2, success: 1, remote_id: *****************, error_code: 0, characteristic_uuid: fff3, value: 01bd}
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff2, success: 1, remote_id: *****************, error_code: 0, characteristic_uuid: fff3, value: ac051d000000f0000000ef3c**************00}
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff2, success: 1, remote_id: *****************, error_code: 0, characteristic_uuid: fff3, value: 01bd}
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff2, success: 1, remote_id: *****************, error_code: 0, characteristic_uuid: fff3, value: ac051d000000f0000000ef3c**************00}
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff2, success: 1, remote_id: *****************, error_code: 0, characteristic_uuid: fff3, value: 01bd}
[... cut ...]
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_SUCCESS, service_uuid: fff2, success: 1, remote_id: *****************, error_code: 0, characteristic_uuid: fff3, value: 01bd}
I/flutter (17687): [FBP] [[ OnCharacteristicReceived ]] result: {error_string: GATT_ERROR, service_uuid: fff2, success: 0, remote_id: *****************, error_code: 133, characteristic_uuid: fff3, value: 00}
I/flutter (17687): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 22, disconnect_reason_string: CONNECTION_TERMINATED_BY_LOCAL_HOST, remote_id: *****************, connection_state: 0}
I/flutter (17687): 连接状态BluetoothConnectionState.disconnected
```

special521, 2024-11-01:
[image]
Click subscribe data stream uninterrupted transmission, Bluetooth automatically disconnects after 30 seconds

chipweinberger, 2024-11-01:
google this: CONNECTION_TERMINATED_BY_LOCAL_HOST

FBP cannot fix it

closed 2024-11-01 by chipweinberger