chipweinberger/flutter_blue_plus #1044: [Help]: StartScan withServiceData Filter not working properly

SamuelBehrmann, 2024-11-06:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.33.6

### Flutter Version

3.24.2

### What OS?

macOS

### OS Version

android 14

### Bluetooth Module

raspberry pi

### What is your problem?

Scanning for BLE devices by service works like a charm and my device is found. The issue arrives when I try using the parameter: `withServiceData`. I entered the exact service data that i was searching for yet no device was found. I also tried to search for partial serviceData by applying a mask, yet no device found found. 
If i on the other hand apply the filter and mask manually in the search results, I am able to find the device. 

My best guess is that some of the information is lost along the way. But im very irritated by the fact that i can not find any other reported issues about this. 

```dart
  // version 1 - full service data
  await FlutterBluePlus.startScan(
      androidUsesFineLocation: true,
      androidScanMode: AndroidScanMode.lowLatency,
      withServiceData: [
        ServiceDataFilter(
          Guid.fromString(_CHIP_UUID),
          data: [0x00, 0x00, 0x0f, 0xf1, 0xff, 0x01, 0x80, 0x00],
          mask: [255, 255, 255, 255, 255, 255, 255, 255],
        ),
      ],
      timeout: const Duration(seconds: 15),
    );
    
    // version 2 - partial search with mask
    await FlutterBluePlus.startScan(
      androidUsesFineLocation: true,
      androidScanMode: AndroidScanMode.lowLatency,
      withServices: [
        Guid.fromString(_CHIP_UUID),
      ],
      withServiceData: [
        ServiceDataFilter(
          Guid.fromString(_CHIP_UUID),
          data: [0, 0, 15],
          mask: [255, 255, 255],
        ),
      ],
      timeout: const Duration(seconds: 15),
    );
```

### Logs

```shell
W/1.raster(27397): type=1400 audit(0.0:16756): avc:  denied  { getattr } for  name="/" dev="dmabuf" ino=1 scontext=u:r:untrusted_app:s0:c60,c257,c512,c768 tcontext=u:object_r:unlabeled:s0 tclass=filesystem permissive=0 app=com.example.matter_jni_example
I/flutter (27397): BluetoothManagerImpl: Turn on bluetooth
D/[FBP-Android](27397): [FBP] onMethodCall: flutterRestart
D/[FBP-Android](27397): [FBP] disconnectAllDevices(flutterRestart)
D/[FBP-Android](27397): [FBP] connectedPeripherals: 0
I/flutter (27397): [FBP] <turnOn> args: null
D/[FBP-Android](27397): [FBP] onMethodCall: turnOn
D/DeferrableSurface(27397): surface closed,  useCount=2 closed=true androidx.camera.core.SurfaceRequest$2@16e7a4c
D/DeferrableSurface(27397): surface closed,  useCount=0 closed=true androidx.camera.core.processing.SurfaceEdge$SettableSurface@859bc20
D/DeferrableSurface(27397): Surface terminated[total_surfaces=2, used_surfaces=2](androidx.camera.core.processing.SurfaceEdge$SettableSurface@859bc20}
D/DeferrableSurface(27397): use count-1,  useCount=1 closed=true androidx.camera.core.SurfaceRequest$2@16e7a4c
D/DeferrableSurface(27397): surface closed,  useCount=1 closed=true androidx.camera.core.impl.ImmediateSurface@52f1b11
D/Camera2CameraImpl(27397): {Camera@cc4c88b[id=0]} Use cases [androidx.camera.core.Preview-b0416419-350d-412a-8a82-2a6e185d0b5b205441204, androidx.camera.core.ImageAnalysis-d17dbb45-ba24-42af-9a79-2b7f65921e30119176413] now DETACHED for camera
D/UseCaseAttachState(27397): All use case: [] for camera: 0
I/flutter (27397): [FBP] (turnOn) result: false
D/Camera2CameraImpl(27397): {Camera@cc4c88b[id=0]} Resetting Capture Session
D/Camera2CameraImpl(27397): {Camera@cc4c88b[id=0]} Releasing session in state OPENED
2
D/UseCaseAttachState(27397): Active and attached use case: [] for camera: 0
D/Camera2CameraImpl(27397): {Camera@cc4c88b[id=0]} Closing camera.
D/Camera2CameraImpl(27397): {Camera@cc4c88b[id=0]} Transitioning camera internal state: OPENED --> CLOSING
D/CameraStateRegistry(27397): Recalculating open cameras:
D/CameraStateRegistry(27397): Camera                                       State                 
D/CameraStateRegistry(27397): -------------------------------------------------------------------
D/CameraStateRegistry(27397): Camera@9515eb2[id=1]                         UNKNOWN               
D/CameraStateRegistry(27397): Camera@cc4c88b[id=0]                         CLOSING               
D/CameraStateRegistry(27397): -------------------------------------------------------------------
D/CameraStateRegistry(27397): Open count: 1 (Max allowed: 1)
D/CameraStateMachine(27397): New public camera state CameraState{type=CLOSING, error=null} from CLOSING and null
D/CameraStateMachine(27397): Publishing new public camera state CameraState{type=CLOSING, error=null}
D/Camera2CameraImpl(27397): {Camera@cc4c88b[id=0]} Resetting Capture Session
D/Camera2CameraImpl(27397): {Camera@cc4c88b[id=0]} Releasing session in state CLOSING
D/CaptureSession(27397): onSessionFinished()
E/BufferQueueProducer(27397): [SurfaceTexture-0-27397-20](id:6b0500000042,api:4,p:1155,c:27397) dequeueBuffer: BufferQueue has been abandoned
D/TransportRuntime.SQLiteEventStore(27397): Storing event with priority=VERY_LOW, name=FIREBASE_ML_SDK for destination cct
I/flutter (27397): [FBP] <getAdapterState> args: null
D/[FBP-Android](27397): [FBP] onMethodCall: getAdapterState
I/flutter (27397): [FBP] (getAdapterState) result: {adapter_state: 4}
D/TransportRuntime.JobInfoScheduler(27397): Upload for context TransportContext(cct, VERY_LOW, MSRodHRwczovL2ZpcmViYXNlbG9nZ2luZy5nb29nbGVhcGlzLmNvbS92MGNjL2xvZy9iYXRjaD9mb3JtYXQ9anNvbl9wcm90bzNc) is already scheduled. Returning...
[log] BluetoothManagerImpl: Start scan with service
      data: [0, 0, 15]
      mask: [255, 255, 255]
E/BufferQueueProducer(27397): [SurfaceTexture-0-27397-20](id:6b0500000042,api:4,p:1155,c:27397) cancelBuffer: BufferQueue has been abandoned
I/flutter (27397): [FBP] <startScan> args: {with_services: [fff6], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [{service: fff6, data: 00000f, mask: ffffff}], continuous_updates: false, continuous_divisor: 1, android_legacy: false, android_scan_mode: 2, android_uses_fine_location: true}
D/[FBP-Android](27397): [FBP] onMethodCall: startScan
D/UseCaseAttachState(27397): Active and attached use case: [] for camera: 0
D/DeferrableSurface(27397): use count-1,  useCount=0 closed=true androidx.camera.core.SurfaceRequest$2@16e7a4c
D/DeferrableSurface(27397): Surface no longer in use[total_surfaces=2, used_surfaces=1](androidx.camera.core.SurfaceRequest$2@16e7a4c}
D/DeferrableSurface(27397): Surface terminated[total_surfaces=1, used_surfaces=1](androidx.camera.core.SurfaceRequest$2@16e7a4c}
D/DeferrableSurface(27397): use count-1,  useCount=0 closed=true androidx.camera.core.impl.ImmediateSurface@52f1b11
D/DeferrableSurface(27397): Surface no longer in use[total_surfaces=1, used_surfaces=0](androidx.camera.core.impl.ImmediateSurface@52f1b11}
D/DeferrableSurface(27397): Surface terminated[total_surfaces=0, used_surfaces=0](androidx.camera.core.impl.ImmediateSurface@52f1b11}
D/Camera2CameraImpl(27397): {Camera@cc4c88b[id=0]} CameraDevice.onClosed()
D/Camera2CameraImpl(27397): {Camera@cc4c88b[id=0]} Transitioning camera internal state: CLOSING --> INITIALIZED
D/CameraStateRegistry(27397): Recalculating open cameras:
D/CameraStateRegistry(27397): Camera                                       State                 
D/CameraStateRegistry(27397): -------------------------------------------------------------------
D/CameraStateRegistry(27397): Camera@9515eb2[id=1]                         UNKNOWN               
D/CameraStateRegistry(27397): Camera@cc4c88b[id=0]                         CLOSED                
D/CameraStateRegistry(27397): -------------------------------------------------------------------
D/CameraStateRegistry(27397): Open count: 0 (Max allowed: 1)
D/CameraStateMachine(27397): New public camera state CameraState{type=CLOSED, error=null} from CLOSED and null
D/CameraStateMachine(27397): Publishing new public camera state CameraState{type=CLOSED, error=null}
I/flutter (27397): [FBP] stopScan: already stopped
```

chipweinberger, 2024-11-06:
please try to debug the issue and open a PR.

I dont think I ever tested this code when I merged it. It's not a common filter to use.

SamuelBehrmann, 2024-11-07:
I have debugged the example app and found the following: 
- The GUID is not parsed the same way for `services` and `serviceData´
this leads to an exception being thrown on java layer. 
- Adding the missing `uuid128(...)` solves the issue. 
- It looks like the error is not properly propagated to the very top dart layer since the exception is only visible when debugging. 

[image]

```java
//FlutterBluePlusPlugin.java
                        // Line - 543 
                        // services
                        for (int i = 0; i < withServices.size(); i++) {
                            ParcelUuid s = ParcelUuid.fromString(uuid128(withServices.get(i)));
                            ScanFilter f = new ScanFilter.Builder().setServiceUuid(s).build();
                            filters.add(f);
                        }

                        // Line - 595 
                         / / service data
                        for (int i = 0; i < withServiceData.size(); i++) {
                            HashMap<String, Object> m = (HashMap<String, Object>) withServiceData.get(i);
                            ParcelUuid s = ParcelUuid.fromString((String) m.get("service"));
                            byte[] mdata =             hexToBytes((String) m.get("data"));
                            byte[] mask =              hexToBytes((String) m.get("mask"));
                            ScanFilter f = null;
                            if (mask.length == 0) {
                                f = new ScanFilter.Builder().setServiceData(s, mdata).build();
                            } else {
                                f = new ScanFilter.Builder().setServiceData(s, mdata, mask).build();
                            }
                            filters.add(f);
                        }

```

This fixed the issue for me. 
`ParcelUuid s = ParcelUuid.fromString(uuid128(((String) m.get("service"))));`

SamuelBehrmann, 2024-11-07:
Created this PR: https://github.com/chipweinberger/flutter_blue_plus/pull/1046

chipweinberger, 2024-11-07:
merged!

SamuelBehrmann, 2024-11-07:
👍 
@chipweinberger 
Small FYI Master is broken due to a missing `,` in the parameter list in line 2605 of `FlutterBluePlusPlugin.java`

chipweinberger, 2024-11-08:
closing. fixed in 1.34.0

closed 2024-11-08 by chipweinberger