chipweinberger/flutter_blue_plus #1047: [Help]: Why is the format of `manufacturerData` different between iOS and Android?

P0labrD, 2024-11-07:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.33.2

### Flutter Version

3.24.3

### What OS?

Android, iOS

### OS Version

iOS 15.8.3, Android 14

### Bluetooth Module

ESP32-WROOM-32E

### What is your problem?

I'm looking at the `advertisementData` of my bluetooth device on both Android and iOS, and I can't figure why the format of 
 `manufacturerData` and `msd` properties are different: the `manufacturerId` key is accessed differently between both os
 
 ```json
{
  "android": {
    "advName": "advName",
    "txPowerLevel": null,
    "appearance": null,
    "connectable": true,
    "manufacturerData": {
      "0": [5, 54 /* ... */],
      "12345": []
    },
    "msd": [
      [0, 0, 5, 54 /* ... */],
      [48, 57]
    ],
    "serviceData": {},
    "serviceUuids": []
  },

  "ios": {
    "advName": "advName",
    "txPowerLevel": null,
    "appearance": null,
    "connectable": true,
    "manufacturerData": {
      "12345": [0, 0, 5, 54 /* ... */]
    },
    "msd": [[48, 57, 0, 0, 5, 54 /* ... */]],
    "serviceData": {},
    "serviceUuids": []
  }
}
 ```
 
I can't find any documentation on this difference, so before introducing platform specific code in production, I would like to be sure it is developed as intented.

Thank you very much!
 
 

### Logs

```shell
{
  "android": {
    "advName": "advName",
    "txPowerLevel": null,
    "appearance": null,
    "connectable": true,
    "manufacturerData": {
      "0": [5, 54 /* ... */],
      "12345": []
    },
    "msd": [
      [0, 0, 5, 54 /* ... */],
      [48, 57]
    ],
    "serviceData": {},
    "serviceUuids": []
  },

  "ios": {
    "advName": "advName",
    "txPowerLevel": null,
    "appearance": null,
    "connectable": true,
    "manufacturerData": {
      "12345": [0, 0, 5, 54 /* ... */]
    },
    "msd": [[48, 57, 0, 0, 5, 54 /* ... */]],
    "serviceData": {},
    "serviceUuids": []
  }
}
```

chipweinberger, 2024-11-07:
they should not be different. please open a PR if there is a bug.

`msd` = full raw bytes
`manufacturerData` = map of <manufactureId, data>

chipweinberger, 2024-11-08:
also, where is this json coming from?

print out the advertisement data directly please and report back

`print(advertisementData)`

```
  @override
  String toString() {
    return 'AdvertisementData{'
        'advName: $advName, '
        'txPowerLevel: $txPowerLevel, '
        'appearance: $appearance, '
        'connectable: $connectable, '
        'manufacturerData: $manufacturerData, '
        'serviceData: $serviceData, '
        'serviceUuids: $serviceUuids'
        '}';
  }
```

P0labrD, 2024-11-08:
I wrote the json with what I saw in my debugger, anonymizing the data
Here's the real (still anonymized) print ouput for iOS: 
```sh
flutter: AdvertisementData{advName: product-name, txPowerLevel: null, appearance: null, connectable: true, manufacturerData: {12345: [0, 0, 5, 54, 48, 48, 85, 48, 54, 57, 48, 57, 50, 49]}, serviceData: {}, serviceUuids: []}
```

And for Android:
```sh
I/flutter (19040): AdvertisementData{advName: product-name, txPowerLevel: null, appearance: null, connectable: true, manufacturerData: {0: [5, 54, 48, 48, 85, 48, 54, 57, 48, 57, 50, 49], 12345: []}, serviceData: {}, serviceUuids: []}
```

I'm not a hardware developer myself, but I could ask the one who did develop on the chip what he implemented in order to give good reproduction info

chipweinberger, 2024-11-08:
the android code is here. you should debug it further :) 

```
    Map<Integer, byte[]> getManufacturerSpecificData(ScanRecord adv) {
        byte[] bytes = adv.getBytes();
        Map<Integer, byte[]> manufacturerDataMap = new HashMap<>();
        int n = 0;
        while (n < bytes.length) {

            // layout:
            // n[0] = fieldlen
            // n[1] = datatype (MSD)
            // n[2] = manufacturerId (low)
            // n[3] = manufacturerId (high)
            // n[4] = data...
            int fieldLen = bytes[n] & 0xFF;

            // no more or malformed data
            if (fieldLen <= 0) {
                break;
            }

            // Ensuring we don't go past the bytes array
            if (n + fieldLen >= bytes.length) {
                break;
            }

            int dataType = bytes[n + 1] & 0xFF;

            // Manufacturer Specific Data magic number
            // At least 3 bytes: 2 for manufacturer ID & 1 for dataType
            if (dataType == 0xFF && fieldLen >= 3) {

                // Manufacturer Id
                int high = (bytes[n + 3] & 0xFF) << 8;
                int low = (bytes[n + 2] & 0xFF);
                int manufacturerId = high | low;

                // the length of the msd data,
                // excluding manufacturerId & dataType
                int msdLen = fieldLen - 3;

                // ptr to msd data
                // excluding manufacturerId & dataType
                int msdPtr = n + 4;

                // add to map
                if (manufacturerDataMap.containsKey(manufacturerId)) {
                    // If the manufacturer ID already exists, append the new data to the existing list
                    byte[] existingData = manufacturerDataMap.get(manufacturerId);
                    byte[] mergedData = new byte[existingData.length + msdLen];
                    // Merge arrays
                    System.arraycopy(existingData, 0, mergedData, 0, existingData.length);
                    System.arraycopy(bytes, msdPtr, mergedData, existingData.length, msdLen);
                    manufacturerDataMap.put(manufacturerId, mergedData);
                } else {
                    // Otherwise, put the new manufacturer ID and its data into the map
                    byte[] data = new byte[msdLen];
                    // Starting from n+4 because manufacturerId occupies n+2 and n+3
                    System.arraycopy(bytes, msdPtr, data, 0, data.length);
                    manufacturerDataMap.put(manufacturerId, data);
                }
            }

            n += fieldLen + 1;
        }

        return manufacturerDataMap;
    }
    ```

chipweinberger, 2024-11-08:
@MrCsabaToth

MrCsabaToth, 2024-11-10:
> @MrCsabaToth

On Sunday I might have a little time to inspect the related code. I don't have an iOS device though, so it'll be Android and code inspection. As far as I know last time I modified it because of the FedEx tag, it meant to be the same. It had an off by one byte bug you corrected later.

chipweinberger, 2024-11-10:
i inspected the code, looked fine to me.

MrCsabaToth, 2024-11-10:
Hey @P0labrD, could you screenshot the raw advertisement data displayed in nRF Connect on Android vs iOS so we can have even more munition? The raw data looks like this https://github.com/chipweinberger/flutter_blue_plus/issues/785#issue-2119339981 and this https://github.com/chipweinberger/flutter_blue_plus/issues/785#issuecomment-2095307561

P0labrD, 2024-11-12:
Hello @MrCsabaToth, thank you for your time.

Here are the two raw advertisement data displayed: 
|Android|iOS|
|-|-|
|[image]|[image]|

P0labrD, 2024-11-13:
Update: my hardware developer apparently did not fully understand the [Core Blutooth Specification Part A 1.4. Manufacturer Specific Data](https://www.bluetooth.com/wp-content/uploads/Files/Specification/HTML/CSS_v11/out/en/supplement-to-the-bluetooth-core-specification/data-types-specification.html): as seen [on the android screen above](https://github.com/chipweinberger/flutter_blue_plus/issues/1047#issuecomment-2470416524), he decided to put the manufacturerId on another "lane" as the specific data, whereas it should be but in the 2 first bytes of the 16 bytes of the manufacturerData.

What I don't understand at the moment is how iOS CoreBluetooth works, and how it "magically" did the matching itself.

chipweinberger, 2024-11-13:
yes we should still try to fix them so they match

chipweinberger, 2024-11-22:
So I think I have the answer.

According to ChatGPT, you should only have one MSD in an advertisement.

iOS just concatenates all the data together.

So we should fix this on Android, and remove all the "map" stuff. And fix/break the API.

chipweinberger, 2025-01-15:
@tnc1997 , would be great to fix this. Requires a platform interface change ideally.

chipweinberger, 2025-04-11:
merged: 115c392e199f1235330a0420cee1532055a9b448

closed 2025-04-11 by chipweinberger

ekuleshov, 2025-04-11:
> merged: [115c392](https://github.com/chipweinberger/flutter_blue_plus/commit/115c392e199f1235330a0420cee1532055a9b448)

Hmm. Couldn't we just use LinkedHashMap() to preserve the order?
Perhaps that is generally good idea to use that one instead of plain HashMap everywhere on Android

chipweinberger, 2025-04-11:
order is preserved with the array.

we don't need a map at all.

ekuleshov, 2025-04-11:
I mean there are 30+ statements like `new HashMap()` in the `FlutterBluePlusPlugin` Java code.

Some could be replaced with Map.of() and some may have some significance, e.g. `bmScanAdvertisement` collects several maps. So, would be good to preserve the order in those as is.