chipweinberger/flutter_blue_plus #1048: [Feature]: BluetoothUtils:1.90接收普通数据 >>>>>>setNotifyValue:  PlatformException(setNotifyValue, secondary service not found '55535343-fe7d-4ae5-8fa9-9fafd205e455', null, null)

Johnjson, 2024-11-08:
### FlutterBluePlus Version

^1.34.1

### Flutter Version

Flutter version 3.19.5

### What OS?

Android

### OS Version

android 32

### Bluetooth Module

连接

### What is your feature request?

result: {error_string: GATT_SUCCESS, success: 1, remote_id: 30:30:F9:6F:2F:46, error_code: 0, services: [{characteristics: [{descriptors: [{service_uuid: 1801, remote_id: 30:30:F9:6F:2F:46, descriptor_uuid: 2902, primary_service_uuid: 1801, characteristic_uuid: 2a05}], service_uuid: 1801, remote_id: 30:30:F9:6F:2F:46, primary_service_uuid: 1801, characteristic_uuid: 2a05, properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 0, authenticated_signed_writes: 0, extended_properties: 0, indicate: 1, indicate_encryption_required: 0, write: 0, notify: 0}}, {descriptors: [], service_uuid: 1801, remote_id: 30:30:F9:6F:2F:46, primary_service_uuid: 1801, characteristic_uuid: 2b29, properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 1, authenticated_signed_writes: 0, extended_properties: 0, indicate: 0, indicate_encryption_required: 0, write: 1, notify: 0}}, {descriptors: [], service_uuid: 1801, remote_id: 30:30
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.1发现服务成功 >>>>>>name: 
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:_handlerServicesservice sUuid: 1801
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:_handlerServicesservice serviceUuid: 1801
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:_handlerServicesservice sUuid: 1800
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:_handlerServicesservice serviceUuid: 1800
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:_handlerServicesservice sUuid: 55535343-fe7d-4ae5-8fa9-9fafd205e455
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:_handlerServicesservice serviceUuid: 55535343-fe7d-4ae5-8fa9-9fafd205e455
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.2找到设置模式的服务 >>>>>>name:   serviceGuid: 55535343-fe7d-4ae5-8fa9-9fafd205e455
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.4_readCharacteristics >>>>>>name:   cUuid: b39b7234-beec-d4a8-f443-418843535349
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.4_readCharacteristics >>>>>>name:   cUuid: 16962447-c623-61ba-d94b-4d1e43535349
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.6找到写数据的特征值 >>>>>>name:   characteristicUUID: 16962447-c623-61ba-d94b-4d1e43535349
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.4_readCharacteristics >>>>>>name:   cUuid: 2359b521-cb23-4ad4-8fa9-431e84535349
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.5接收普通回复命令 >>>>>>name:   cUuid:2359b521-cb23-4ad4-8fa9-431e84535349
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.8设置为通知模式 >>>>>>name: 
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.7当前mtu: 500 请求设置mtu为512 >>>>>>name: 
I/flutter (26340): [FBP] <setNotifyValue> args: {remote_id: 30:30:F9:6F:2F:46, service_uuid: 55535343-fe7d-4ae5-8fa9-9fafd205e455, characteristic_uuid: 2359b521-cb23-4ad4-8fa9-431e84535349, primary_service_uuid: 55535343-fe7d-4ae5-8fa9-9fafd205e455, force_indications: false, enable: true}
D/[FBP-Android](26340): [FBP] onMethodCall: setNotifyValue
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   — — — — — — — — — — st — — — — — — — — — — — — —
I/flutter (26340): LogUtils d  2024-11-08 15:28:44  | BluetoothUtils:1.90接收普通数据 >>>>>>setNotifyValue:  PlatformException(setNotifyValue, secondary service not found '55535343-fe7d-4a
I/flutter (26340): LogUtils d  2024-11-08 15:28:44  | e5-8fa9-9fafd205e455', null, null)

### Logs

```shell
result: {error_string: GATT_SUCCESS, success: 1, remote_id: 30:30:F9:6F:2F:46, error_code: 0, services: [{characteristics: [{descriptors: [{service_uuid: 1801, remote_id: 30:30:F9:6F:2F:46, descriptor_uuid: 2902, primary_service_uuid: 1801, characteristic_uuid: 2a05}], service_uuid: 1801, remote_id: 30:30:F9:6F:2F:46, primary_service_uuid: 1801, characteristic_uuid: 2a05, properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 0, authenticated_signed_writes: 0, extended_properties: 0, indicate: 1, indicate_encryption_required: 0, write: 0, notify: 0}}, {descriptors: [], service_uuid: 1801, remote_id: 30:30:F9:6F:2F:46, primary_service_uuid: 1801, characteristic_uuid: 2b29, properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 1, authenticated_signed_writes: 0, extended_properties: 0, indicate: 0, indicate_encryption_required: 0, write: 1, notify: 0}}, {descriptors: [], service_uuid: 1801, remote_id: 30:30
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.1发现服务成功 >>>>>>name: 
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:_handlerServicesservice sUuid: 1801
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:_handlerServicesservice serviceUuid: 1801
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:_handlerServicesservice sUuid: 1800
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:_handlerServicesservice serviceUuid: 1800
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:_handlerServicesservice sUuid: 55535343-fe7d-4ae5-8fa9-9fafd205e455
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:_handlerServicesservice serviceUuid: 55535343-fe7d-4ae5-8fa9-9fafd205e455
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.2找到设置模式的服务 >>>>>>name:   serviceGuid: 55535343-fe7d-4ae5-8fa9-9fafd205e455
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.4_readCharacteristics >>>>>>name:   cUuid: b39b7234-beec-d4a8-f443-418843535349
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.4_readCharacteristics >>>>>>name:   cUuid: 16962447-c623-61ba-d94b-4d1e43535349
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.6找到写数据的特征值 >>>>>>name:   characteristicUUID: 16962447-c623-61ba-d94b-4d1e43535349
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.4_readCharacteristics >>>>>>name:   cUuid: 2359b521-cb23-4ad4-8fa9-431e84535349
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.5接收普通回复命令 >>>>>>name:   cUuid:2359b521-cb23-4ad4-8fa9-431e84535349
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.8设置为通知模式 >>>>>>name: 
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   BluetoothUtils:1.7当前mtu: 500 请求设置mtu为512 >>>>>>name: 
I/flutter (26340): [FBP] <setNotifyValue> args: {remote_id: 30:30:F9:6F:2F:46, service_uuid: 55535343-fe7d-4ae5-8fa9-9fafd205e455, characteristic_uuid: 2359b521-cb23-4ad4-8fa9-431e84535349, primary_service_uuid: 55535343-fe7d-4ae5-8fa9-9fafd205e455, force_indications: false, enable: true}
D/[FBP-Android](26340): [FBP] onMethodCall: setNotifyValue
I/flutter (26340): LogUtils d  2024-11-08 15:28:44   — — — — — — — — — — st — — — — — — — — — — — — —
I/flutter (26340): LogUtils d  2024-11-08 15:28:44  | BluetoothUtils:1.90接收普通数据 >>>>>>setNotifyValue:  PlatformException(setNotifyValue, secondary service not found '55535343-fe7d-4a
I/flutter (26340): LogUtils d  2024-11-08 15:28:44  | e5-8fa9-9fafd205e455', null, null)
```

chipweinberger, 2024-11-08:
Hi, can you try this change?

```
    private ChrFound locateCharacteristic(BluetoothGatt gatt,
                                                 String serviceUuid,
                                                 String characteristicUuid, 
                                                 String primaryServiceUuid)
    {
        // remember this
        boolean isSecondaryService = primaryServiceUuid != null && primaryServiceUuid.length() != 0;
        
        ```

shingohu, 2024-11-08:
same issue,just update library from 1.33.5 to 1.34.1

chipweinberger, 2024-11-08:
please use 1.33.6, 1.34.1 is broken.

chipweinberger, 2024-11-08:
fixed in 1.34.2

reopen a new issue if you have new problems.

Johnjson, 2024-11-11:
flutter_blue_plus: ^1.34.4 Update to this version The issue has been resolved