chipweinberger/flutter_blue_plus #1049: [Help]: exceptions in startScan() are not propagated to Flutter code

mbalamut, 2024-11-08:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.34.2

### Flutter Version

3.24.3

### What OS?

Android

### OS Version

12

### Bluetooth Module

unknown

### What is your problem?

When starting a scan is not possible, i.e. due to Bluetooth turned off or refused permissions, startScan() doesn't throw any exception, also no error is added to scanResults.
Logs are from the example app after commenting out the systemDevices() call.

### Logs

```shell
Launching lib\main.dart on SM G975F in debug mode...
You are applying Flutter's app_plugin_loader Gradle plugin imperatively using the apply script method, which is deprecated and will be removed in a future release. Migrate to applying Gradle plugins with the declarative plugins block: https://flutter.dev/to/flutter-gradle-plugin-apply

You are applying Flutter's main Gradle plugin imperatively using the apply script method, which is deprecated and will be removed in a future release. Migrate to applying Gradle plugins with the declarative plugins block: https://flutter.dev/to/flutter-gradle-plugin-apply

Warning: unable to detect project KGP version. Skipping version checking.
This may be because you have applied KGP after the Flutter Gradle Plugin.
√ Built build\app\outputs\flutter-apk\app-debug.apk
Connecting to VM Service at ws://127.0.0.1:65010/vhDHgCMWatc=/ws
Connected to the VM Service.
I/flutter (26249): [FBP] <setLogLevel> args: 5
D/[FBP-Android](26249): [FBP] onMethodCall: setLogLevel
D/[FBP-Android](26249): [FBP] initializing BluetoothAdapter
I/BluetoothAdapter(26249): BluetoothAdapter() : com.lib.flutter_blue_plus_example
I/Gralloc4(26249): mapper 4.x is not supported
W/Gralloc3(26249): mapper 3.x is not supported
I/gralloc (26249): Arm Module v1.0
W/Gralloc4(26249): allocator 4.x is not supported
W/Gralloc3(26249): allocator 3.x is not supported
I/flutter (26249): [FBP] (setLogLevel) result: true
D/[FBP-Android](26249): [FBP] onMethodCall: flutterRestart
I/BluetoothAdapter(26249): STATE_ON
D/[FBP-Android](26249): [FBP] disconnectAllDevices(flutterRestart)
D/[FBP-Android](26249): [FBP] connectedPeripherals: 0
I/flutter (26249): [FBP] <getAdapterState> args: null
D/[FBP-Android](26249): [FBP] onMethodCall: getAdapterState
I/flutter (26249): [FBP] (getAdapterState) result: {adapter_state: 4}
D/ViewRootImpl@969423[MainActivity](26249): Creating frameDrawingCallback nextDrawUseBlastSync=false reportNextDraw=true hasBlurUpdates=false
D/ViewRootImpl@969423[MainActivity](26249): Creating frameCompleteCallback
I/SurfaceView@72c531c(26249): uSP: rtp = Rect(0, 0 - 1080, 2154) rtsw = 1080 rtsh = 2154
D/ViewRootImpl@969423[MainActivity](26249): Received frameDrawingCallback frameNum=1. Creating transactionCompleteCallback=false
I/SurfaceView@72c531c(26249): onSSPAndSRT: pl = 0 pt = 0 sx = 1.0 sy = 1.0
I/SurfaceView@72c531c(26249): aOrMT: uB = true t = android.view.SurfaceControl$Transaction@13dc262 fN = 1 android.view.SurfaceView.access$500:124 android.view.SurfaceView$SurfaceViewPositionUpdateListener.positionChanged:1728 android.graphics.RenderNode$CompositePositionUpdateListener.positionChanged:319
I/SurfaceView@72c531c(26249): aOrMT: vR.mWNT, vR = ViewRootImpl@969423[MainActivity]
[... 22 lines cut ...]
D/InsetsSourceConsumer(26249): ensureControlAlpha: for ITYPE_STATUS_BAR on com.lib.flutter_blue_plus_example/com.lib.flutter_blue_plus_example.MainActivity
I/ViewRootImpl@969423[MainActivity](26249): Relayout returned: old=(0,0,1080,2280) new=(0,0,1080,2280) req=(1080,2280)0 dur=3 res=0x1 s={true 473546728480} ch=false fn=6
I/ViewRootImpl@969423[MainActivity](26249): updateBoundsLayer: t = android.view.SurfaceControl$Transaction@491fa61 sc = Surface(name=Bounds for - com.lib.flutter_blue_plus_example/com.lib.flutter_blue_plus_example.MainActivity@0)/@0x4775986 frame = 6
I/ViewRootImpl@969423[MainActivity](26249): mWNT: t = android.view.SurfaceControl$Transaction@491fa61 fN = 6 android.view.ViewRootImpl.prepareSurfaces:2783 android.view.ViewRootImpl.performTraversals:4029 android.view.ViewRootImpl.doTraversal:2924
I/ViewRootImpl@969423[MainActivity](26249): mWNT: merge t to BBQ
I/ViewRootImpl@969423[MainActivity](26249): MSG_WINDOW_FOCUS_CHANGED 1 1
D/InputMethodManager(26249): startInputInner - Id : 0
I/InputMethodManager(26249): startInputInner - mService.startInputOrWindowGainedFocus
I/flutter (26249): [FBP] stopScan: already stopped
I/flutter (26249): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 1, android_legacy: false, android_scan_mode: 2, android_uses_fine_location: false}
D/[FBP-Android](26249): [FBP] onMethodCall: startScan
I/ViewRootImpl@969423[MainActivity](26249): MSG_WINDOW_FOCUS_CHANGED 0 1
I/ViewRootImpl@969423[MainActivity](26249): stopped(false) old=false
I/DecorView(26249): notifyKeepScreenOnChanged: keepScreenOn=false
D/InsetsSourceConsumer(26249): ensureControlAlpha: for ITYPE_NAVIGATION_BAR on com.lib.flutter_blue_plus_example/com.lib.flutter_blue_plus_example.MainActivity
D/InsetsSourceConsumer(26249): ensureControlAlpha: for ITYPE_STATUS_BAR on com.lib.flutter_blue_plus_example/com.lib.flutter_blue_plus_example.MainActivity
I/ViewRootImpl@969423[MainActivity](26249): Relayout returned: old=(0,0,1080,2280) new=(0,0,1080,2280) req=(1080,2280)0 dur=4 res=0x1 s={true 473546728480} ch=false fn=11
I/ViewRootImpl@969423[MainActivity](26249): updateBoundsLayer: t = android.view.SurfaceControl$Transaction@491fa61 sc = Surface(name=Bounds for - com.lib.flutter_blue_plus_example/com.lib.flutter_blue_plus_example.MainActivity@0)/@0x4775986 frame = 11
I/ViewRootImpl@969423[MainActivity](26249): mWNT: t = android.view.SurfaceControl$Transaction@491fa61 fN = 11 android.view.ViewRootImpl.prepareSurfaces:2783 android.view.ViewRootImpl.performTraversals:4029 android.view.ViewRootImpl.doTraversal:2924
I/ViewRootImpl@969423[MainActivity](26249): mWNT: merge t to BBQ
I/ViewRootImpl@969423[MainActivity](26249): MSG_WINDOW_FOCUS_CHANGED 1 1
D/InputMethodManager(26249): startInputInner - Id : 0
I/InputMethodManager(26249): startInputInner - mService.startInputOrWindowGainedFocus
I/flutter (26249): [FBP] stopScan: already stopped
```

chipweinberger, 2024-11-08:
I think you are mistaken. We check for bluetooth on and permissions here.

```
                case "startScan":
                {
                    // see: BmScanSettings
                    HashMap<String, Object> data = call.arguments();
                    List<String> withServices =    (List<String>) data.get("with_services");
                    List<String> withRemoteIds =   (List<String>) data.get("with_remote_ids");
                    List<String> withNames =       (List<String>) data.get("with_names");
                    List<String> withKeywords =    (List<String>) data.get("with_keywords");
                    List<Object> withMsd =         (List<Object>) data.get("with_msd");
                    List<Object> withServiceData = (List<Object>) data.get("with_service_data");
                    boolean continuousUpdates =         (boolean) data.get("continuous_updates");
                    boolean androidLegacy =             (boolean) data.get("android_legacy");
                    int androidScanMode =                   (int) data.get("android_scan_mode");
                    boolean androidUsesFineLocation =   (boolean) data.get("android_uses_fine_location");

                    ArrayList<String> permissions = new ArrayList<>();

                    if (Build.VERSION.SDK_INT >= 31) { // Android 12 (October 2021)
                        permissions.add(Manifest.permission.BLUETOOTH_SCAN);
                        if (androidUsesFineLocation) {
                            permissions.add(Manifest.permission.ACCESS_FINE_LOCATION);
                        }
                        // it is unclear why this is needed, but some phones throw a
                        // SecurityException AdapterService getRemoteName, without it
                        permissions.add(Manifest.permission.BLUETOOTH_CONNECT);
                    }

                    if (Build.VERSION.SDK_INT <= 30) { // Android 11 (September 2020)
                        permissions.add(Manifest.permission.ACCESS_FINE_LOCATION);
                    }

                    ensurePermissions(permissions, (granted, perm) -> {

                        if (granted == false) {
                            result.error("startScan",
                                String.format("FlutterBluePlus requires %s permission", perm), null);
                            return;
                        }

                        // check adapter
                        if (isAdapterOn() == false) {
                            result.error("startScan", String.format("bluetooth must be turned on"), null);
                            return;
                        }

                        // get scanner
                        BluetoothLeScanner scanner = mBluetoothAdapter.getBluetoothLeScanner();
                        if(scanner == null) {
                            result.error("startScan", String.format("getBluetoothLeScanner() is null. Is the Adapter on?"), null);
                            return;
                        }
```

mbalamut, 2024-11-08:
Sorry, I meant the Flutter user code.
Exceptions are caught in onError and not propagated further. Please see my PR.

chipweinberger, 2024-11-08:
thanks. fixed in 1.34.3

closed 2024-11-08 by chipweinberger