chipweinberger/flutter_blue_plus #1050: [Help]: setNotifyValue, secondary service not found 'ff22', null, null

kukafei, 2024-11-08:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.34.2

### Flutter Version

3.19.1

### What OS?

Android

### OS Version

android 13

### Bluetooth Module

unkown

### What is your problem?

After upgrading to the latest version, executing 'await characteristic. setNotifyValue (true)' resulted in errors such as' setNotifyValue ',' secondary service not found 'ff22', 'null', and 'null'.  The previous version 1.33.5 did not have this issue.

### Logs

```shell
[38;5;196m[E]PlatformException(setNotifyValue, secondary service not found 'ff22', null, null)
```

chipweinberger, 2024-11-08:
it would be great if you could figure out the root cause and open a PR.

roberto1984, 2024-11-08:
Hi,

Same issue here,can not figure out since this morning,I'm using 128 bit uuids

chipweinberger, 2024-11-09:
this bug was fixed in 1.34.2. please check you are on 1.34.2 or later.

chipweinberger, 2024-11-09:
try 1.34.4, and create a new issue if you still have problems.

closed 2024-11-09 by chipweinberger