chipweinberger/flutter_blue_plus #1053: [Help]: discoverServices() method is not working after disconnect and connect back again

vontdeux, 2024-11-09:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.33.5

### Flutter Version

3.16.0

### What OS?

Android

### OS Version

10

### Bluetooth Module

HC-05

### What is your problem?

Connect to a device, discover services, subscribe to it's characteristic - works fine
disconnect , connect back again, try discover device, it will time out 15s
what i did is i just add a disconnect button and method
```
ScanResultTile(
            result: r,
            onTap: () => onConnectPressed(r.device),
            onDis: () => onDisconnectPressed(r.device),
          ),
```

and 
```
return ExpansionTile(
      title: _buildTitle(context),
      leading: Text(widget.result.rssi.toString()),
      trailing: Container(
        child: isConnected ? _buildDisconnectButton(context) : _buildConnectButton(context),
      ),
```

tested on this modified example app and my app, both having same results

### Logs

```shell
Restarted application in 2,514ms.
I/flutter (14514): [FBP] <setLogLevel> args: 5
D/[FBP-Android](14514): [FBP] onMethodCall: setLogLevel
I/flutter (14514): [FBP] <setLogLevel> result: true
D/[FBP-Android](14514): [FBP] onMethodCall: flutterRestart
D/[FBP-Android](14514): [FBP] disconnectAllDevices(flutterRestart)
D/[FBP-Android](14514): [FBP] calling disconnect: C0:20:12:14:10:01
D/BluetoothGatt(14514): cancelOpen() - device: C0:20:12:14:10:01
D/[FBP-Android](14514): [FBP] calling close: C0:20:12:14:10:01
D/BluetoothGatt(14514): close()
D/BluetoothGatt(14514): unregisterApp() - mClientIf=31
D/[FBP-Android](14514): [FBP] connectedPeripherals: 0
I/flutter (14514): [FBP] <getAdapterState> args: null
D/[FBP-Android](14514): [FBP] onMethodCall: getAdapterState
[... 137 lines cut ...]
D/[FBP-Android](14514): [FBP] onMethodCall: readRssi
I/flutter (14514): [FBP] <connect> args: {remote_id: C0:20:12:14:10:01, auto_connect: 0}
D/[FBP-Android](14514): [FBP] onMethodCall: connect
D/BluetoothGatt(14514): connect() - device: C0:20:12:14:10:01, auto: false
D/BluetoothGatt(14514): registerApp()
D/BluetoothGatt(14514): registerApp() - UUID=098334a0-d1d4-4875-91a8-0a2fdc242ba8
D/BluetoothGatt(14514): onClientRegistered() - status=0 clientIf=31
I/flutter (14514): [FBP] <connect> result: true
D/BluetoothGatt(14514): onClientConnectionState() - status=0 clientIf=31 device=C0:20:12:14:10:01
D/[FBP-Android](14514): [FBP] onConnectionStateChange:connected
D/[FBP-Android](14514): [FBP]   status: SUCCESS
I/flutter (14514): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: C0:20:12:14:10:01, connection_state: 1}
I/flutter (14514): [FBP] <readRssi> args: C0:20:12:14:10:01
D/[FBP-Android](14514): [FBP] onMethodCall: readRssi
D/BluetoothGatt(14514): readRssi() - device: C0:20:12:14:10:01
D/[FBP-Android](14514): [FBP] onReadRemoteRssi:
D/[FBP-Android](14514): [FBP]   rssi: -56
D/[FBP-Android](14514): [FBP]   status: GATT_SUCCESS (0)
I/flutter (14514): [FBP] <readRssi> result: true
I/flutter (14514): [FBP] [[ OnReadRssi ]] result: {error_string: GATT_SUCCESS, rssi: -56, success: 1, remote_id: C0:20:12:14:10:01, error_code: 0}
I/flutter (14514): [FBP] <discoverServices> args: C0:20:12:14:10:01
D/[FBP-Android](14514): [FBP] onMethodCall: discoverServices
D/BluetoothGatt(14514): discoverServices() - device: C0:20:12:14:10:01
I/flutter (14514): [FBP] <discoverServices> result: true
```

chipweinberger, 2024-11-09:
seems like a bug in android, or your peripheral's firmware.

If you call `discoverServices`, but the `OnDiscoveredServices` callback is never called, then there is nothing FBP can do about it.

chipweinberger, 2024-11-09:
if you find out the underlying cause, please report back. I'm curious. 

For now, I'll close because this is working as expected.

closed 2024-11-09 by chipweinberger

hezhipengzipp, 2025-03-27:
??? I  met this quesion too

ekuleshov, 2025-08-13:
> I’m also getting this error: `discoverService` was called, but the `OnDiscoveredServices` callback is never triggered. However, I can still connect to the device and discover services using the `nRF Connect ` app.

@ppupha try to connect to your service from FBP example app

ppupha, 2025-08-14:
> I’m also getting this error: discoverService was called, but the OnDiscoveredServices callback is never triggered. However, I can still connect to the device and discover services using the nRF Connect  app.
>
>@ppupha try to connect to your service from FBP example app

Tried with FBP example app. The device connects successfully but fails to discover services.
Here is the log:

```
I/flutter (26851): [FBP] <connect> args: {remote_id: C0:BF:BE:89:12:B7, auto_connect: 0}
D/[FBP-Android](26851): [FBP] onMethodCall: connect
D/CompatChangeReporter(26851): Compat change id reported: 265103382; UID 10397; state: ENABLED
D/BluetoothGatt(26851): connect() - device: XX:XX:XX:XX:12:B7, auto: false
D/BluetoothGatt(26851): registerApp()
D/BluetoothGatt(26851): registerApp() - UUID=73858745-5cbf-4b07-8d90-549e6ec5119d
D/BluetoothGatt(26851): onClientRegistered() - status=0 clientIf=50
W/WindowOnBackDispatcher(26851): OnBackInvokedCallback is not enabled for the application.
W/WindowOnBackDispatcher(26851): Set 'android:enableOnBackInvokedCallback="true"' in the application manifest.
I/flutter (26851): [FBP] (connect) result: true
D/BluetoothGatt(26851): onClientConnectionState() - status=0 clientIf=50 connected=true device=C0:BF:BE:89:12:B7
D/[FBP-Android](26851): [FBP] onConnectionStateChange:connected
D/[FBP-Android](26851): [FBP]   status: SUCCESS
I/flutter (26851): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: C0:BF:BE:89:12:B7, connection_state: 1}
I/flutter (26851): [FBP] <readRssi> args: C0:BF:BE:89:12:B7
D/[FBP-Android](26851): [FBP] onMethodCall: readRssi
D/BluetoothGatt(26851): readRssi() - device: XX:XX:XX:XX:12:B7
D/[FBP-Android](26851): [FBP] onReadRemoteRssi:
D/[FBP-Android](26851): [FBP]   rssi: -43
D/[FBP-Android](26851): [FBP]   status: GATT_SUCCESS (0)
I/flutter (26851): [FBP] (readRssi) result: true
I/flutter (26851): [FBP] [[ OnReadRssi ]] result: {error_string: GATT_SUCCESS, rssi: -43, success: 1, remote_id: C0:BF:BE:89:12:B7, error_code: 0}
D/BluetoothGatt(26851): onConnectionUpdated() - Device=C0:BF:BE:89:12:B7 interval=6 latency=0 timeout=500 status=0
I/flutter (26851): [FBP] <discoverServices> args: C0:BF:BE:89:12:B7
D/[FBP-Android](26851): [FBP] onMethodCall: discoverServices
D/BluetoothGatt(26851): discoverServices() - device: XX:XX:XX:XX:12:B7
I/flutter (26851): [FBP] (discoverServices) result: true
I/flutter (26851): FlutterBluePlusException | discoverServices | fbp-code: 1 | Timed out after 15s
I/flutter (26851): backtrace: #0      FutureTimeout.fbpTimeout.<anonymous closure> (package:flutter_blue_plus/src/utils.dart:18:7)
I/flutter (26851): #1      _RootZone.run (dart:async/zone.dart:1670:54)
I/flutter (26851): #2      Future.timeout.<anonymous closure> (dart:async/future_impl.dart:998:34)
I/flutter (26851): <asynchronous suspension>
I/flutter (26851): #3      BluetoothDevice.discoverServices (package:flutter_blue_plus/src/bluetooth_device.dart:281:43)
I/flutter (26851): <asynchronous suspension>
I/flutter (26851): #4      _DeviceScreenState.onDiscoverServicesPressed (package:flutter_blue_plus_example/screens/device_screen.dart:130:19)
I/flutter (26851): <asynchronous suspension>
D/BluetoothGatt(26851): onClientConnectionState() - status=22 clientIf=50 connected=false device=C0:BF:BE:89:12:B7
D/[FBP-Android](26851): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](26851): [FBP]   status: CONNECTION_TERMINATED_BY_LOCAL_HOST
D/BluetoothGatt(26851): close()
D/BluetoothGatt(26851): unregisterApp() - mClientIf=50
I/flutter (26851): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 22, disconnect_reason_string: CONNECTION_TERMINATED_BY_LOCAL_HOST, remote_id: C0:BF:BE:89:12:B7, connection_state: 0}
```