chipweinberger/flutter_blue_plus #1054: [Help]: 1.33.6 to 1.34.0 changes result in failed discoverServices on Android

MrCsabaToth, 2024-11-10:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.34.4

### Flutter Version

3.24.4

### What OS?

Android

### OS Version

14

### Bluetooth Module

N/A

### What is your problem?

For some reason between the version 1.33.6 and 1.34.0 and since then there were some changes which cause basic code to fail.

### Logs

```shell
I/flutter (27428): FlutterBluePlusException | discoverServices | fbp-code: 6 | device is not connected
I/flutter (27428): trace:
I/flutter (27428): #0      BluetoothDevice.discoverServices (package:flutter_blue_plus/src/bluetooth_device.dart:250:7)
I/flutter (27428): #1      DeviceBase.discover (package:track_my_indoor_exercise/devices/gadgets/device_base.dart:214:32)
I/flutter (27428): #2      FitnessEquipment.discover (package:track_my_indoor_exercise/devices/gadgets/fitness_equipment.dart:675:33)
I/flutter (27428): #3      FitnessEquipment.connectOnDemand (package:track_my_indoor_exercise/devices/gadgets/fitness_equipment.dart:517:27)
I/flutter (27428): <asynchronous suspension>
I/flutter (27428): #4      FindDevicesState.goToRecording (package:track_my_indoor_exercise/ui/find_devices.dart:781:21)
I/flutter (27428): <asynchronous suspension>
I/flutter (27428): #5      FindDevicesState.build.<anonymous closure>.<anonymous closure>.<anonymous closure> (package:track_my_indoor_exercise/ui/find_devices.dart:981:29)
I/flutter (27428): <asynchronous suspension>
I/flutter (27428): FlutterBluePlusException | discoverServices | fbp-code: 6 | device is not connected
I/flutter (27428): trace:
I/flutter (27428): #0      BluetoothDevice.discoverServices (package:flutter_blue_plus/src/bluetooth_device.dart:250:7)
I/flutter (27428): #1      DeviceBase.discover (package:track_my_indoor_exercise/devices/gadgets/device_base.dart:223:34)
I/flutter (27428): <asynchronous suspension>
I/flutter (27428): #2      FitnessEquipment.discover (package:track_my_indoor_exercise/devices/gadgets/fitness_equipment.dart:675:21)
I/flutter (27428): <asynchronous suspension>
I/flutter (27428): #3      FitnessEquipment.connectOnDemand (package:track_my_indoor_exercise/devices/gadgets/fitness_equipment.dart:517:21)
I/flutter (27428): <asynchronous suspension>
I/flutter (27428): #4      FindDevicesState.goToRecording (package:track_my_indoor_exercise/ui/find_devices.dart:781:21)
I/flutter (27428): <asynchronous suspension>
I/flutter (27428): #5      FindDevicesState.build.<anonymous closure>.<anonymous closure>.<anonymous closure> (package:track_my_indoor_exercise/ui/find_devices.dart:981:29)
I/flutter (27428): <asynchronous suspension>
```
```

MrCsabaToth, 2024-11-10:
I tried to debug the example app but it pukes a long ass message. I've seen that before but I need to leave the hotel soon where I'm debugging these fitness machines and I don't have more time:
```
You are applying Flutter's app_plugin_loader Gradle plugin imperatively using the apply script method, which is deprecated and will be removed in a future release. Migrate to applying Gradle plugins with the declarative plugins block: https://flutter.dev/to/flutter-gradle-plugin-apply

You are applying Flutter's main Gradle plugin imperatively using the apply script method, which is deprecated and will be removed in a future release. Migrate to applying Gradle plugins with the declarative plugins block: https://flutter.dev/to/flutter-gradle-plugin-apply

Warning: unable to detect project KGP version. Skipping version checking. 
This may be because you have applied KGP after the Flutter Gradle Plugin.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:checkDebugAarMetadata'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.CheckAarMetadataWorkAction
   > 14 issues were found when checking AAR metadata:

       1.  Dependency 'androidx.fragment:fragment:1.7.1' requires libraries and applications that
           depend on it to compile against version 34 or later of the
           Android APIs.

           :app is currently compiled against android-33.

           Also, the maximum recommended compile SDK version for Android Gradle
           plugin 8.0.2 is 33.

           Recommended action: Update this project's version of the Android Gradle
           plugin to one that supports 34, then update this project to use
           compileSdk of at least 34.

           Note that updating a library or application's compileSdk (which
           allows newer APIs to be used) can be done separately from updating
           targetSdk (which opts the app in to new runtime behavior) and
           minSdk (which determines which devices the app can be installed
           on).

       2.  Dependency 'androidx.window:window:1.2.0' requires libraries and applications that
           depend on it to compile against version 34 or later of the
           Android APIs.

           :app is currently compiled against android-33.

           Also, the maximum recommended compile SDK version for Android Gradle
           plugin 8.0.2 is 33.

           Recommended action: Update this project's version of the Android Gradle
           plugin to one that supports 34, then update this project to use
           compileSdk of at least 34.

           Note that updating a library or application's compileSdk (which
           allows newer APIs to be used) can be done separately from updating
           targetSdk (which opts the app in to new runtime behavior) and
           minSdk (which determines which devices the app can be installed
           on).

       3.  Dependency 'androidx.window:window-java:1.2.0' requires libraries and applications that
           depend on it to compile against version 34 or later of the
           Android APIs.

           :app is currently compiled against android-33.

           Also, the maximum recommended compile SDK version for Android Gradle
           plugin 8.0.2 is 33.

           Recommended action: Update this project's version of the Android Gradle
           plugin to one that supports 34, then update this project to use
           compileSdk of at least 34.

           Note that updating a library or application's compileSdk (which
           allows newer APIs to be used) can be done separately from updating
           targetSdk (which opts the app in to new runtime behavior) and
[... 197 lines cut ...]
           :app is currently compiled against android-33.

           Also, the maximum recommended compile SDK version for Android Gradle
           plugin 8.0.2 is 33.

           Recommended action: Update this project's version of the Android Gradle
           plugin to one that supports 34, then update this project to use
           compileSdk of at least 34.

           Note that updating a library or application's compileSdk (which
           allows newer APIs to be used) can be done separately from updating
           targetSdk (which opts the app in to new runtime behavior) and
           minSdk (which determines which devices the app can be installed
           on).

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 2s
Error: Gradle task assembleDebug failed with exit code 1
```

MrCsabaToth, 2024-11-10:
The bottom line is that `discoverServices` fail, while `1.33.6` has no problem.

chipweinberger, 2024-11-10:
> I/flutter (27428): FlutterBluePlusException | discoverServices | fbp-code: 6 | device is not connected

this error seems unrelated to anything I've changed recently.

I think your device disconnected for some reason.

chipweinberger, 2024-11-10:
the basic code failing, chr writes, made sense because I changed how characteristics are located to support includedServices. That should be fixed now.

but discoverServices did not change.

chipweinberger, 2024-11-10:
closing because I dont think this is a real issue. just a disconnected device. more logs would be necessary.

closed 2024-11-10 by chipweinberger

MrCsabaToth, 2024-11-10:
I don't know the root cause but I narrowed it down to that version bump and then verified it switching between them back and forth three times, so it was very deterministically reproducible.

chipweinberger, 2024-11-10:
okay. thanks for reporting. maybe more reports will come in.

chipweinberger, 2024-11-10:
its worth point out, 1.34.0 is broken. dont use it. use 1.34.3

chipweinberger, 2024-11-10:
at the very least, please log the disconnect reason.

chipweinberger, 2024-11-11:
good luck! thanks for your work.

MrCsabaToth, 2024-11-12:
Thanks for your work! I softened the title because I'm not sure now. If next time there will be some exception I'll try to debug more.